fwupd/fwupd #5227: cros-ec firmware parser broken since 1.8.3: not searching magic due to install flags: invalid magic for file

bleungatchromium, 2022-11-01:
Filing an issue for Boris on my team.

Boris noticed that all recent versions of fwupd are failing to flash/open firmware files with this error:
`not searching magic due to install flags: invalid magic for file`

Which is coming from [here](https://github.com/fwupd/fwupd/blob/main/libfwupdplugin/fu-firmware.c#L826) and the magic check is for fmap.

Boris checked the __FMAP__ magic is still present in firmware binaries. Not sure why [this check](https://github.com/fwupd/fwupd/blob/f0980cbddaee9b50bbf8df85f280aff6f71ca442/libfwupdplugin/fu-fmap-firmware.c#L29) is failing.

My question is why is fwupd checking for the magic at all inside the "NO SEARCH" condition? 
```
if (!fu_firmware_has_flag(self, FU_FIRMWARE_FLAG_ALWAYS_SEARCH) &&
   (flags & FWUPD_INSTALL_FLAG_NO_SEARCH) > 0) {
    if (!klass->check_magic(self, fw, *offset, error)) {   <----- shouldn't that be skipped?
```

bleungatchromium, 2022-11-01:
@the-sneaky-fox  who reported this, and is his report above.

bleungatchromium, 2022-11-01:
This fails for both the `cros-ec` and `fmap` firmware types when selecting firmware-parse.

hughsie, 2022-11-01:
> when selecting firmware-parse

I guess that's explained, but not expected; but that shouldn't affect updating. If you specify `--force` we don't set `FWUPD_INSTALL_FLAG_NO_SEARCH` in `fu-tool.c` and it should work. Maybe we shouldn't be re-using the --force command line to mean two different things. Maybe having `--no-search` (default off) is probably better.

Flashing should be 100% unaffected, and that's the bit I'm worried about. Is this perhaps from using `fwupdtool install-blob` perhaps? If not, can you get a daemon log please.

the-sneaky-fox, 2022-11-01:
The flashing command is `fwupdtool --plugins=cros-ec install-blob firmware.bin`, it fails with the same error message

hughsie, 2022-11-01:
I can fix in the morning.

bleungatchromium, 2022-11-02:
Thanks Richard. 

For your reference, [0c51630991725d29255a4e4774f92185f6a24aaa](https://github.com/fwupd/fwupd/commit/0c51630991725d29255a4e4774f92185f6a24aaa) is the first commit that fails.

closed 2022-11-02 by hughsie