fwupd/fwupd #5251: Firmware update (command line) Fails without LVFS:

sureshmanisns, 2022-11-09:
**Describe the question**
When I try to use the .cab file to update my dongle device, I get an error that says, "Message recipient disconnected from message bus without replying."  Actually firmware did not get updated.

**fwupd version information**
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.13.0-30-generic
compile   com.hughsie.libjcat           0.1.4
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.4

More Info:
Screenshot attaced 
[image]
[image]
[image]

hughsie, 2022-11-09:
Is that using the synaptics-cape protocol? If so, can you get a `sudo fwupdtool -v --plugins synaptics-cape install *.cab` -- and if it crashes, also a backtrace please?

sureshmanisns, 2022-11-09:
Hi [hughsie](https://github.com/hughsie),
No it uses the qualcomm(com.qualcomm.dfu) protocol (dfu_csr plugin).

superm1, 2022-11-09:
> I don't think it's this if you're running 1.7.10: https://github.com/fwupd/fwupd/commit/824f634244d9ca04ae405b70c66452a00e37d181
> 
> Can you do the same thing with -- `sudo fwupdtool -v --plugins dfu_csr install *.cab` instead please. A backtrace would be wonderful too.

He said 1.7.9 in description

hughsie, 2022-11-09:
> He said 1.7.9 in description

Ohh by bad. Then it'll be that. I think 1.7.10 will fix your problem.

sureshmanisns, 2022-11-09:
Hi ,
It fails in the following place:

Writing…                 [*************************************  ]13:50:34:0537 FuPluginDfuCsr       writing 0 bytes of data
13:50:34:0537 FuCommon             fu_memcpy_safe: assertion 'src != NULL' failed
13:50:34:0539 FuPlugin             attach(dfu_csr)
13:50:34:0542 FuDeviceList         no replug or re-enumerate required

sureshmanisns, 2022-11-09:
> > He said 1.7.9 in description
> 
> Ohh my bad. Then it'll be that. I think 1.7.10 will fix your problem.

Oh ok. where can I get the 1.7.10 version ? I just used  "sudo apt-get install fwupd" it installs 1.7.9

hughsie, 2022-11-09:
>  13:50:34:0537 FuCommon fu_memcpy_safe: assertion 'src != NULL' failed

If you run gdb like `sudo G_DEBUG=fatal-criticals fwupdtool -v --plugins dfu_csr install *.cab` then it'll tell us the callchain of where exactly it's failing.

sureshmanisns, 2022-11-09:
> > 13:50:34:0537 FuCommon fu_memcpy_safe: assertion 'src != NULL' failed
> 
> If you run gdb like `sudo G_DEBUG=fatal-criticals fwupdtool -v --plugins dfu_csr install *.cab` then it'll tell us the callchain of where exactly it's failing.

Backtrace:
Writing…                 [*************************************  ]14:28:02:0575 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0609 FuPluginDfuCsr       timeout=28
14:28:02:0609 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0609 FuPluginDfuCsr       status=OK
14:28:02:0609 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0644 FuPluginDfuCsr       timeout=28
14:28:02:0644 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0644 FuPluginDfuCsr       status=OK
14:28:02:0644 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0677 FuPluginDfuCsr       timeout=28
14:28:02:0677 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0677 FuPluginDfuCsr       status=OK
14:28:02:0677 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0712 FuPluginDfuCsr       timeout=1053
14:28:02:0712 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0712 FuPluginDfuCsr       status=OK
14:28:02:0712 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0769 FuPluginDfuCsr       timeout=28
14:28:02:0769 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0769 FuPluginDfuCsr       status=OK
14:28:02:0769 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0803 FuPluginDfuCsr       timeout=28
14:28:02:0803 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0803 FuPluginDfuCsr       status=OK
14:28:02:0803 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0837 FuPluginDfuCsr       timeout=28
14:28:02:0837 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0837 FuPluginDfuCsr       status=OK
14:28:02:0837 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0871 FuPluginDfuCsr       timeout=28
14:28:02:0871 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0871 FuPluginDfuCsr       status=OK
14:28:02:0871 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0905 FuPluginDfuCsr       timeout=28
14:28:02:0905 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0905 FuPluginDfuCsr       status=OK
14:28:02:0905 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0939 FuPluginDfuCsr       timeout=28
14:28:02:0939 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0939 FuPluginDfuCsr       status=OK
14:28:02:0939 FuPluginDfuCsr       writing 1017 bytes of data
14:28:02:0973 FuPluginDfuCsr       timeout=28
14:28:02:0973 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:02:0973 FuPluginDfuCsr       status=OK
14:28:02:0973 FuPluginDfuCsr       writing 1017 bytes of data
14:28:03:0008 FuPluginDfuCsr       timeout=9720
14:28:03:0008 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:03:0008 FuPluginDfuCsr       status=OK
14:28:03:0008 FuPluginDfuCsr       writing 236 bytes of data
14:28:03:0133 FuPluginDfuCsr       timeout=35415
14:28:03:0133 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
14:28:03:0133 FuPluginDfuCsr       status=OK
Writing…                 [*************************************  ]14:28:03:0133 FuPluginDfuCsr       writing 0 bytes of data
14:28:03:0133 FuCommon             fu_memcpy_safe: assertion 'src != NULL' failed
Trace/breakpoint trap

hughsie, 2022-11-09:
Sorry, i mean like `sudo G_DEBUG=fatal-criticals gdb --args fwupdtool -v --plugins dfu_csr install *.cab` then do `bt` when it crashes.

sureshmanisns, 2022-11-10:
> Sorry, i mean like `sudo G_DEBUG=fatal-criticals gdb --args fwupdtool -v --plugins dfu_csr install *.cab` then do `bt` when it crashes.

After backtrace:
Writing…                 [*************************************  ]06:52:41:0070 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0104 FuPluginDfuCsr       timeout=28
06:52:41:0104 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0104 FuPluginDfuCsr       status=OK
06:52:41:0104 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0138 FuPluginDfuCsr       timeout=28
06:52:41:0138 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0138 FuPluginDfuCsr       status=OK
06:52:41:0138 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0172 FuPluginDfuCsr       timeout=28
06:52:41:0172 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0172 FuPluginDfuCsr       status=OK
06:52:41:0172 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0207 FuPluginDfuCsr       timeout=1053
06:52:41:0207 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0207 FuPluginDfuCsr       status=OK
06:52:41:0207 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0263 FuPluginDfuCsr       timeout=28
06:52:41:0263 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0263 FuPluginDfuCsr       status=OK
06:52:41:0263 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0297 FuPluginDfuCsr       timeout=28
06:52:41:0297 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0297 FuPluginDfuCsr       status=OK
06:52:41:0297 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0331 FuPluginDfuCsr       timeout=28
06:52:41:0331 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0331 FuPluginDfuCsr       status=OK
06:52:41:0331 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0366 FuPluginDfuCsr       timeout=28
06:52:41:0366 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0366 FuPluginDfuCsr       status=OK
06:52:41:0366 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0400 FuPluginDfuCsr       timeout=28
06:52:41:0400 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0400 FuPluginDfuCsr       status=OK
06:52:41:0400 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0434 FuPluginDfuCsr       timeout=28
06:52:41:0434 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0434 FuPluginDfuCsr       status=OK
06:52:41:0434 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0468 FuPluginDfuCsr       timeout=28
06:52:41:0468 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0468 FuPluginDfuCsr       status=OK
06:52:41:0468 FuPluginDfuCsr       writing 1017 bytes of data
06:52:41:0503 FuPluginDfuCsr       timeout=9720
06:52:41:0503 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0503 FuPluginDfuCsr       status=OK
06:52:41:0503 FuPluginDfuCsr       writing 236 bytes of data
06:52:41:0628 FuPluginDfuCsr       timeout=35415
06:52:41:0628 FuPluginDfuCsr       state=dfuDNLOAD-IDLE
06:52:41:0628 FuPluginDfuCsr       status=OK
Writing…                 [*************************************  ]06:52:41:0628 FuPluginDfuCsr       writing 0 bytes of data
06:52:41:0628 FuCommon             fu_memcpy_safe: assertion 'src != NULL' failed

Thread 1 "fwupdtool" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff7bd0295 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt full
#0  0x00007ffff7bd0295 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#1  0x00007ffff7bd1579 in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#2  0x00007ffff7bd1743 in g_log () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#3  0x00007ffff7f0c27d in fu_memcpy_safe () from /lib/x86_64-linux-gnu/libfwupdplugin.so.5
No symbol table info available.
#4  0x00007ffff0ff5cb2 in ?? () from /usr/lib/x86_64-linux-gnu/fwupd-plugins-5/libfu_plugin_dfu_csr.so
No symbol table info available.
#5  0x00007ffff0ff6601 in ?? () from /usr/lib/x86_64-linux-gnu/fwupd-plugins-5/libfu_plugin_dfu_csr.so
No symbol table info available.
#6  0x00007ffff7f1c217 in fu_device_write_firmware () from /lib/x86_64-linux-gnu/libfwupdplugin.so.5
No symbol table info available.
#7  0x00007ffff7f2b22f in fu_plugin_runner_write_firmware () from /lib/x86_64-linux-gnu/libfwupdplugin.so.5
No symbol table info available.
#8  0x000055555558a7c1 in fu_engine_install_blob ()
No symbol table info available.
#9  0x000055555558b217 in ?? ()
No symbol table info available.
#10 0x000055555558ba12 in fu_engine_install ()
No symbol table info available.
#11 0x000055555558beaf in fu_engine_install_tasks ()
No symbol table info available.
#12 0x0000555555576069 in ?? ()
No symbol table info available.
#13 0x000055555557a7cd in fu_util_cmd_array_run ()
No symbol table info available.
#14 0x0000555555570a4b in main ()
No symbol table info available.

hughsie, 2022-11-10:
@sureshmanisns I've spotted something that might be the issue. I don't have any DFU-CSR hardware any more (my AIAIAI headphones no longer boot) but if you could try with that PR that would be wonderful. I can build a Fedora test .rpm if that would be helpful.

sureshmanisns, 2022-11-10:
> @sureshmanisns I've spotted something that might be the issue. I don't have any DFU-CSR hardware any more (my AIAIAI headphones no longer boot) but if you could try with that PR that would be wonderful. I can build a Fedora test .rpm if that would be helpful.

Ok, thanks for your update. Is it possible for you to build the Debian package for Ubuntu 20.04? because I don't have Fedora.

hughsie, 2022-11-10:
> Is it possible for you to build the Debian package for Ubuntu

I don't know how, sorry.

sureshmanisns, 2022-11-10:
Hi [hughsie](https://github.com/hughsie),
It would be helpful if you build a Fedora test.rpm and shared it so that I could verify the fix.

hughsie, 2022-11-10:
Done: https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

g0-n, 2023-04-03:
Hi @hughsie,

I tried to update my AIAIAI H05 in SteamOS (over Steam Deck). Console showed this message "Message recipient disconnected from message bus without replying" while writing the firmware (1.5 to 1.6). Now I can't power the headphones on, nor downgrade them to 1.5 as I keep getting that same message.
I'm not very good with command line, I don't know how to report this properly.
Any help?
Thank u very much

hughsie, 2023-04-03:
@g0-n what fwupd version?

g0-n, 2023-04-03:
Oh, thank you!
1.8.14 is the version. Got it from Discover SC (flatpack).

hughsie, 2023-04-04:
@g0-n Hmm, so that *should* work. My AIAIAI H05 device I use for CI testing no longer works -- I think it's been updated one too many times perhaps. I've asked about a new device but no luck so far. Is it possible for you to get a coredump or backtrace of the daemon process?

g0-n, 2023-04-04:
Ok @hughsie 
Do I need to do that using --verbose?

fwupdmgr update --verbose

g0-n, 2023-04-05:
Hi again @hughsie!
I'm sorry this is taking me so long. Need to figure out many things, as I told you I'm very new with all of this :-(
`sudo gdb --args /usr/libexec/fwupd/fwupd -vv` when running this I get 

> fwupd no such file or directory

I looked for the fwupd directory and I think I have it in /usr/lib. I wonder, shouldn't I run `sudo gdb --args fwupdmgr update -vv` ?

superm1, 2023-04-05:
> Hi again @hughsie!
> I'm sorry this is taking me so long. Need to figure out many things, as I told you I'm very new with all of this :-(
> `sudo gdb --args /usr/libexec/fwupd/fwupd -vv` when running [...]

It needs to be the daemon to get daemon crashes.

g0-n, 2023-04-05:
oh, ok, i see

thank u!

g0-n, 2023-04-05:
I moved the query over there....