fwupd/fwupd #5254: COD is disabled by default because of problems on Insyde

superm1, 2022-11-09:
We've intentionally disabled CoD for anything but aarch64 intentionally because of a bug found in Insyde BIOS.
This was changed in 1.8.3 in https://github.com/fwupd/fwupd/commit/77880751cddab056ebc653b1e991ff88f60603eb

That was also brought back to 1.7.x release.  It's unfortunate that we have to flip the default though for everyone because of this issue happening in Insyde.

I noticed that in the [press release from Insyde](https://www.insyde.com/press_news/blog/uefi-24-review-part-15-capsule-disk) they were a major proponent for this feature.  I wonder if perhaps there is a nuance to the fwupd implementation that is missed from the spec.

@jbobzin-insyde is explicitly mentioned in that press release.  Would you be able to review the fwupd implementation for using COD against the Insyde implementation for supporting it?  If we have a bug, perhaps we can fix it in fwupd and then let CoD be allowed again by default for X86.

superm1, 2022-11-09:
@jbobzin-insyde The background of why we disabled it by default is in https://github.com/fwupd/fwupd/issues/4823

superm1, 2022-12-06:
@hughsie I had some offline discussion about this.  It's claimed to be fixed in the latest Insyde code.  So we've got a few approaches we can use:

1. Quirk the broken cases.
2. Drop the `uefi-allow-cod` check and use a heuristic to conditionally enable on non-aarch64.
3. Some combination of these.

The problem with the quirking approach is I don't currently know right now how widespread Insyde BIOS that advertises support for CoD but has this bug actually is.
The problem with the heuristic approach is that we don't know the right criteria.
My current best guess for a criteria would be check the to look at SMBIOS DMI type 0.  If we see BIOS vendor as "INSYDE Corp." then look for a BIOS release date of 11/01/2022 or later, or even make it 2023 or later.

The problem will end up being if a broken system has a BIOS update but still has the older Insyde code.  So maybe a quirk to block the HP Dev One until we know it's really got the fix in place, and a heuristic together?

Thoughts?

superm1, 2022-12-08:
Having just filed https://github.com/fwupd/fwupd/pull/5306 I had a thought on what the root cause for this might be...
Could it be that we're not calling `fu_uefi_device_fixup_firmware` for the cod case?

superm1, 2022-12-08:
> Having just filed #5306 I had a thought on what the root cause for this might be... Could it be that we're not calling `fu_uefi_device_fixup_firmware` for the cod case?

I gave this a try on such a system, but it didn't help :/

superm1, 2022-12-09:
> Doesn't CoD actually require a capsule header? I suppose if it's just looking for a file with a specific extension maybe it could blindly just apply it to the system BIOS, but you'd think you need to read the header to know what to flash, if you see what I mean.

I've never seen CoD working, so I don't know for sure.  I suppose by that argument we should be calling `fu_uefi_device_fixup_firmware` in CoD case even if it doesn't fix the Insyde bug.

> Back a bit tho; why do we want to support CoD -- do newer platforms only support this rather than NVMRAM? Or is there a compelling reason to switch if it's available?

It would be a simpler and quicker flash process, don't need fwupdx64.efi involved and as many reboots.

hughsie, 2022-12-09:
> we should be calling fu_uefi_device_fixup_firmware in CoD case

I think we should be stricter than that, i.e. only allow for the header matching payload or to be the FMP GUID -- we really don't want to be auto-prepending a capsule header if the end result might be wildly not the right thing.

> It would be a simpler and quicker flash process

Could Insyde tell us the number of models that we'd have to quirk to work around the "even tho CoD is advertised, it doesn't work" bug? I think slower flashing process is better than unintentional regression.

superm1, 2022-12-09:
> we really don't want to be auto-prepending a capsule header if the end result might be wildly not the right thing.

By that logic how come we add a header for NVRAM and GRUB case though?  

> Could Insyde tell us the number of models that we'd have to quirk to work around the "even tho CoD is 
> advertised, it doesn't work" bug? I think slower flashing process is better than unintentional regression.

I don't think they can.  I don't think they track what the IBVs and OEMs put into all the machines that closely.

hughsie, 2022-12-09:
> By that logic how come we add a header for NVRAM and GRUB case though?

I guess just legacy. If a RT->UpdateCapsule2 was invented now I'd probably argue we shouldn't do the fixup either.

> I don't think they track what the IBVs and OEMs put into all the machines that closely.

Maybe they don't but maybe we could? i.e. record in the LVFS report metadata if CoD was advertised as supported?

superm1, 2022-12-09:
> I guess just legacy. If a RT->UpdateCapsule2 was invented now I'd probably argue we shouldn't do the fixup either.
>
>
> Maybe they don't but maybe we could? i.e. record in the LVFS report metadata if [...]

Given it needs a client side change I think it would take quite a while to data that was at large enough scale to trust enough to make a policy decision.

I guess for now we don't have a pressing reason to re-enable COD, we should probably just revisit this if we end up with machines some day that don't support UpdateCapsule RT service.

closed 2022-12-09 by superm1