fwupd/fwupd #5258: fwupd follows infinite symlink cycles until memory full or error.

keskival, 2022-11-13:
**Describe the bug**
A clear and concise description of what the bug is.

Cannot run `fwupdmgr update` successfully, because `fwupd` process gets into an almost infinite loop going through symbolic links in `/var` and `/proc` directories.

See the stored error output of `fwupdmgr get-devices` here:
```
Devices that were not updated correctly:

 • UEFI dbx (78 → 217)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]:
Y
Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "c6ae2e9be984534b5e3c515d9bc17046bf8b7bdbf3a7f05725ec39c1ae4453e4",
                           "Metadata" : {
                             "DistroId" : "ubuntu",
                             "DistroVersion" : "22.04"
                           },
                           "Reports" : [
                             {
                               "Checksum" : "ef237384c9099e0a7724e0769ae3887d3e11ca81",
                               "ReleaseId" : null,
                               "UpdateState" : 3,
                               "UpdateError" : "Blocked executable in the ESP, ensure grub and shim are up to date: Error opening directory “/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/proc/self/task/2555/fd/16/sys/kernel/debug/tracing”: Too many levels of symbolic links",
                               "Guid" : [
                                 "fda6234b-adcb-5105-8515-9af647d29775"
                               ],
                               "Plugin" : "uefi_dbx",
                               "VersionOld" : "78",
                               "VersionNew" : "217",
                               "Flags" : 4194595,
                               "Created" : 1668362835,
                               "Modified" : 1668363660,
                               "Metadata" : {
                                 "HostSku" : "All",
                                 "DistroVersion" : "22.04",
                                 "RuntimeVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "FwupdSupported" : "True",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.7.9",
                                 "SecureBoot" : "Disabled",
                                 "HostFamily" : "ASUS MB",
                                 "UEFIUXCapsule" : "Disabled",
                                 "KernelName" : "Linux",
                                 "RuntimeVersion(org.kernel)" : "5.15.0-52-generic",
                                 "CompileVersion(com.hughsie.libjcat)" : "0.1.9",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.3.10",
                                 "DistroId" : "ubuntu",
                                 "CpuArchitecture" : "x86_64",
                                 "BootTime" : "1668362739",
                                 "HostVendor" : "ASUS",
                                 "EfivarNvramUsed" : "30235",
                                 "HostProduct" : "All Series",
                                 "FwupdTainted" : "False",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.7.9",
                                 "KernelVersion" : "5.15.0-52-generic"
                               }
                             }
                           ]
                         }
```

I also got a similar error previously from `/var` directory which I fixed by removing the offending symlink.

**Steps to Reproduce**
Steps to reproduce the behavior.
[... 483 lines cut ...]
├── initrd.img.old -> initrd.img-5.15.0-50-generic
├── lost+found
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.13.0-28-generic
├── System.map-5.15.0-50-generic
├── System.map-5.15.0-52-generic
├── vmlinuz -> vmlinuz-5.15.0-52-generic
├── vmlinuz-5.13.0-28-generic
├── vmlinuz-5.15.0-50-generic
├── vmlinuz-5.15.0-52-generic
└── vmlinuz.old -> vmlinuz-5.15.0-50-generic

10 directories, 314 files
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting? Yes.
- Is this a regression? Nope.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? I believe it is, but not completely sure.
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No.

superm1, 2022-11-13:
Something like this happened recently in another bug report. By chance is your ESP not marked as an ESP?

hughsie, 2022-11-13:
We fixed this in 1.8.7 last week.

closed 2022-11-13 by hughsie

superm1, 2022-11-13:
Right it's probably a duplicate of #5211.  If it's the same root cause they could at least work around it until they upgrade.