fwupd/fwupd #5272: For DW5821e: How to solve "failed to device before update reload: failed to wait for detach replug: device {id} did not come back" when fwupdmgr update to upgrade firmware?

Fanice-luo, 2022-11-21:
**Describe the bug**
I was trying update or downgrade the firmware to fwupdmgr  update or downgrade, but the output is : "failed to device before update reload: failed to wait for detach replug: device {id} did not come back".

**Steps to Reproduce**
1.fwupdate get-devices
2.sudo fwupdmgr downgrade XXX
mcs@mcs-Lenovo-Product:~$ sudo fwupdmgr downgrade 699fb7ed17170b9dfc89cf89162c623c0919bef4
Choose a release:
0.	Cancel
1.	T77W968.F1.0.0.4.9.GC.012
2.	T77W968.F1.0.0.4.7.GC.011
1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Downgrade DW5821e-eSIM Snapdragon X20 LTE from                               ║
║ T77W968.F1.0.0.5.2.DF.032.049 to T77W968.F1.0.0.4.9.GC.012?                  ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release T77W968.F1.0.0.4.9 packaged for LVFS.                       ║
║                                                                              ║
║ DW5821e-eSIM Snapdragon X20 LTE and all connected devices may not be usable  ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Waiting…                 [*************************************  ]failed to get device before update reload: failed to wait for detach replug: device 1f914e619b232dc4da68ec3389b98b20163003a2 did not come back

This is the same as fwupdmgr update.

**fwupd version information**
mcs@mcs-Lenovo-Product:~$ fwupdmgr --version 
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.0-20220806
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.3.10

fwupdmgr get-devices
mcs@mcs-Lenovo-Product:~$ sudo fwupdmgr get-devices 
LENOVO FFFFFFFFFF
│
├─Core™ i3-8145U CPU @ 2.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internal device
│   
├─DW5821e-eSIM Snapdragon X20 LTE:
│     Device ID:          699fb7ed17170b9dfc89cf89162c623c0919bef4
│     Summary:            Dell DW5821e/eSIM LTE modem
│     Current version:    T77W968.F1.0.0.5.2.DF.032.049
│     Vendor:             Dell Inc. (USB:0x413C)
│     GUIDs:              4014afd7-7bc7-596d-a2c9-25344cb8c80b
│                         8782e171-6010-563f-8696-ec7020cd8c6f ← USB\VID_413C&PID_81E0&REV_0318&CARRIER_DEFAULT
│                         f8f7b616-b179-5957-b17e-69826c481e73 ← USB\VID_413C&PID_81E0&REV_0318
│                         f93bb794-00c7-569d-b6c1-30b4a5641cdb ← USB\VID_413C&PID_81E0
│                         795e079d-093b-5503-aa59-35b832480e95 ← USB\VID_413C
│     Device Flags:       • Updatable
│                         • System requires external power source
[... 83 lines cut ...]
│     Device ID:          f95c9218acd12697af946874bfe4239587209232
│     Summary:            UEFI ESRT device
│     Current version:    4784136
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               13aaf4e2-706f-4ed3-af7b-b61634eff1a8
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─WhiskeyLake-U GT2 [UHD Graphics 620]:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              52aa8ac7-c837-561f-af32-b7979cd2f6ed ← PCI\VEN_8086&DEV_3EA0
                          6986c21a-0f3f-5538-bcef-fe4e8cda7ab7 ← PCI\VEN_8086&DEV_3EA0&REV_02
                          f61f4d93-6678-5248-bd05-08de10e12af8 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA314D
                          8f489980-a33a-599f-a220-bffb133a77e2 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA314D&REV_02
                          1bb467a3-4af9-53a9-ab48-f588ee6e8066 ← PCI\VEN_8086&DEV_3EA0&REV_00
                          1dc190d6-5b18-58a7-a7e6-560bbc2e2c06 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA314D&REV_00
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
Best Wishes, Fanice

aleksander0m, 2022-11-21:
Hello Fanice-luo!

If I'm not mistaken you're working for the manufacturer, and you recently tested this procedure for a different bug and it was working fine, right? Can you remember which version of fwupd you tested?

Is this a different module with different vid:pids than the original ones?

I suggest you run fwupd with verbose logs enabled to see the whole flow and to see exactly where it breaks.

Fanice-luo, 2022-12-07:
Hi @aleksander0m 
1.When I recently tested this procedure for a different bug and it wasn't working fine for fwupd1.8.7.
2.I tested fwupd1.6.5 can update successfully but return info: "failed to wait for detach replug: device {id} did not come back", command "lsusb" can identify my device T77W968(DW5821e, PID:VID is 413c:81d7), "mmcli -L" return "No modems were found", need restart ModemManager.service can obtain device.
3.I compared the difference of fwupd1.6.2 and fwupd1.7.0, found something affect the update progress.
Here is some changes at commit :
From 40cd18fa976d521e99efca0a7eefc3b0b2029e84 Mon Sep 17 00:00:00 2001
From: Richard Hughes <richard@hughsie.com>
Date: Thu, 26 Aug 2021 07:52:29 +0100

 static gboolean
-fu_fastboot_device_attach(FuDevice *device, GError **error)
+fu_fastboot_device_attach(FuDevice *device, FuProgress *progress, GError **error)
 {
-	fu_device_set_status(device, FWUPD_STATUS_DEVICE_RESTART);
-	if (!fu_fastboot_device_cmd(device, "reboot", FU_FASTBOOT_DEVICE_READ_FLAG_NONE, error))
+	if (!fu_fastboot_device_cmd(device,
+				    "reboot",
+				    progress,
+				    FU_FASTBOOT_DEVICE_READ_FLAG_NONE,
+				    error))
 		return FALSE;
 	fu_device_add_flag(device, FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG);
 	return TRUE;
 }
 
+static void
+fu_fastboot_device_set_progress(FuDevice *self, FuProgress *progress)
+{
+	fu_progress_set_id(progress, G_STRLOC);
+	fu_progress_add_flag(progress, FU_PROGRESS_FLAG_GUESSED);
+	fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 2); /* detach */
+	fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_WRITE, 94);	/* write */
+	fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_RESTART, 2); /* attach */
+	fu_progress_add_step(progress, FWUPD_STATUS_DEVICE_BUSY, 2);	/* reload */
+}
+
 static void
 fu_fastboot_device_init(FuFastbootDevice *self)
 {
@@ -709,4 +768,5 @@ fu_fastboot_device_class_init(FuFastbootDeviceClass *klass)
 	klass_device->set_quirk_kv = fu_fastboot_device_set_quirk_kv;
 	klass_device->open = fu_fastboot_device_open;
 	klass_device->close = fu_fastboot_device_close;
+	klass_device->set_progress = fu_fastboot_device_set_progress;

aleksander0m, 2022-12-14:
Not really sure that commit has anything to do with the issue.

I guess if the upgrade worked fine with 1.6.2 but not with 1.7.0 you could probably do a `git bisect` to try to find exactly the culprit?

hughsie, 2023-09-23:
Agreed, a bisect would be the only way we can debug this without the actual hardware. Sorry!

closed 2023-09-23 by hughsie