fwupd/fwupd #5275: fwupdmgr security: incorrect detection of IOMMU being present on host or not.

johannbg, 2022-11-21:
**Describe the bug**
When running fupdmgr security on Lenovo X1 Carbon Gen 5 ( W10DG ) it incorrectly reports in the HSI-2 section that IOMMU is not-found on the host system when it is most certainly present and found via simple `dmesg | grep IOMMU` and enabling it as suggested by the resolution section as in  

`Resolution: If available, turn on IOMMU in the system BIOS. You may also have to use additional kernel boot parameters, for example intel_iommu=on iommu=pt.`

does work.

```shell
dmesg | grep -i -e DMAR -e IOMMU
[    0.000000] Command line: initrd=\aab1f5db3229412687c416467bce6bf7\6.0.9-300.fc37.x86_64\initrd root=UUID=4d8f51db-7aeb-49f5-8b5d-1dd543116e60 ro rootflags=subvol=root rhgb quiet systemd.machine_id=aab1f5db3229412687c416467bce6bf7 intel_iommu=on iommu=pt
[    0.017204] ACPI: DMAR 0x000000005B1C0000 0000A8 (v01 LENOVO TP-N1M   00001560 PTEC 00000002)
[    0.017243] ACPI: Reserving DMAR table memory at [mem 0x5b1c0000-0x5b1c00a7]
[    0.100049] Kernel command line: initrd=\aab1f5db3229412687c416467bce6bf7\6.0.9-300.fc37.x86_64\initrd root=UUID=4d8f51db-7aeb-49f5-8b5d-1dd543116e60 ro rootflags=subvol=root rhgb quiet systemd.machine_id=aab1f5db3229412687c416467bce6bf7 intel_iommu=on iommu=pt
[    0.100185] DMAR: IOMMU enabled
[    0.180694] DMAR: Host address width 39
[    0.180695] DMAR: DRHD base: 0x000000fed90000 flags: 0x0
[    0.180702] DMAR: dmar0: reg_base_addr fed90000 ver 1:0 cap 1c0000c40660462 ecap 7e3ff0505e
[    0.180705] DMAR: DRHD base: 0x000000fed91000 flags: 0x1
[    0.180709] DMAR: dmar1: reg_base_addr fed91000 ver 1:0 cap d2008c40660462 ecap f050da
[    0.180711] DMAR: RMRR base: 0x0000005a69f000 end: 0x0000005a6befff
[    0.180713] DMAR: RMRR base: 0x0000005d000000 end: 0x0000005f7fffff
[    0.180729] DMAR-IR: IOAPIC id 2 under DRHD base  0xfed91000 IOMMU 1
[    0.180733] DMAR-IR: HPET id 0 under DRHD base 0xfed91000
[    0.180734] DMAR-IR: Queued invalidation will be enabled to support x2apic and Intr-remapping.
[    0.182558] DMAR-IR: Enabled IRQ remapping in x2apic mode
[    0.334506] iommu: Default domain type: Passthrough (set via kernel command line)
[    0.375610] DMAR: No ATSR found
[    0.375611] DMAR: No SATC found
[    0.375612] DMAR: IOMMU feature fl1gp_support inconsistent
[    0.375613] DMAR: IOMMU feature pgsel_inv inconsistent
[    0.375614] DMAR: IOMMU feature nwfs inconsistent
[    0.375615] DMAR: IOMMU feature eafs inconsistent
[    0.375616] DMAR: IOMMU feature prs inconsistent
[    0.375617] DMAR: IOMMU feature nest inconsistent
[    0.375618] DMAR: IOMMU feature mts inconsistent
[    0.375619] DMAR: IOMMU feature sc_support inconsistent
[    0.375620] DMAR: IOMMU feature dev_iotlb_support inconsistent
[    0.375621] DMAR: dmar0: Using Queued invalidation
[    0.375625] DMAR: dmar1: Using Queued invalidation
[    0.375821] pci 0000:00:00.0: Adding to iommu group 0
[    0.375833] pci 0000:00:02.0: Adding to iommu group 1
[    0.375843] pci 0000:00:08.0: Adding to iommu group 2
[    0.375861] pci 0000:00:14.0: Adding to iommu group 3
[    0.375870] pci 0000:00:14.2: Adding to iommu group 3
[    0.375886] pci 0000:00:16.0: Adding to iommu group 4
[    0.375899] pci 0000:00:1c.0: Adding to iommu group 5
[    0.375910] pci 0000:00:1c.2: Adding to iommu group 6
[    0.375930] pci 0000:00:1c.4: Adding to iommu group 7
[    0.375944] pci 0000:00:1d.0: Adding to iommu group 8
[    0.375970] pci 0000:00:1f.0: Adding to iommu group 9
[    0.375980] pci 0000:00:1f.2: Adding to iommu group 9
[    0.375990] pci 0000:00:1f.3: Adding to iommu group 9
[    0.375999] pci 0000:00:1f.4: Adding to iommu group 9
[    0.376009] pci 0000:00:1f.6: Adding to iommu group 9
[    0.376022] pci 0000:02:00.0: Adding to iommu group 10
[    0.376033] pci 0000:04:00.0: Adding to iommu group 11
[    0.376052] pci 0000:05:00.0: Adding to iommu group 12
[    0.376165] DMAR: Intel(R) Virtualization Technology for Directed I/O
[    6.451636] pci 0000:00:1f.1: Adding to iommu group 13
[    6.452422] pci 0000:00:1f.1: Removing from iommu group 13
```

Note: There is no bios option on this model to toggle IOMMU support on or off as the document is suggesting should be done.

**Steps to Reproduce**
Install Fedora 37 on the specific model and run `fwupdmgr security`, check reported IOMMU support.

**Expected behavior**
I expected in the security report that the IOMMU support was reported being found but not enabled/not configured and it being suggested to add `intel_iommu=on iommu=pt` to the kernel command line to enable it.

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.7
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.1
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.2
runtime   org.kernel                    6.0.9-300.fc37.x86_64
runtime   org.freedesktop.fwupd         1.8.7
```

johannbg, 2022-11-21:
> > IOMMU is not-found on the host system
> 
> Without the kernel args, what devices does /sys/class/iommu/dmar have? 

none, it's only after I enabled on the kernel commandline that dmar becomes present. 

The dmesg output is the following without the kernel args.

```shell
dmesg | grep -i -e DMAR -e IOMMU
[    0.017176] ACPI: DMAR 0x000000005B1C0000 0000A8 (v01 LENOVO TP-N1M   00001560 PTEC 00000002)
[    0.017215] ACPI: Reserving DMAR table memory at [mem 0x5b1c0000-0x5b1c00a7]
[    0.180585] DMAR: Host address width 39
[    0.180586] DMAR: DRHD base: 0x000000fed90000 flags: 0x0
[    0.180593] DMAR: dmar0: reg_base_addr fed90000 ver 1:0 cap 1c0000c40660462 ecap 7e3ff0505e
[    0.180596] DMAR: DRHD base: 0x000000fed91000 flags: 0x1
[    0.180600] DMAR: dmar1: reg_base_addr fed91000 ver 1:0 cap d2008c40660462 ecap f050da
[    0.180602] DMAR: RMRR base: 0x0000005a69f000 end: 0x0000005a6befff
[    0.180604] DMAR: RMRR base: 0x0000005d000000 end: 0x0000005f7fffff
[    0.180607] DMAR-IR: IOAPIC id 2 under DRHD base  0xfed91000 IOMMU 1
[    0.180609] DMAR-IR: HPET id 0 under DRHD base 0xfed91000
[    0.180610] DMAR-IR: Queued invalidation will be enabled to support x2apic and Intr-remapping.
[    0.182433] DMAR-IR: Enabled IRQ remapping in x2apic mode
[    0.333386] iommu: Default domain type: Translated
[    0.333386] iommu: DMA domain TLB invalidation policy: lazy mode
```

>That's what fwupd means by not-found; from a _userspace point of view_ it literally doesn't exist.

Should the check not be checking whether IOMMU is supported instead of something not being found?

> If you have to use `intel_iommu=on iommu=pt` then we should probably report that as a kernel bug, as IOMMU is supposed to be used by default on Fedora.

Something being used by default on Fedora does not mean that its automatically used by the rest of the ca 500 distro's out there + defaults on Fedora have a tendency to change with each Fedora release so there is that so defaults might not be the same between releases in other words downstream defaults cannot and thus should not be depended upon.

hughsie, 2022-11-21:
> Something being used by default on Fedora does not mean that its automatically used by the rest of the ca 500 distro's out there

Fedora uses the upstream kernel default.

johannbg, 2022-11-21:
Given that IOMMU went through quite the disorganized development process leading for it becoming incoherent mess in the kernel, how came it about that it got added to the hsi specification when it should be turned off by default since it has an impact on bandwidth performance ( which is probably why the kernel upstream is seemingly default to not be enabling it ), so encouraging users to enable it via it being in red letters, is that such a wise thing todo? 
Should the security report simply not report it if it's not available ( as in not found in this case )  and if it's available flag it if insecurely configured?

hughsie, 2022-11-21:
Without an IOMMU it's easy to build a malicious ThunderBolt device that can just access main unencrypted RAM.

johannbg, 2022-11-21:
Well that requires physical access to the device and we all know it's game over if the perpetrator has physical access to the device so no matter if IOMMU is enabled or not the perpetrator could just seize the device and circumvent the security chip by replacing it or short circuit between pins on the security chip to reset the bios password to disable the IOMMU support if set in the bios if the ThunderBolt device attack was unsuccessful at the point of the attack. 

From a security standpoint I would think it would be better to check for if the Thunderbolt port(s) are enabled in the BIOS and recommend individuals to permanently disable it if it's not used or between usages rather then checking for if IOMMU exist on the devices itself and flag it if it does not exist or is not enabled since it probably would be more effective to thwart thunderbolt port's attacks.

mcatanzaro, 2022-11-21:
Users plugging in malicious USB devices is one of the oldest tricks in the book....

johannbg, 2022-11-21:
Yup handing out malicious USB at a conference or label them some sponsor on site and slip them into swag bags even simply mail them to some corporation and rely on peoples own ignorance of inserting them all tried and tested methods. That said going through open windows predates that book but the solution to both remain the same, if you dont want someone or something enter through something that is open and accessible, keep it closed thus inaccessible.

hughsie, 2022-11-22:
> rely on peoples own ignorance of inserting them

I guess I'm in the ignorant camp them. I think IOMMU is an important line of defence, and hence why it's in HSI. I think you need to file a kernel bug (either upstream at kernel.org or downstream in Fedora bugzilla) about why the kernel args are required (they shouldn't be required). Ill close this issue as fwupd is doing as designed; notifying you that the IOMMU isn't actually being used.

closed 2022-11-22 by hughsie