fwupd/fwupd #5277: Intel Management Engine (CSME) version is invalid

gdonval, 2022-11-22:
**Describe the bug**

CSME version is reported as invalid. Version downgrade seems impossible.

This is on a Lenovo X1 Carbon Gen 7 using the "non-absolute persistent module" version of the BIOS (i.e. "P" bios with computrace disabled).

**Steps to Reproduce**

1. Boot PC
2. `sudo fwupdtool security --force --verbose` (see last section)
3. See "✘ CSME v0:12.0.49.1534:          Invalid"
4. Try to downgrade (gnome-firmware program).
5. Reboot when prompted.
6. See downgrade fails.

**Expected behavior**

Either my version is valid and should be accepted as such or it is invalid and downgrade to a valid one should be doable.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version

runtime   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
runtime   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    6.0.9-arch1-1
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.gusb          0.4.2
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): **pacman**

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

Please note that the report on device not updated correctly only appeared after I tried to force the downgrade.

```shell
$ fwupdmgr get-devices --show-all-devices

LENOVO 20QESAMA00
│
├─Unknown Device:
│     Device ID:          59a11bc1795b6b83caa7eface68d0ad302cf5f86
│     GUID:               0422c218-5b8a-504d-8fba-43f8a29f4312 ← GPIO\ID_INT34BB:00
│   
├─Bolt Receiver:
│ │   Device ID:          82091cc2e67e8e248d8dad9ac847708118484457
│ │   Summary:            Miniaturised USB wireless receiver
│ │   Current version:    MPR05.01_B0010
│ │   Bootloader Version: BOT30.01_B0010
│ │   Vendor:             Logitech (HIDRAW:0x046D, USB:0x046D)
│ │   Install Duration:   30 seconds
│ │   GUIDs:              02d049be-c3f5-51c4-a312-b2284b43415a
│ │                       f70d9241-2656-546b-a8e4-211bac348667 ← HIDRAW\VEN_046D&DEV_C548
│ │                       af1404c4-f038-5b3e-92b0-09bf4aa84f1c ← HIDRAW\VEN_046D&DEV_C548&REV_00
│ │   Device Flags:       • Updatable
│ │                       • Signed Payload
│ │ 
│ ├─MX MCHNCL M:
│ │ │   Device ID:        3bd87d7ac1d74ec4419fd42f0813d7702e8059c0
[... 246 lines cut ...]
                          1bb467a3-4af9-53a9-ab48-f588ee6e8066 ← PCI\VEN_8086&DEV_3EA0&REV_00
                          a0b1b319-e792-5f3c-aa9e-3c5b8a512207 ← PCI\VEN_8086&DEV_3EA0&SUBSYS_17AA2292&REV_00
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
    
________________________________________________

Devices that were not updated correctly:

 • Intel Management Engine (192.49.1534 → 192.47.1524)

Devices that have been updated successfully:

 • Embedded Controller (0.1.24 → 0.1.25)
 • System Firmware (0.1.54 → 0.1.55)
 • Prometheus (10.01.3478575 → 10.01.3478575)
```

</details>

**Additional questions**

- Operating system and version: **Archlinux**
- Have you tried rebooting? Yep.
- Is this a regression? Probably not.

gdonval, 2022-11-22:
**Verbose output**

<details>

<summary>**Verbose fwupd output**</summary>

```shell
$ sudo fwupdtool security --force --verbose

15:28:08.629 FuDebug              Verbose debugging enabled (on console 1)
15:28:08.631 FuMain               locked /run/lock/fwupdtool
15:28:08.635 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Loading…                 [-                                      ]15:28:08.662 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
15:28:08.662 FuCommon             reading /etc/fwupd/daemon.conf with 2224 bytes
15:28:08.662 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
15:28:08.662 FuConfig             using autodetected max archive size 4.1 GB
15:28:08.665 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]15:28:08.666 FuRemoteList         loading /usr/share/fwupd/metainfo
15:28:08.667 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
15:28:08.667 XbSilo               file: 45fd1523-4794-003f-5b30-12b8c7d68373, current:45fd1523-4794-003f-5b30-12b8c7d68373, cached: (null)
15:28:08.667 XbSilo               loading silo with file contents
15:28:08.667 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
15:28:08.667 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
15:28:08.668 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
15:28:08.668 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
15:28:08.668 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
15:28:08.669 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
15:28:08.669 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
15:28:08.669 FuRemoteList         ordering lvfs-testing=lvfs+1
15:28:08.669 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]15:28:08.669 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
15:28:08.669 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:28:08.669 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
15:28:08.670 Jcat                 loaded 1 certificates
15:28:08.670 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
15:28:08.670 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
15:28:08.670 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
15:28:08.670 Jcat                 loaded 1 certificates
15:28:08.670 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
15:28:08.670 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
15:28:08.670 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]15:28:08.670 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
15:28:08.671 FuHistory            got schema version of 8
Loading…                 [*                                      ]15:28:08.694 FuPlugin             load(/usr/lib/fwupd-1.8.7/libfu_plugin_modem_manager.so)
15:28:08.697 FuContext            added udev subsystem watch of platform
15:28:08.697 FuContext            added udev subsystem watch of block
15:28:08.697 FuContext            added udev subsystem watch of pci
15:28:08.697 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
15:28:08.697 FuPluginDell         Using libsmbios 2.4
15:28:08.697 FuContext            added udev subsystem watch of i2c
15:28:08.697 FuContext            added udev subsystem watch of i2c-dev
15:28:08.697 FuContext            added udev subsystem watch of hidraw
15:28:08.698 FuContext            added udev subsystem watch of gpio
15:28:08.698 FuContext            added udev subsystem watch of mei
15:28:08.698 FuContext            added udev subsystem watch of iommu
15:28:08.698 FuContext            added udev subsystem watch of msr
15:28:08.698 FuContext            added udev subsystem watch of mtd
15:28:08.699 FuContext            added udev subsystem watch of nvme
15:28:08.702 FuContext            added udev subsystem watch of drm
15:28:08.703 FuContext            added udev subsystem watch of drm_dp_aux_dev
15:28:08.703 FuContext            added udev subsystem watch of serio
15:28:08.703 FuPluginTest         init
15:28:08.703 FuContext            added udev subsystem watch of thunderbolt
15:28:08.703 FuContext            added udev subsystem watch of tpm
15:28:08.704 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
15:28:08.704 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
15:28:08.704 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
15:28:08.704 XbSilo               file: 6d430a8c-988e-72af-c653-32f375c0bc9c, current:6d430a8c-988e-72af-c653-32f375c0bc9c, cached: (null)
15:28:08.704 XbSilo               loading silo with file contents
[... cut ...]
  Guid:                 561403e8-143a-5071-ab09-bf5e4c146983
  Guid:                 d5092916-1868-5182-8c70-12f83dab6372

15:28:13.844 FuPlugin             finalize(modem_manager)
15:28:13.845 FuPlugin             finalize(flashrom)
15:28:13.845 FuPluginTest         destroy
15:28:13.849 FuProgress           raw timing data was { 0.201, 0.002, 0.049, 0.003, 0.202, 0.004 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 44 0 11 1 44 1 ] at ../fwupd-1.8.7/plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906

```

</details>

hughsie, 2022-11-22:
Not valid is telling you you need to *upgrade* as the version you're using is vulnerable to some critical security issues.

gdonval, 2022-11-22:
Hoping this clear things out:

[image]

----

So as I said, it looks like either this should be recognised as a valid version or there should be a mechanism to downgrade (downgrade fails).

hughsie, 2022-11-23:
> So as I said, it looks like either this should be recognised as a valid version

It's valid, but it's insecure; maybe the fwupd text is confusing. I think filing an issue in https://github.com/fwupd/firmware-lenovo/issues asking for a new version might be the best plan I'm afraid.

closed 2022-11-23 by hughsie