fwupd/fwupd #5280: https://fwupd.org/lvfs/devices/com.dell.uefib9870160.firmware not found

w5pny, 2022-11-24:
The latest Dell Precision 7760 firmware today is announced as available but clicking on the cab file yields "Not Found" and the fwupd update procedure gets "Not Found" as well when trying to download it.

hughsie, 2022-11-24:
I get the same. Doing a fsck on the database now.

hughsie, 2022-11-24:
Issue opened https://jira.linuxfoundation.org/plugins/servlet/desk/portal/2/IT-24877 will report back when we know what happened. It's very odd!

hughsie, 2022-11-24:
Well, thankfully it's only one..

[image]

hughsie, 2022-11-25:
I've removed this file from the metadata in the meantime as all the sysadmin team is doing Thanksgiving and I don't have direct access to the EFS share.

SicLuceatLux, 2022-12-01:
Since #5282 is realted to this issue. 404 Error is gone. It is downloading properly but now i got this 
[image]

SicLuceatLux, 2022-12-05:
opend dedicated issue #5296

w5pny, 2022-12-05:
Dell finally uploaded this missing file(s)...

closed 2022-12-05 by w5pny