fwupd/fwupd #5281: 04F3:311C ELAN touchpad unusable after upgrade

dequis, 2022-11-24:
Laptop: Dell Precision 5560
Touchpad: `DLL0945:00 04F3:311C` via i2c_designware

```
runtime   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
compile   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    5.15.77-1-lts
compile   org.freedesktop.gusb          0.4.2
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.2
```

Hey there!

I think I bricked the touchpad of my laptop.

Title inspired by a very similar issue, https://github.com/fwupd/fwupd/issues/2926 (which is actually about both the touchpad and the fingerprint reader)

I was trying to fix the issue described [in this issue from the libinput tracker](https://gitlab.freedesktop.org/libinput/libinput/-/issues/618) using the steps described [in this comment](https://gitlab.freedesktop.org/libinput/libinput/-/issues/618#note_1379374) to extract and flash the firmware from the official updater, which i downloaded from [dell](https://www.dell.com/support/home/en-eu/drivers/driversdetails?driverid=4w96v&lwp=rt).

I realize this probably isn't a supported workflow since there's no ELAN touchpad firmware on LVFS but I figured I might have a chance if I ask for help here.

It no longer shows up in get-devices (which used to mention ELAN more explicitly) or dmesg, I got the IDs from logs of previous boots:

```
kernel: input: DLL0945:00 04F3:311C Mouse as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-DLL0945:00/0018:04F3:311C.0009/input/input51
kernel: input: DLL0945:00 04F3:311C Touchpad as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-DLL0945:00/0018:04F3:311C.0009/input/input52
kernel: hid-generic 0018:04F3:311C.0009: input,hidraw2: I2C HID v1.00 Mouse [DLL0945:00 04F3:311C] on i2c-DLL0945:00
```

Interestingly only the touchpad is missing in current boots, and the "mouse" is 04F3:0400 instead of the usual 04F3:311C (but it reports no input events) 

```
input: DLL0945:00 04F3:0400 as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-DLL0945:00/0018:04F3:0400.0004/input/input11
hid-generic 0018:04F3:0400.0004: input,hidraw2: I2C HID v1.00 Mouse [DLL0945:00 04F3:0400] on i2c-DLL0945:00
```

This seems to match [this comment in the other issue](https://github.com/fwupd/fwupd/issues/2926#issuecomment-784162419), I guess that's bootloader mode? 

I lost the output of the original update attempt because i rebooted in hopes that it'd fix it (it kinda looked like it got interrupted halfway through), but I got this:

[image]

Near the top is the install-blob command i used, and 4 seconds later at 16:09:34 there's a handful of i2c errors:

```
kernel: i2c_hid_acpi i2c-DLL0945:00: i2c_hid_get_input: incomplete report (14/60420)
kernel: i2c_hid_acpi i2c-DLL0945:00: i2c_hid_get_input: IRQ triggered but there's no data
kernel: i2c_designware i2c_designware.1: controller timed out
kernel: i2c_designware i2c_designware.1: timeout in disabling adapter
kernel: i2c_designware i2c_designware.1: timeout waiting for bus ready
kernel: i2c_hid_acpi i2c-DLL0945:00: failed to set a report to device.
```

Also, after this, i ran `fwupdmgr get-devices` (at 16:09:56, 20 seconds later) and got this:

```
systemd[1]: Starting Firmware update daemon...
fwupd[1294218]: 15:09:57.803 FuEngine             failed to probe device usb:03:00:0b: USB error on device 0c45:672e : Entity not found [-5]
kernel: i2c_designware i2c_designware.1: timeout in disabling adapter
fwupd[1294218]: 15:10:00.083 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-DLL0945:00/0018:04F3:311C.0009/hidraw/hidraw2: failed to read HID ID: ioctl error: Connection timed out [110]
kernel: i2c_designware i2c_designware.1: timeout waiting for bus ready
kernel: i2c_hid_acpi i2c-DLL0945:00: failed to set a report to device.
```

The USB error is irrelevant (that's the built-in webcam which is disabled), but there's this:

`failed to add device /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-14/i2c-DLL0945:00/0018:04F3:311C.0009/hidraw/hidraw2: failed to read HID ID: ioctl error: Connection timed out [110]`

Shortly after that I rebooted, and it didn't help.

As suggested in the other thread, I tried to add a quirks file, which didn't work, and i then edited the `builtin.quirk.gz`, also no luck.

```
$ zcat /usr/share/fwupd/quirks.d/builtin.quirk.gz | grep -B4 -A2 precision
# Acer Aspire V3-372T
[513cde3d-d939-59bd-a634-5c1645ebb93b]
Flags = elantp-recovery

# precision 5560
[97034138-3b7a-5e6d-8512-1b03980ce672]
Flags = elantp-recovery
```

[fwupdtool hwids](https://github.com/fwupd/fwupd/files/10086586/hwids.txt)

[fwupdtool get-devices --plugins elantp --verbose](https://github.com/fwupd/fwupd/files/10086583/get-devices-elantp.txt)

[FWUPD_PROBE_VERBOSE=1 fwupdtool get-devices --plugins elantp --verbose](https://github.com/fwupd/fwupd/files/10086659/fwupd_probe_verbose.txt)

Like the other issue said, the verbose output has this:

```
  SysfsPath:            /sys/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-15/i2c-DLL0945:00/0018:04F3:0400.0004/hidraw/hidraw2

17:47:58.493 FuPlugin             fu_plugin_device_created(elantp)
17:47:58.493 FuEngine             elantp ignoring: not i2c-hid touchpad
```

Next thing I'm gonna test is this https://github.com/fwupd/fwupd/issues/2926#issuecomment-785367823, which *seems* to result in fwupd seeing the device in the other thread. But first dinner.

dequis, 2022-11-24:
Okay, applied a variation on the patch on that comment (the product id is `0x400` in bootloader mode and the cutoff was `< 0x3000`, i picked `< 0x100` as an arbitrary lower-than-0x400 number. A cleaner thing would probably be checking for exactly `0x400`)

```diff
$ diff -Naur fu-elantp-hid-device.c*
--- fu-elantp-hid-device.c      2022-11-24 19:12:03.767213014 +0100
+++ fu-elantp-hid-device.c.orig 2022-11-24 19:14:28.796461756 +0100
@@ -64,7 +64,7 @@
        }

        /* i2c-hid */
-       if (fu_udev_device_get_model(FU_UDEV_DEVICE(device)) < 0x100 ||
+       if (fu_udev_device_get_model(FU_UDEV_DEVICE(device)) < 0x3000 ||
            fu_udev_device_get_model(FU_UDEV_DEVICE(device)) >= 0x4000) {
                g_set_error_literal(error,
                                    FWUPD_ERROR,
```

And now it is visible:

```
$ sudo fwupdtool get-devices --plugins elantp
Loading…                 [**************                         ]18:13:13.337 FuEngine             failed to probe device usb:03:00:0b: USB error on device 0c45:672e : Entity not found [-5]
Loading…                 [************************************** ]18:13:13.531 FuEngine             failed to update history database: device ID 8a21cacfb0a8d2b30c5ee9290eb71db021619f8b was not found
Loading…                 [-                                      ]
Dell Inc. Precision 5560
│
└─DLL0945:00 04F3:0400:
      Device ID:          0eeea9b449795ed307fcba6fa46f596d9679b87b
      Summary:            Touchpad
      Current version:    0x0000
      Bootloader Version: 0x0001
      Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
      GUIDs:              a27f0ead-f345-501e-ac88-9f2d6b999ad7 ← HIDRAW\VEN_04F3&DEV_0400
                          d13ad68b-20fd-5b2f-ad7c-4d2741f6bd1a ← HIDRAW\VEN_04F3&DEV_0400&REV_00
                          a3060df1-2988-562d-b461-b9dcd549b374 ← HIDRAW\VEN_04F3&DEV_0400&MOD_0000
                          4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
                          6cf7ae6a-8b98-5b63-9af8-b23b932f65de ← ELANTP\ICTYPE_13&MOD_0000
                          ac3627a9-8463-5e63-b6b1-be9e0bcf27d2 ← ELANTP\ICTYPE_13&MOD_0000&DRIVER_HID
      Device Flags:       • Internal device
                          • Updatable
                          • Is in bootloader mode
```

Yay.

```
$ sudo fwupdtool install-blob TP_Bin.bin 0eeea9b449795ed307fcba6fa46f596d9679b87b

Loading…                 [****                                   ]18:22:22.784 FuEngine             failed to probe device usb:03:00:0b: USB error on device 0c45:672e : Entity not found [-5]
Loading…                 [************                           ]18:22:25.312 FuEngine             failed to update history database: device ID 8a21cacfb0a8d2b30c5ee9290eb71db021619f8b was not found
Loading…                 [-                                      ]
Waiting…                 [***************************************] Less than one minute remaining…
```

```
Dell Inc. Precision 5560
│
└─DLL0945:00 04F3:0400:
      Device ID:          0eeea9b449795ed307fcba6fa46f596d9679b87b
      Summary:            Touchpad
      Current version:    0x000c
      Bootloader Version: 0x0001
      Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
      GUIDs:              a27f0ead-f345-501e-ac88-9f2d6b999ad7 ← HIDRAW\VEN_04F3&DEV_0400
                          d13ad68b-20fd-5b2f-ad7c-4d2741f6bd1a ← HIDRAW\VEN_04F3&DEV_0400&REV_00
                          a3060df1-2988-562d-b461-b9dcd549b374 ← HIDRAW\VEN_04F3&DEV_0400&MOD_0000
                          4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
                          6cf7ae6a-8b98-5b63-9af8-b23b932f65de ← ELANTP\ICTYPE_13&MOD_0000
                          ac3627a9-8463-5e63-b6b1-be9e0bcf27d2 ← ELANTP\ICTYPE_13&MOD_0000&DRIVER_HID
      Device Flags:       • Internal device
                          • Updatable

```

I might just have fixed the issue myself? Gotta post this comment before I reboot.

dequis, 2022-11-24:
Yup. Works.

So: the conclusions from the previous issue (#2926) were:

* Entering bootloader was caused by verify-update
    * Fixed by commits disallowing verify-update https://github.com/fwupd/fwupd/commit/32e4b2b8f0e07cff5cc21c3a21b923406bfcdbac
    * OP didn't have a valid firmware in the first place, what if that was the reason they ended up in bootloader mode?
* Exiting bootloader mode needs a valid firmware
    * OP didn't have a valid firmware blob, and ended up acquiring one

What was missing:
* 0x400 should have been added as a valid product ID for elan touchpads

What I still don't understand:
* I entered bootloader mode without verify-update and with a valid firmware. Why? Other people on the [gitlab thread](https://gitlab.freedesktop.org/libinput/libinput/-/issues/618) successfully applied the update with this method.
    * I could speculate that the fact that i was actively having the touchpad acceleration/sensitivity issues meant i2c wasn't working very well.

But yeah if 0x400 was accepted as i2c-hid in fu-elantp-hid-device.c, I wouldn't have needed to post this issue at all. Please change that!

dequis, 2022-11-24:
Oh, a small note: the elantp-recovery flag isn't relevant to get this working. When I installed the patched version it undid my changes to `/usr/share/fwupd/quirks.d/builtin.quirk.gz`, so all the steps from the first successful get-devices were done without that flag.

jinglewu, 2022-11-25:
PID 0400 is ELAN touchpad bootloader mode only.
Some kernel cannot detect this PID 0400 in the i2c-hid interface. So this works with elan-recovery mode in fwupd.
It should be possible to increase PID 0400 in fu-elantp-hid-device.c .

hughsie, 2022-11-25:
@dequis could you do a PR changing the `if` statement to have a lower bound of 0x400 please?

dequis, 2022-11-25:
@hughsie Gladly! https://github.com/fwupd/fwupd/pull/5283

apedicdev, 2024-05-03:
is this by any chance fixing the annoying elan touchpad issue on linux?

dequis, 2024-05-03:
No, this is just for when you get it stuck on bootloader mode while trying to upgrade the firmware while trying to fix it, which should be rare. The firmware upgrade, at least the one i applied here, didn't fix the issues I had, but didn't make it worse.

I have no idea if the issue still happens. I might have gotten used to it, or maybe unrelated environmental factors made my i2c bus less bad, or maybe someone fixed the issue. Couldn't tell you which.

porteusconf, 2025-03-07:
> is this by any chance fixing the annoying elan touchpad issue on linux?

I may try and let you know, but I've read that others tried firmware updates and nothing changed... that is, still had apparently unfixed [bug 209167](https://bugzilla.kernel.org/show_bug.cgi?id=209167) which describes dmesg spammed with:
```
Touchpad at isa0060/serio1/input0 lost sync at byte 6
```