fwupd/fwupd #5284: Unsupported CPU on AMD Thinkpad?

blitz, 2022-11-25:
**Describe the question**

I'm currently debugging why I get a `?` as security status in the GNOME Device Security page. I think the issue is the one below and I'm not sure whether this is a bug in fwupd or in the kernel or whether the CPU is actually unsupported. (If so, why?). I would appreciate if anyone can shed some light here. :)

My CPU is: AMD Ryzen 7 PRO 4750U with Radeon Graphics
The laptop in question is a Lenovo Thinkpad L14 Gen1 (AMD).

Verbose log of the below commands: [fwupd-verbose.log](https://github.com/fwupd/fwupd/files/10094238/fwupd.log)
Linux patch to get around `think-lmi` not providing `type` attributes: [0001-platform-x86-think-lmi-expose-type-attribute.patch.log](https://github.com/fwupd/fwupd/files/10094254/0001-platform-x86-think-lmi-expose-type-attribute.patch.log)

```shell
❯ sudo fwupdtool security                
Loading…                 [-                                      ]

Not enough data was provided to make an HSI calculation.
 » https://fwupd.github.io/hsi.html#not-enough-info
To ignore this warning, use --force
❯ sudo fwupdtool security --force
Loading…                 [-                                      ]
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ fwupd plugins:                 Untainted
✘ Linux kernel lockdown:         Failed
✘ Linux swap:                    Unencrypted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

```

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

```shell
❯ fwupdmgr --version 
runtime   org.freedesktop.fwupd         1.8.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    6.0.9
runtime   com.hughsie.libjcat           0.1.12
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.gusb          0.3.10
```

RaitoBezarius, 2022-11-28:
The patch in question addresses https://github.com/fwupd/fwupd/issues/5077 partially by making everything a string.

hughsie, 2022-11-28:
This is a dupe of https://github.com/fwupd/fwupd/issues/4887 unfortunately.

closed 2022-11-28 by hughsie

superm1, 2022-11-29:
> The patch in question addresses https://github.com/fwupd/fwupd/issues/5077 partially by making everything a string.

I think that patch needs some work. I suspect with it applied you don't get the warning but you also won't see possible values anymore in fwupd.

superm1, 2024-05-29:
Here is a [patch series](https://lore.kernel.org/linux-crypto/oCZE3H8ZnqlQsatOdb4HH7_0t7E9fsRqVv1QodA3IeUSDCdx9DP1bWsLOVq6T5iGffMjzMequirUJSM4UpcA1y2LZVUvMsffulaArsPy2cs=@hughsie.com/T/#t) for the kernel that enables the feature on the older SoCs.