fwupd/fwupd #5287: wacom-raw can fail to appear on HIDRAW

flying-elephant, 2022-11-29:
**Describe the bug**
When a system is installed with specific wacom hid driver -sources in the kernel is located at /drivers/hid/- , wacom-raw
devices will fail to be loaded with HIDRAW and seems disappearning afterwords.
That causees the whole flash processes of wacom-raw devices fail neithger.

**Steps to Reproduce**
#1. Compile your kernel with wacom hid driver located at "/drivers/hid/" -either built-in or a loadable module works
#2. Start LVFS upgrade to our devices on your system
#3. Once our device become what we call "boot" mode, then, the device disappears

**Expected behavior**
The flash can never be continued after wacom-raw devices turn into the boot mode

**fwupd version information**
Any

**Additional questions**
This happens with the vendor ID 0x56a of our devices; however, all our boot devices have that ID except that they are in the normal mode.
I'd like to know if any subclass to unload a loaded module and move a certain loadable module to somewhere during flash.
Then, the module is move back to the original place after the flash is done.
If there is a such operation, I'd like to modify wacom-raw in right way.

Tats

hughsie, 2022-11-29:
> unload a loaded module and move a certain loadable module to somewhere during flash

You can unload the kernel driver using `fu_device_unbind_driver` but you can't move the module .ko as the file system might be immutable. I've not got anything here with wacom-raw hardware to test with, but can you show me what's happening with a `fwupdtool install-blob --verbose` log perhaps?

flying-elephant, 2022-11-30:
Thank you, but the log was looking like this:
ubuntu@ubuntu-ThinkPad-X1-Yoga-Gen-7:~/tmp/fwupd_modification/fwupd$ sudo ./build/src/fwupdtool install-blob --verbose 
00:57:55:0029 FuDebug              Verbose debugging enabled (on console 1)
Loading…                 [-                                      ]Invalid arguments

Use fwupdtool --help for help
ubuntu@ubuntu-ThinkPad-X1-Yoga-Gen-7:~/tmp/fwupd_modification/fwupd$ 

Therefore, I tested my system with our device equipped with and with wacom HID driver.
It succeeded to flash firmware. I guess it is OK with I2C HID devices because they are not disconnected once it is loaded.
Up to this moment, we haven't had any request for LVFS update for our devices with USB.
Also., all of the listed device in the quirk are with I2C.

If our device with USB I/F, this failure will happen, but unfortunately I don't have a unit with such device either. 
I can prepare a device with USB I/F, but I have no idea if such testing-purpose device can be evaluated with LVFS.
Do you have any idea ?

hughsie, 2022-12-05:
> I can prepare a device with USB I/F

So this isn't wacom-usb protocol, but something wacom-raw over some kind of USB endpoint? I'm a bit confused what hardware this is trying to target; feel free to switch to email if it's sensitive.

flying-elephant, 2022-12-06:
>So this isn't wacom-usb protocol, but something wacom-raw over some kind of USB endpoint? 
the devices controled under "wacom-usb" are supposed to be our branded product solutions (BS), Bamboo, Intuos, Cintiq, etc[1]
On the other hand our company owns Component-business and those devices are sold only to either our OEMs / ODMs..
It is called "Technology solution(TS)"[2]

Products from those two different business are not managed by a single department group.
Therefore, the products -inside, outside, idea, firmware, etc- are different too in many component categories.
Here we have "wacom-raw" targettomg TS business products and those products can be eqiuppedwith either I2C or USB.
Does it make sense ? 

[1]:https://estore.wacom.com/en-US
[2]:https://www.wacom.com/en-us/about-wacom/businesses

hughsie, 2022-12-06:
>  Here we have "wacom-raw" targettomg TS business products and those products can be eqiuppedwith either I2C or USB.
> Does it make sense ?

It does, but I need to know a lot more about the USB interface before I can give you any advice about wacom-raw. The original design of FuWacomDevice literally subclasses FuUdevDevice. Can you share what device is expected to disconnect on FW_CMD_DETACH and what happens at each point?

flying-elephant, 2022-12-07:
>Can you share what device is expected to disconnect on FW_CMD_DETACH and what happens at each point?
Yes, ofcause.
When "Detach" is passed to our device, our device is going to be into "Boot" mode.
The descriptors are changed in this case, so the device is reconnected as another USB device.
Touch and pen devices are disconnected before it turns to the boot mode (boot device).

Once the boot device appears, wacom hid driver grabs this "Boot" device, but stops to register it as a HID device because it is neither touch or pen device.
The generic HID driver, on the other hand, doesn't pick this up neighther afterwords. 

Once the boot device is abondoned by both drivers, hidraw driver is not prepared for it. 
At the end, our flash program, wacom-raw, loses our boot device and cannot proceed to rewrite our firmware.

This whole thing happens only when our device has USB interface and the specific VID 0x56a.
This incident never happens to the devices with the combinations of :

I/F-USB and VID-0x2d1f
I/F-I2C and VID-0x2d1f
I/F-I2C and VID-0x56a

hughsie, 2022-12-08:
> Once the boot device is abondoned by both drivers, hidraw driver is not prepared for it.

Can you expand a bit; do you not get a /dev/hidrawX node added by the kernel? If there's no /dev/hidrawX then fwupd won't be able to talk to the device.

flying-elephant, 2022-12-08:
>do you not get a /dev/hidrawX node added by the kernel?

No. hidraw node is never created with our boot device in the case that I explained. That's what I was trying to say too!

We submitted a patch to avoid that : https://www.spinics.net/lists/linux-input/msg82524.html
However, we don't know how long it takes fo the patch to be accepted and merged into the kernel.
I can currently think of to avoid this until the patch is in is just to move wacom HID module once fwupd finds our device somewhere while trying to flash

Only good thing is that we are rarely asked to adopt USB interfaces for our products. I2C interface is mainly used.

hughsie, 2022-12-09:
> We submitted a patch to avoid that : https://www.spinics.net/lists/linux-input/msg82524.html

That looks entirely correct.

>  However, we don't know how long it takes fo the patch to be accepted and merged into the kernel.

If it helps, you can require a specific kernel version with fwupd updates.

> just to move wacom HID module

There are issues with this; the filesystem may be read-only, the module may be "built in" the the kernel image, and various anti-malware projects are going to start screaming if one process starts changing stuff with a .ko extension :)

We could add a blacklist entry at runtime, but that seems kinda fragile. Maybe less fragile if you need to do *something*.  Maybe have a look at /etc/modprobe.d/ as a solution?

superm1, 2022-12-09:
> Once the boot device appears, wacom hid driver grabs this "Boot" device, but stops to register it as a HID device because it is neither touch or pen device.
The generic HID driver, on the other hand, doesn't pick this up neighther, afterwords.

Could you perhaps unbind it from the wacom driver in the kernel using the sysfs interface at this time?

hughsie, 2022-12-09:
> Could you perhaps unbind it from the wacom driver in the kernel using the sysfs interface at this time

There's no device node at all at this point from my understanding.

superm1, 2022-12-09:
> > Could you perhaps unbind it from the wacom driver in the kernel using the sysfs interface at this time
> 
> There's no device node at all at this point from my understanding.

No devfs node sure; but there should be sysfs nodes.

hughsie, 2022-12-09:
@flying-elephant can you get a udev log of what happens when the device is in bootloader mode? e.g. all the kernel events.

flying-elephant, 2022-12-12:
>There are issues with this; the filesystem may be read-only, the module may be "built in" the the kernel image,

This is totally right. I need to keep it mind. Thanks!

>No devfs node sure; but there should be sysfs nodes.

hidraw driver seems like to create child classes and their devfs at the same time when a device is detected:
https://elixir.bootlin.com/linux/latest/source/drivers/hid/hidraw.c#L567

>can you get a udev log of what happens when the device is in bootloader mode? e.g. all the kernel events.

Hope the logs on the bottom hold enough information.
For this evaluation, I used a device with VID 0x56a and PID 0x5346; then I turned it to be the boot mode, which comes with
VID 0x56a and PID 0x0094.
No event looks happnening after "wacom" driver ignores the boot device.

*dmesg:
[   11.563824] rfkill: input handler disabled
[   14.799564] wlxcce1d539ebc4: authenticate with a8:9d:21:83:9c:1d
[   14.816671] wlxcce1d539ebc4: send auth to a8:9d:21:83:9c:1d (try 1/3)
[   14.817503] wlxcce1d539ebc4: authenticated
[   14.820040] wlxcce1d539ebc4: associate with a8:9d:21:83:9c:1d (try 1/3)
[   14.821423] wlxcce1d539ebc4: RX AssocResp from a8:9d:21:83:9c:1d (capab=0x11 status=0 aid=17)
[   14.825213] wlxcce1d539ebc4: associated
[   14.842020] IPv6: ADDRCONF(NETDEV_CHANGE): wlxcce1d539ebc4: link becomes ready
[   14.901654] wlxcce1d539ebc4: Limiting TX power to 13 dBm as advertised by a8:9d:21:83:9c:1d
[   24.745441] rfkill: input handler enabled
[   24.824025] snd_hda_codec_hdmi hdaudioC0D0: HDMI: invalid ELD data byte 3
[   26.648774] rfkill: input handler disabled
[   44.430773] EXT4-fs (sdb): mounted filesystem with ordered data mode. Opts: (null)
[   67.484334] usb 5-4: new full-speed USB device number 2 using xhci_hcd
[   67.660875] usb 5-4: New USB device found, idVendor=056a, idProduct=5346, bcdDevice=80.02
[   67.660878] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   67.660880] usb 5-4: Product: Pen and multitouch sensor
[   67.660882] usb 5-4: Manufacturer: Wacom Co.,Ltd.
[   67.699188] input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0005/input/input31
[   67.699324] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0005/input/input32
[   67.699391] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0005/input/input33
[   67.699478] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0005/input/input34
[   67.699565] input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0005/input/input35
[   67.699720] hid-generic 0003:056A:5346.0005: input,hiddev1,hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:2a:00.3-4/input0
[   82.732956] usb 5-4: USB disconnect, device number 2
[   87.792835] usb 5-4: new full-speed USB device number 3 using xhci_hcd
[   87.968871] usb 5-4: New USB device found, idVendor=056a, idProduct=5346, bcdDevice=80.02
[   87.968874] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   87.968876] usb 5-4: Product: Pen and multitouch sensor
[   87.968877] usb 5-4: Manufacturer: Wacom Co.,Ltd.
[   88.006884] input: Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0006/input/input36
[   88.068624] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0006/input/input37
[   88.068717] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0006/input/input38
[   88.068845] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0006/input/input39
[   88.068987] input: Wacom Co.,Ltd. Pen and multitouch sensor Mouse as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0006/input/input40
[   88.069213] hid-generic 0003:056A:5346.0006: input,hiddev1,hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:2a:00.3-4/input0
[  116.269081] usb 5-4: USB disconnect, device number 3
[  163.984942] wacom: disagrees about version of symbol module_layout
[  291.256076] usb 5-4: new full-speed USB device number 4 using xhci_hcd
[  291.433388] usb 5-4: New USB device found, idVendor=056a, idProduct=5346, bcdDevice=80.02
[  291.433391] usb 5-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  291.433393] usb 5-4: Product: Pen and multitouch sensor
[  291.433395] usb 5-4: Manufacturer: Wacom Co.,Ltd.
[  291.473612] input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0007/input/input41
[  291.473796] input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:5346.0007/input/input42
[  291.532385] wacom 0003:056A:5346.0007: hidraw4: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:2a:00.3-4/input0
[  321.412030] usb 5-4: USB disconnect, device number 4
[... cut ...]
UDEV  [340.025544] remove   /kernel/slab/kmalloc-512/cgroup/kmalloc-512(1307:systemd-timedated.service) (cgroup)
UDEV  [340.025710] remove   /kernel/slab/kmalloc-32/cgroup/kmalloc-32(1307:systemd-timedated.service) (cgroup)
UDEV  [340.026192] remove   /kernel/slab/kmalloc-4k/cgroup/kmalloc-4k(1307:systemd-timedated.service) (cgroup)
UDEV  [340.026385] remove   /kernel/slab/anon_vma/cgroup/anon_vma(1307:systemd-timedated.service) (cgroup)
UDEV  [340.026537] remove   /kernel/slab/kmalloc-64/cgroup/kmalloc-64(1307:systemd-timedated.service) (cgroup)
^C
tats@Timothy:~$

hughsie, 2022-12-13:
>  UDEV [322.262883] add /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:0094.0008 (hid)

This looks more promising; you get a hid node, just not hidraw. What does `ls /sys/devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:0094.0008` show out of interest?

flying-elephant, 2022-12-13:
>This looks more promising; you get a hid node, just not hidraw. What does ls >/sys/devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb5/5-4/5-4:1.0/0003:056A:0094.0008 show out of interest?

Here's the logs. It looks like somehow the device was recognized as a HID device.
Can fwupd treat this information useful?

$tats@Timothy:/media/tats/Extended_1TB/3_Programs/Linux-app/hid_feature$ ls -l /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-5/1-5:1.0/0003:056A:0094.0008
total 0
-r--r--r-- 1 root root 4096 Dec 14 07:33 modalias
drwxr-xr-x 2 root root    0 Dec 14 07:33 power
-r--r--r-- 1 root root 4096 Dec 14 07:33 report_descriptor
lrwxrwxrwx 1 root root    0 Dec 14 07:32 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Dec 14 07:32 uevent

$tats@Timothy:/media/tats/Extended_1TB/3_Programs/Linux-app/hid_feature$ cat /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-5/1-5:1.0/0003:056A:0094.0008/uevent
HID_ID=0003:0000056A:00000094
HID_NAME=Wacom Co.,Ltd. Multitouch sensor
HID_PHYS=usb-0000:03:00.0-5/input0
HID_UNIQ=
MODALIAS=hid:b0003g0101v0000056Ap00000094

hughsie, 2022-12-14:
>  Can fwupd treat this information useful?

Hmm, that interface looks about as useful as a _chocolate teapot_. I think we do need that kernel patch to do this properly.

flying-elephant, 2022-12-15:
It sounds like what we can do is only pray not to have our devices USB I/F :(
Then, I'll keep this mind and ask a customer to have VID 0x2D1F definately if their product is operated by USB I/F.
Or do you have any other idea? ; perhaps, we could actively patch wacom.ko ?

superm1, 2022-12-15:
Land the patch and CC it to stable and actively push it to land everywhere you can. Ubuntu, chrome, fedora, debian etc.

flying-elephant, 2022-12-15:
>push it to land everywhere you can. Ubuntu, chrome, fedora, debian etc.

Would you like to explain it with more detail?
It sounds like we need to send the patch to as many Linux distribution community as possible, but don't know exactly how.
Maybe, could you indicate any reference that leads to what to do?

hughsie, 2022-12-15:
> push it to land everywhere you can

I think getting it into the upstream linus tree and then *waiting* would be what I'd do. Fedora/RHEL is hard to take patches, but the former we rebase pretty often. Another thought. What happens if you use a _different VID_ (non-Wacom) for the botloader mode. Is that a viable option, or is the hardware already in the field, so to speak?

flying-elephant, 2022-12-16:
>Another thought. What happens if you use a different VID (non-Wacom) for the botloader mode. Is that a viable option, or is the
>hardware already in the field, so to speak?

Unfortunately, it is not possible. Changing the boot loader may cost a lot not only expences, but also human resources, time etc.
What about using USB commuication components, like wacom-usb for branded products, once we find a USB device with
VID 0x56a and PID 0x0094 ?

hughsie, 2022-12-16:
> What about using USB commuication components

The wacom-usb plugin is using FuHidDevice rather than using hidraw -- maybe that might work? It's certainly worth a try.

hughsie, 2022-12-16:
Does https://github.com/fwupd/fwupd/blob/main/plugins/wacom-usb/fu-wac-device.c help? If you can share a protocol specification maybe I can give you some sample code.

flying-elephant, 2022-12-19:
>Does https://github.com/fwupd/fwupd/blob/main/plugins/wacom-usb/fu-wac-device.c help? 

Thank you, I'll take a look at it and compare the components between that of wacom-raw and wacom-usb.

>If you can share a protocol specification maybe I can give you some sample code.

The protocol is the exactly the same suc has using feature get/set reports, the same report ID's and following commands if available, etc. Only the change is the interface from I2C to USB,

I'm getting busier, but I will definately try to implement the USB interfaced version protocols for the boot loader.

hughsie, 2022-12-19:
> Only the change is the interface from I2C to USB

I wonder if we can do something better than just re-implement a FuDevice object that inherits from FuHidDevice rather than FuUdevDevice... i.e. maybe we could make the HID an *interface* rather than a concrete object. Will ponder. In the meantime, the duplication would be 100% fine.

flying-elephant, 2022-12-20:
Thanks, I will work on it :)
Once I thought to use of "uhid" to forcely create "hidraw" device file for our boot loader, but didn't come up an idea.

hughsie, 2023-02-19:
Closing, feel free to reopen with the PR.

closed 2023-02-19 by hughsie