fwupd/fwupd #5290: German translation about failed upgrade show same, non-working URL three times https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot

tyll, 2022-11-29:
**Describe the bug**
German translation about failed upgrade show same, non-working URL three times:

"""
Der Aktualisierungsfehler ist ein bekanntes Problem, besuchen Sie diese URL für weitere Informationen: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot; https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot; https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
Bereit …                 [-                                      ]
"""

**Steps to Reproduce**
Have a failed upgrade, allow to upload a report of the update error "Aktualisierungsfehler"

**Expected behavior**
URLs that are working and only one of them.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
runtime   com.hughsie.libjcat           0.1.12
compile   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    6.0.8-300.fc37.x86_64
compile   org.freedesktop.gusb          0.4.1
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.2

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

dnf on Fedora 37

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
LANG=C fwupdmgr get-devices --show-all-devices
LENOVO 20TJS2F41T
?
??Unknown Device:
?     Device ID:          4a4907dbb1b96c6a1177dfd1b95eb41c781d1265
?     GUID:               1199a818-4c52-5137-b536-d59e2e2cada9 ? GPIO\ID_INT3450:00
?   
??CometLake-H GT2 [UHD Graphics]:
?     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
?     Current version:    05
?     Vendor:             Intel Corporation (PCI:0x8086)
?     GUIDs:              3777783a-3f83-56a5-95f4-533eb6a2bd19 ? PCI\VEN_8086&DEV_9BC4
?                         6c3dbf6c-4e6f-5309-9954-c5ab7aca617e ? PCI\VEN_8086&DEV_9BC4&REV_05
?                         5fde5d20-db24-5f21-afdd-247c1bf1efa1 ? PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2
?                         07168636-0f3b-565c-8fe1-0f0a77d82cd8 ? PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2&REV_05
?                         7ffe1cb7-395a-52a9-a172-70ec6caaf310 ? PCI\VEN_8086&DEV_9BC4&REV_00
?                         b813dc18-ddf2-508d-a7eb-0e2fc8752b03 ? PCI\VEN_8086&DEV_9BC4&SUBSYS_17AA22C2&REV_00
?     Device Flags:       ? Internal device
?                         ? Cryptographic hash verification is available
?   
??Core? i7-10850H CPU @ 2.70GHz:
?     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
?     Current version:    0x000000f0
?     Vendor:             Intel
?     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ? cpu
?                         30249f37-d140-5d3e-9319-186b1bd5cac3 ? CPUID\PRO_0&FAM_06
?                         a45b0522-5722-54bd-b802-86cd044262df ? CPUID\PRO_0&FAM_06&MOD_A5
?                         7b9b6e8c-226c-5db6-86cb-ea3187578013 ? CPUID\PRO_0&FAM_06&MOD_A5&STP_2
[... 303 lines cut ...]
?                         ? Device is usable for the duration of the update
?   
??UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    22552
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               6255bb23-b2a8-469a-9161-bb9e1348ca23
      Device Flags:       ? Internal device
                          ? Updatable
                          ? System requires external power source
                          ? Needs a reboot after installation
                          ? Device is usable for the duration of the update

```

</details>

**Additional questions**

- Operating system and version: Fedora 37
- Have you tried rebooting? since it is an error in the translation, I don't see that this would help.
- Is this a regression? I don't know.

hughsie, 2022-11-30:
How do you mean "non-working URL"? It seems to work here. I can fix the duplication of course.

hughsie, 2022-11-30:
I've fixed half the problem with https://gitlab.com/fwupd/lvfs-website/-/commit/0b803b3490772476120523f1c9d9daf167d4299a

tyll, 2022-12-02:
Thank you for the quick partial fix. My terminal URL parser identifies the semicolon as part of the URL, so when I click on it, it opens:

```
https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot;
```

instead of the intended URL:

```
https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot
```

So maybe use only space chars (maybe two or more) to separate the URLs instead of `; ` since semicolons are valid characters in URIs:

https://www.rfc-editor.org/rfc/rfc3986#section-2.2

hughsie, 2022-12-02:
Fixed in:

	commit 77ee3eac4c1f2fca7d8a3d7f8501a28b72528552 (HEAD -> master)
	Author: Richard Hughes <richard@hughsie.com>
	Date:   Fri Dec 2 18:20:32 2022 +0000

	    Do not use ; to split URIs, it's a valid char in RFC3986

I'll push on next deployment, thanks.

closed 2022-12-02 by hughsie