fwupd/fwupd #5294: IPv6 for fwupd.org

jpds, 2022-11-30:
**Describe the solution you'd like**

Please enable IPv6 for fwupd.org and related domains. This is hosted on AWS which fully supports this:

- https://aws.amazon.com/vpc/ipv6/

hughsie, 2022-11-30:
Sounds simple enough; the LVFS is hosted on AWS and provisioned using Terraform. So I can ask the LVFS sysadmin team for the right thing, do you happen to know what I should ask for? And also a way to test this please also.

jpds, 2022-11-30:
The steps off the top of my head for enabling v6 are:

- on the VPC settings, Amazon autoassigns a range once enabled
- on the subnets, you have to select an octet for the subnet from within the range
- on the instances - these are autoassigned by AWS from the above subnet
- verify that the security groups allow TCP/80, and TCP/443 (+an other required ports) from `::/0` (global internet)
- if ELBs are used, those have to be set to `dualstack`

You should then be able to test with:

```
curl -v -H "Host: fwupd.org" https://[insert:long:ipv6:address:of:ec2:instance:here]/
```

And then the AAAA records need to be added to DNS.

hughsie, 2022-11-30:
Created as https://jira.linuxfoundation.org/plugins/servlet/desk/portal/2/IT-24896 -- this will need one of the Linux Foundation sysadmin team to make this change. I'll close this ticket, but keep it updated with progress. Thanks for all the extra info, I pasted that into the LF support ticket. It's much appreciated.

closed 2022-11-30 by hughsie

jpds, 2022-12-15:
> qa.fwupd.org

Just noticed that this uses `cdn-staging.fwupd.org` which points at the fastly domain. Apparently adding `dualstack` to the CNAME Just Works: https://docs.fastly.com/en/guides/ipv6-support#switching-to-dualstack-for-fastly-shared-hostnames

```
$ host p2.shared.global.fastly.net.
p2.shared.global.fastly.net has address 151.101.38.49
$ host dualstack.p2.shared.global.fastly.net.
dualstack.p2.shared.global.fastly.net has address 151.101.38.49
dualstack.p2.shared.global.fastly.net has IPv6 address 2a04:4e42:9::561
```

hughsie, 2022-12-15:
Is that an extra thing we need to do?

jpds, 2022-12-15:
Yes, as it's separate to AWS - however it'll require zero testing as it's just all transparently handled by Fastly once the DNS configuration bit is done.

So you could probably also enable it on `cdn.fwupd.org`. Might also be worth auditing other DNS records for other aubdomains (I obviously do not have access to this).

hughsie, 2022-12-15:
Proxied to Ryan, thanks.

hughsie, 2023-01-20:
Ryan delivered:

    $ host cdn.fwupd.org
    cdn.fwupd.org is an alias for dualstack.p2.shared.global.fastly.net.
    dualstack.p2.shared.global.fastly.net has address 199.232.54.49
    dualstack.p2.shared.global.fastly.net has IPv6 address 2a04:4e42:4a::561

@jpds can you confirm it's working for you now please?

jpds, 2023-01-20:
@hughsie That indeed Just Worked. Excellent. :tada:

Maybe the redirector at `www.fwupd.org` should be enabled but other than that - everything else is working great - thank you both.

jpds, 2023-01-27:
`www.fwupd.org` has a separate v4 address than the ones on `fwupd.org` - I do not know if this means it's on a separate VPC on Amazon (as it's a simple redirector) - however it might just be the case of adding the `AAAA` record for the instance it's on if it's on the same one.

I ran a few IPv6-only networks which occasionally trip up on things like this where some bits are on v4 and others are on v6. Hence why I'm mindful of little details like this.

hughsie, 2023-01-30:
@jpds can you phase that as a todo action please? e.g. so I'm opening another ticket with the sysadmins it's crystal clear.