fwupd/fwupd #5296: Dell 7650 1.16.0 -> 1.17.1  -- error  no HWIDs matched

SicLuceatLux, 2022-12-05:
**Describe the bug**
fwupdmgr informs me about new system-frimware/bios update for my Dell 7560.
1.16.0 -> 1.17.1
After agreeing to perfom an update this error meassage appears
[image]

**Steps to Reproduce**
sudo fwupdmgr update

**Expected behavior**
It should reboot and flash the latest bios

**fwupd version information**
fwupdmgr 1.8.0

```shell
fwupdmgr --version

runtime   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.0.11
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.0
runtime   org.freedesktop.gusb          0.3.10

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
sudo apt-get install fwupdmgr

<details>

<summary>**fwupd device information**</summary>

```shell
Precision 7560
│
├─WD19TB:
│ │   Device ID:          570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.03
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      55XM363/3193205146923953
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        073624c16dd99abe01ba1da223a70321e4f29beb
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.25.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUIDs:            d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        0acceea54e71c4d8002593822afe0f4705616613
[... 485 lines cut ...]
├── vmlinuz-6.0.6
├── vmlinuz-6.0.6-76060006-generic
├── vmlinuz-6.0.7
├── vmlinuz-6.0.8
├── vmlinuz-6.0.9
└── vmlinuz.old -> vmlinuz-6.0.7

1 directory, 53 files

```

**Additional questions**

- Operating system and version:
PopOS 22.04 latest
- Have you tried rebooting?
sure multiple times
- Is this a regression?
Yes
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No

hughsie, 2022-12-05:
If you do "fwupdmgr refresh --force" and then try to update again, does the same update get offered?

SicLuceatLux, 2022-12-05:
yep same

skipperTux, 2022-12-05:
Same error, Dell Precision 7560, Fedora 37 with all updates

```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.7
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.1
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.2
runtime   org.kernel                    6.0.11-300.fc37.x86_64
runtime   org.freedesktop.fwupd         1.8.7
```

```
$ fwupdmgr refresh --force
$ fwupdmgr get-devices 
$ fwupdmgr update
Downloading…             [***************************************]
no HWIDs matched 34335281-6f25-5a0d-9eca-4f709cb8fb87
```

skymoo, 2022-12-06:
Same with Fedora 36. Force refreshing doesn't help either.

SicLuceatLux, 2022-12-06:
Is it possible that the firmware cap contains a wrong hardwareid?  Or only the one for 7760?

CragW, 2022-12-06:
@skipperTux the metadata for 1.17.1 has been updated, pl. refresh and check if issue gets resolved.

skymoo, 2022-12-06:
That did the trick for me, thanks!