fwupd/fwupd #5303: [duke] mbb device fw: EM05GFAR07A07M1G_01.012.01.012 but  Can't downgrade EM05GFAR07A07M1G_01.013.01.013

dukexinaw, 2022-12-08:
**Describe the question**
1 mbb device fw: EM05GFAR07A07M1G_01.012.01.012

2 fwupdmgr   get-reaeases 
    list :  EM05GFAR07A07M1G_01.013.01.013
    list :  EM05GFAR07A07M1G_01.012.01.012

3 fwupdmgr    downgrade 
   have no  EM05GFAR07A07M1G_01.013.01.013    <---------- extect 

4 enable fwupd -v debug mode , do [fwupdmgr    downgrade ]
....
11:12:19.253 FuEngine             ignoring EM05GFAR07A07M1G_01.013.01.013 as newer than EM05GFAR07A07M1G_01.012.01.012
.....

We expect EM05GFAR07A07M1G_01.012.01.012 to upgrade EM05GFAR07A07M1G_01.013.01.013  , but I also check that fwupd code seems to be designed this way; How to solve it?
  

fwupdmgr --version
1.8.8

hughsie, 2022-12-08:
Don't you want `fwupdmgr upgrade` if you want to go from old to new?

dukexinaw, 2022-12-09:
Don't you want fwupdmgr upgrade if you want to go from old to new?
>>  yes ，do [fwupdmgr upgrade]  can update EM05GFAR07A07M1G_01.012.01.012 to EM05GFAR07A07M1G_01.013.01.013

so ，Do I understand it this way:
fwupdmgr upgrade  old to newest
right ?  There are related documents 

bug fwupdmgr downgrade  can not list EM05GFAR07A07M1G_01.013.01.013  May I ask the purpose of this design。

hughsie, 2022-12-09:
The basics is that `fwupdmgr upgrade` goes from old->new, and `fwupdmgr downgrade` goes from new->old, and `fwupdmgr reinstall` just reinstalls old->old.

dukexinaw, 2022-12-09:
Got it

Thanks  a lot。