fwupd/fwupd #5314: Thunderbolt being disabled causes IOMMU device protection alert (but only needed fot thunderbolt?)

BryanQuigley, 2022-12-12:
**Describe the bug**

2022-12-12 06:52:59:  ✔ IOMMU device protection enabled - Thunderbolt is enabled in BIOS.
2022-11-16 02:40:06:  ✘ IOMMU device protection disabled - Thunderbolt is disabled in BIOS.

This seems to encourage me to enable Thunderbolt to get IOMMU protection even though I don't think it's needed without Thunderbolt.

**Steps to Reproduce**

1. Set Thunderbolt to disabled in BIOS
2. Note warning that IOMMU device protection is disabled.

**Expected behavior**

IOMMU device protection should only show up if it would provide some protection.  Disabling unnecessary capabilities should generally be considered more secure.

Obviously there are other possibilities:
 * The BIOS ties together IOMMU and Thunderbolt even though IOMMU protection does help with other threats.
 * The kernel ties together IOMMU and Thunderbolt even though IOMMU protection does help with other threats.
 * Thunderbolt still poses a DMA threat even when disabled.
 * I'm mistaken in some of my assumptions

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
runtime   com.hughsie.libjcat           0.1.12
runtime   org.kernel                    6.0.12-300.fc37.x86_64
compile   com.hughsie.libjcat           0.1.12
compile   org.freedesktop.gusb          0.4.1
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Fedora default / dnf.

<summary>**fwupd device information**</summary>

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20RYCTO1WW
│
├─USB2.0 Hub:
│     Device ID:          301046452a49d84af6356d23e43a684b8f10660f
│     Summary:            USB 3.x hub
│     Current version:    4.124
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              fd36ff26-79e6-51bc-a9f2-558b7fe9ee34 ← USB\VID_17EF&PID_7229
│                         f7f83e6f-5ff1-54ae-8884-08d0686e1745 ← USB\VID_17EF&PID_7229&REV_04C4
│                         ac2e3a7b-7327-5e3e-aed4-c330647eb902 ← USB\VID_17EF&PID_7229&HUB_28
│                         d0233816-614f-5d85-ac93-ee9b299262ac ← USB\VID_17EF&PID_7229&SPI_6840
│                         e28cf82f-c61d-58dc-8f86-779900872f7c ← USB\VID_17EF&PID_7229&SPI_6840&REV_04C4
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│                         • Unsigned Payload
│   
├─Unknown Device:
│     Device ID:          59a11bc1795b6b83caa7eface68d0ad302cf5f86
│     GUID:               0422c218-5b8a-504d-8fba-43f8a29f4312 ← GPIO\ID_INT34BB:00
│   
├─CometLake-U GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
[... 251 lines cut ...]
DMAR: Intel(R) Virtualization Technology for Directed I/O
PCI-DMA: Using software bounce buffering for IO (SWIOTLB)
software IO TLB: mapped [mem 0x0000000079912000-0x000000007d912000] (64MB)
Initialise system trusted keyrings
....
ACPI: bus type thunderbolt registered
thunderbolt 0000:05:00.0: enabling device (0000 -> 0002)
```

Disabled Thunderbolt in BIOS

```
DMAR: Host address width 39
DMAR: DRHD base: 0x000000fed90000 flags: 0x0
DMAR: dmar0: reg_base_addr fed90000 ver 1:0 cap 1c0000c40660462 ecap 19e2ff0505e
DMAR: DRHD base: 0x000000fed91000 flags: 0x1
DMAR: dmar1: reg_base_addr fed91000 ver 1:0 cap d2008c40660462 ecap f050da
DMAR: RMRR base: 0x00000096e50000 end: 0x00000096e6ffff
DMAR: RMRR base: 0x0000009b000000 end: 0x0000009f7fffff
DMAR-IR: IOAPIC id 2 under DRHD base  0xfed91000 IOMMU 1
DMAR-IR: HPET id 0 under DRHD base 0xfed91000
DMAR-IR: Queued invalidation will be enabled to support x2apic and Intr-remapping.
DMAR-IR: Enabled IRQ remapping in x2apic mode
x2apic enabled
```

hughsie, 2022-12-12:
> This seems to encourage me to enable Thunderbolt to get IOMMU protection even though I don't think it's needed without Thunderbolt.

So enabling ThunderBolt in the BIOS also enables IOMMU? i.e. it's one "switch" for two things?

I guess the biggest threat we're trying to mitigate is a plugging in a malicious TBT3 peripheral that enables PCI-passthrough -- which without IOMMU gives it access to everything on the system. @superm1 do you know anything else (do we care about FireWire at this point?!) that allows the same kind of tunnelling? Do you think it makes sense to "obsolete" the requirement of IOMMU if ThunderBolt/USB4 is disabled? And on the last point, how can we know if TBT is actually disabled?

superm1, 2022-12-12:
> So enabling ThunderBolt in the BIOS also enables IOMMU? i.e. it's one "switch" for two things?

Wait, where does this string come from?  I have never seen `Thunderbolt is enabled in BIOS` or `Thunderbolt is disabled in BIOS`.

>  @superm1 do you know anything else (do we care about FireWire at this point?!) that allows the same kind of tunnelling? Do you think it makes sense to "obsolete" the requirement of IOMMU if ThunderBolt/USB4 is disabled?

USB4/Thunderbolt is the big one but there are other attack surfaces.  If the laptop has a relatively easily accessible M.2 port then you can take the chassis cover off and attack from that too.  We don't know a lot about the hardware design in fwupd.

Furthermore turning on the IOMMU will segment the memory access between PCIe devices that are internal to the system.  My WLAN won't be able to access the memory regions used by my NVME for example.

BryanQuigley, 2022-12-12:
Here is where I disable Thunderbolt in BIOS.  [image]

I'm not sure if the BIOS or kernel decides to not enable iommu.

superm1, 2022-12-12:
So it's not the fwupd error string. That was your amendment to the error message.

BryanQuigley, 2022-12-12:
Oh, sorry yes, I just added that for context.  fwupdmgr is not literally saying go enable thunderbolt to turn on IOMMU.

superm1, 2022-12-12:
> Oh, sorry yes, I just added that for context. fwupdmgr is not literally saying go enable thunderbolt to turn on IOMMU.

OK, that makes more sense to me now then.   Would you might sharing the output of `fwupdmgr get-bios-settings` for your system?

fwupd didn't suggest you to turn on Thunderbolt, I think that's actually our bug here.  Which I admit is the opposite of what you want :)

BryanQuigley, 2022-12-12:
I also diffed the output between having Thunderbolt enabled and disabled and the ThunderboltAccess item was the only change.

fwupdmgr get-bios-settings 
```
SecureBoot:
  Setting type:         Enumeration
  Current Value:        Enable
  Description:          SecureBoot
  Read Only:            True
  Possible Values:
    0:                  Disable
    1:                  Enable

LockBIOSSetting:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          LockBIOSSetting
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

WindowsUEFIFirmwareUpdate:
  Setting type:         Enumeration
  Current Value:        Enable
  Description:          BIOS updates delivered via LVFS or Windows Update
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

BluetoothAccess:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          BluetoothAccess
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

FingerprintReaderAccess:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          FingerprintReaderAccess
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

WakeOnLANDock:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          WakeOnLANDock
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

BIOSPasswordAtUnattendedBoot:
  Setting type:         Enumeration
  Current Value:        Enable
  Description:          BIOSPasswordAtUnattendedBoot
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

PasswordBeep:
  Setting type:         Enumeration
  Current Value:        Disable
[... 769 lines cut ...]
    1:                  Enable

BootTimeExtension:
  Setting type:         Enumeration
  Current Value:        Disable
  Description:          BootTimeExtension
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  1
    2:                  2
    3:                  3
    4:                  5
    5:                  10

EnhancedWindowsBiometricSecurity:
  Setting type:         Enumeration
  Current Value:        Enable
  Description:          EnhancedWindowsBiometricSecurity
  Read Only:            False
  Possible Values:
    0:                  Disable
    1:                  Enable

```

commit 3cabf5a7 (2022-12-14): iommu: Suggest to turn on ThunderboltAccess for Lenovo systems

Lenovo systems that offer a BIOS setting for ThunderboltAccess will
use this option to control whether the IOMMU is enabled by default
or not.

It may be counter-intuitive; but as there are other more physically
difficult to attack PCIe devices it's better to have the IOMMU
enabled pre-boot even if it enables access to Thunderbolt/USB4.

Fixes: #5314
commit 9e22e9c1 (2022-12-14): iommu: Suggest to turn on ThunderboltAccess for Lenovo systems

Lenovo systems that offer a BIOS setting for ThunderboltAccess will
use this option to control whether the IOMMU is enabled by default
or not.

It may be counter-intuitive; but as there are other more physically
difficult to attack PCIe devices it's better to have the IOMMU
enabled pre-boot even if it enables access to Thunderbolt/USB4.

Fixes: #5314
closed 2022-12-14 by superm1