fwupd/fwupd #5344: TPM PCR0 reconstruction failed on ThinkPad Z13

hexchain, 2022-12-17:
**Describe the bug**
On a ThinkPad Z13 system, `fwupdmgr security` reports that TPM PCR0 reconstruction is invalid:

```
% fwupdmgr security
Host Security ID: HSI:1! (v1.8.8)

HSI-1
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ SPI write protection:          Enabled
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ SPI replay protection:         Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Encrypted RAM:                 Encrypted
✔ Processor rollback protection: Enabled

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
✘ Linux kernel lockdown:         Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

The TPM PCR0 differs from reconstruction.
 » https://fwupd.github.io/hsi.html#pcr0-tpm-event-log-reconstruction

Host Security Events
  2022-12-07 13:47:12:  ✔ Encrypted RAM changed: Not supported → Encrypted
```

**fwupd version information**
Please provide the version of the daemon and client.

```shell
% fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.8
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.2
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.2
runtime   org.freedesktop.fwupd         1.8.8
runtime   org.kernel                    6.1.0-arch1-1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman (Arch Linux)

<details>

<summary> **fwupd device information** </summary>

Please provide the output of the fwupd devices recognized in your system.
[... 281 lines cut ...]
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│
├─USB4 host controller:
│     Device ID:          518a601a1d260069142204e3a8efc3c8232268d4
│     Summary:            Unmatched performance for high-speed I/O
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│
└─USB4 host controller:
      Device ID:          e68fb7ebd73ca51bfbb1b75867e9f9babf3636be
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
```

</details>

**Additional questions**
See https://gist.github.com/hexchain/39de779d7a189bb94b782542ebf93a93 for the output of `fwupdtpmevlog` and `tpm2_pcrread`.

superm1, 2022-12-17:
Hmm interesting.  I have a Z13 here but don't reproduce that.  I also don't see sha384 reported though, even with a tpm2-tools git snapshot from today.
Your system actually reports sha384 on the TPM, but we don't calculate that for the event log.  Can you please try PR?  https://github.com/fwupd/fwupd/pull/5345

I don't _think_ it will fix the underlying issue, but I would like if you could confirm it and give a new `fwupdtpmevlog` output.

superm1, 2022-12-17:
Also; please can I see output for `fwupdtool security --verbose`

hexchain, 2022-12-17:
That PR seems to have been merged. I built the latest main branch but `./plugins/tpm/fwupdtpmevlog` didn't show any `ChecksumSha384` entries.

fwupdtool output: 
[fwupdtool.stderr.log](https://github.com/fwupd/fwupd/files/10251886/fwupdtool.stderr.log)
[fwupdtool.stdout.log](https://github.com/fwupd/fwupd/files/10251887/fwupdtool.stdout.log)

superm1, 2022-12-17:
```
14:12:52.480 FuPluginTpm          comparing TPM 7d3d291a5428a0059b8759c3e185b6fa752eec55da253b3f0c0760e9de42c9d4 and EVT 7d3d291a5428a0059b8759c3e185b6fa752eec55da253b3f0c0760e9de42c9d4
```

It looks like the string comparison passed...

superm1, 2022-12-17:
Have a try with this one: https://github.com/fwupd/fwupd/pull/5347

hexchain, 2022-12-18:
Now it says "Valid"!

[fwupdtool.stderr.log](https://github.com/fwupd/fwupd/files/10253449/fwupdtool.stderr.log)
[fwupdtool.stdout.log](https://github.com/fwupd/fwupd/files/10253450/fwupdtool.stdout.log)

hughsie, 2022-12-18:
Thanks @superm1 !

closed 2022-12-18 by hughsie