fwupd/fwupd #5346: No volumes of type `c12a7328-f81f-11d2-ba4b-00a0c93ec93b` (ie EFI System) with EFI in RAID1 soft volume

flaf, 2022-12-17:
**Describe the bug**

The command `fwupdmgr update` fails with the message `no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b` because, in my case, EFI is put on RAID1 soft volume and I have no GPT partition of type `EFI System`.

**Steps to Reproduce**

```sh
~$ sudo fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Devices with no available firmware updates: 
 • INTEL SSDPEKKW512G7
 • INTEL SSDPEKKW512G7
 • MTFDDAK240TCC-1AR1ZABHB
 • MZVKW512HMJP-000H1
 • MZVKW512HMJP-000H1
 • WDC WD10EZEX-60WN4A0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 78 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures. If the installation fails, you will    ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly. You may have to       ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Blocked executable in the ESP, ensure grub and shim are up to date: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b

~$ echo $? # exit code is 1
1
```

**Expected behavior**

I would like to update UEFI dbx correctly.

**fwupd version information**

```shell
$ sudo fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-56-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```
I'm on a updated Ubuntu 22.04 and fwupdate and the daemon fwupd are installed via APT (classical install): 

[... 382 lines cut ...]
Device           Start        End   Size Type       Type-UUID                            Name          UUID
/dev/nvme1n1p1    2048    1050623   512M Linux RAID A19D880F-05FC-4D3B-A006-743F0F84911E efi-disk-2    8A1101BC-39AF-D24D-B775-60B9884E65CD
/dev/nvme1n1p2 1050624 1000215182 476.4G Linux RAID A19D880F-05FC-4D3B-A006-743F0F84911E system-disk-2 9BDA0AE3-7C9E-2A4F-B7E9-9C224EFA6FAC
```

So I have this error:

```
~$ sudo fwupdtool esp-list --verbose
14:00:50:0293 FuDebug              Verbose debugging enabled (on console 1)
14:00:50:0386 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0400 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0416 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0445 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0462 FuMain               no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b, falling back to ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
14:00:50:0562 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0580 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0597 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
14:00:50:0623 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b: no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
```

And I'm unable to update my UEFI dbx.

Thanks in advance for your help.

hughsie, 2022-12-17:
Can you try with something more modern please? i.e. 1.8.7 or 1.8.8

flaf, 2022-12-17:
Yes, but I'm not sure if it's a good test because I can try a modern version easily via a **snap** installation. Here is my test:

```sh
~$ snap search fwupd
Name            Version              Publisher       Notes    Summary
fwupd           1.8.7-52-g1024d97d6  richard-hughes  classic  Firmware updates for Linux
uefi-fw-tools   1.5.4-0.7.2+git      canonical✓      -        UEFI Firmware update tools for Ubuntu Core
gnome-firmware  3.34.0-32-g4dc0170   superm1         -        Install firmware on devices

~$ sudo snap install fwupd && echo INSTALLED
error: This revision of snap "fwupd" was published using classic confinement and thus may perform
       arbitrary system changes outside of the security sandbox that snaps are usually confined to,
       which may put your system at risk.

       If you understand and want to proceed repeat the command including --classic.

~$ sudo snap install fwupd --classic && echo INSTALLED
fwupd 1.8.7-52-g1024d97d6 from Richard Hughes installed
INSTALLED

~$ /snap/bin/fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: Will measure elements of system integrity around an update
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:measure-system-integrity for more information.
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • System Firmware
 • INTEL SSDPEKKW512G7
 • INTEL SSDPEKKW512G7
 • MTFDDAK240TCC-1AR1ZABHB
 • MZVKW512HMJP-000H1
 • MZVKW512HMJP-000H1
 • UEFI dbx
 • WDC WD10EZEX-60WN4A0
No updatable devices

$ echo $?
2
```

Am I missing something?

flaf, 2022-12-17:
Sorry, maybe this test is more relevant:

```sh
~$ /snap/bin/fwupdmgr refresh
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: Will measure elements of system integrity around an update
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:measure-system-integrity for more information.
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Successfully downloaded new metadata: 2 local devices supported

~$ /snap/bin/fwupdmgr update
WARNING: UEFI ESP partition not detected or configured
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
WARNING: Will measure elements of system integrity around an update
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:measure-system-integrity for more information.
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 • INTEL SSDPEKKW512G7
 • INTEL SSDPEKKW512G7
 • MTFDDAK240TCC-1AR1ZABHB
 • MZVKW512HMJP-000H1
 • MZVKW512HMJP-000H1
Devices with the latest available firmware version:
 • System Firmware
 • WDC WD10EZEX-60WN4A0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures. If the installation fails, you will    ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly. You may have to       ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Decompressing…           [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: No ESP or BDP found

~$ echo $?
1
```

superm1, 2022-12-17:
What do you mean ESP is on raid1 software volume? md? It's not really cloned across your disks and you manually sync it?

I think we need to better understand the specifics of the layout of your disks and partitions for guidance.

flaf, 2022-12-17:
In my first message, I have put some command outputs to explain my installation (see the output of lsblk, sfdisk commands). Isn't that enough? I try to explain again. I have 2 disks:
- `/dev/nvme0n1` 
- `/dev/nvme1n1`

These 2 disks have the same GPT partition where the first partition is a Linux Raid partition type (size==512MiB). These 2 partitions are:
- `/dev/nvme0n1p1` 
- `/dev/nvme1n1p1`

And in these 2 partitions a RAID1 soft volume is defined (`/dev/md127`) via mdadm. Here, I put again an extract of the lsblk command:

```sh
$ sudo lsblk 
NAME             MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS

# [...]

nvme0n1          259:2    0 476.9G  0 disk  
├─nvme0n1p1      259:6    0   512M  0 part  
│ └─md127          9:127  0 511.9M  0 raid1 /boot/efi # <==== Here!
└─nvme0n1p2      259:7    0 476.4G  0 part  
  └─md126          9:126  0 476.3G  0 raid1 
    ├─vg0-system 253:0    0    50G  0 lvm   /var/snap/firefox/common/host-hunspell
    │                                       /
    └─vg0-swap   253:1    0    64G  0 lvm   [SWAP]
nvme1n1          259:3    0 476.9G  0 disk  
├─nvme1n1p1      259:4    0   512M  0 part  
│ └─md127          9:127  0 511.9M  0 raid1 /boot/efi # <==== Here!
└─nvme1n1p2      259:5    0 476.4G  0 part  
  └─md126          9:126  0 476.3G  0 raid1 
    ├─vg0-system 253:0    0    50G  0 lvm   /var/snap/firefox/common/host-hunspell
    │                                       /
    └─vg0-swap   253:1    0    64G  0 lvm   [SWAP]
```

Is it more clear or do you want another information?

superm1, 2022-12-17:
So sorry; I was looking on my phone and missed that section.  The BIOS won't actually use redundancy here between the ESP's.  If there is corruption on the first one I don't expect the system uses the redundancy of the second to boot..

I think this situation needs manually setting what the ESP is in the conffile.

flaf, 2022-12-17:
> So sorry; I was looking on my phone and missed that section.

No problem.

> The BIOS won't actually use redundancy here between the ESP's. If there is corruption on the first one I don't expect the system uses the redundancy of the second to boot..

At that time, I had tested and it worked even by disconnecting one of 2 disks... but well... it was a long time ago. :)

> I think this situation needs manually setting what the ESP is in the conffile.

How can I do that? Sorry, I don't know at all fwupd which I have discovered today with my issue.

flaf, 2022-12-17:
I have tried this:

1. In the file `/etc/fwupd/uefi_capsule.conf`, I have added `OverrideESPMountPoint=/boot/efi` (in the section `[uefi_capsule]`,
2. And then `sudo systemctl restart fwupd.service`

But that seems to be not good because I have:

```sh
$ sudo systemctl status fwupd.service
● fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Sat 2022-12-17 17:47:02 CET; 8s ago
       Docs: https://fwupd.org/
   Main PID: 21057 (fwupd)
      Tasks: 6 (limit: 76756)
     Memory: 7.2M
        CPU: 927ms
     CGroup: /system.slice/fwupd.service
             └─21057 /usr/libexec/fwupd/fwupd

Dec 17 17:47:01 flhpz4 systemd[1]: Starting Firmware update daemon...
Dec 17 17:47:02 flhpz4 fwupd[21057]: 16:47:02:0424 FuEngine             failed to update history database: device ID 5dec21036bd49fc89b48a2f7ccfef58e764ae26f was not found
Dec 17 17:47:02 flhpz4 systemd[1]: Started Firmware update daemon.
```

hughsie, 2022-12-17:
Is that the correct conf file for the snap version?

flaf, 2022-12-18:
> Is that the correct conf file for the snap version?

Indeed. Sorry I'm not well familiar with snap. So I have done this:

```sh
# Just in case, to avoid getting mixed up between apt and snap.
~$ sudo apt-get purge fwupd

# Installation of the snap version (a recent version).
~$ sudo snap install fwupd --classic

~$ sudo /snap/bin/fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.8
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.8
runtime   org.kernel                    5.15.0-56-generic

# The good conf file for snap. I put the line OverrideESPMountPoint=/boot/efi in
# the section [uefi_capsule].
~$ sudo vim /root/snap/fwupd/current/etc/fwupd/uefi_capsule.conf

~$ sudo /snap/bin/fwupdmgr refresh

# Sorry I didn't think to copy the output. This command ran correctly and offered me to reboot the machine.
~$ sudo /snap/bin/fwupdmgr update

# Reboot of the machine and after the reboot...
~$ /snap/bin/fwupdmgr get-history
WARNING: This package has not been validated, it may not work properly.
HP HP Z4 G4 Workstation PC
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Previous version:   77
  │   Update State:       Success
  │   Last modified:      2022-12-18 11:05
  │   GUID:               20716f78-5f65-5fe7-b32b-55a6a16392ab
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      217
        Remote ID:        lvfs
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.   
```

We can see the new version of UEFI dbx (217). So it's OK and we can close this issue, no?

closed 2022-12-18 by superm1

flaf, 2022-12-18:
Ok, thanks everybody for your help. Bye. :)

ThePhatLeee, 2026-09-11:
Confirming this same "no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b" pattern still shows up in 2026 on Fedora CoreOS 44 (fwupd 2.1.7), with a layout that's a variant on the original report worth noting for anyone else who lands here from a search.

On Fedora CoreOS's `boot_device.mirror` Butane/Ignition config (mdraid1 across two NVMe drives), the layout differs from the original report in one relevant way: `/boot` itself *is* a mirrored, mounted ext4 filesystem (`/dev/md126`), but the two real EFI System Partitions (vfat, GUID `c12a7328-...`) are **not** combined into a RAID member and are **not mounted anywhere** by default — they sit as plain, unmounted per-disk partitions (`esp-1`/`esp-2`). So this isn't quite "ESP inside a RAID1 volume" like the original report — it's "ESP exists as a real, correctly-typed partition, just never gets mounted, on either disk."

```
$ lsblk -f
nvme1n1
├─nvme1n1p2 vfat FAT16 esp-1  <-- real ESP, GUID c12a7328-..., unmounted
├─nvme1n1p3 linux_raid_member  -> md126 ext4 /boot
└─nvme1n1p4 linux_raid_member  -> md127 crypto_LUKS -> root (btrfs)
nvme0n1
├─nvme0n1p2 vfat FAT16 esp-2  <-- mirror twin, also unmounted
├─nvme0n1p3 linux_raid_member  -> md126 (same /boot mirror)
└─nvme0n1p4 linux_raid_member  -> md127 (same root mirror)
```

`/boot/efi` exists as an empty directory but has nothing mounted on it, so `OverrideESPMountPoint=/boot/efi` (the fix that resolved the original report) isn't directly usable as-is here without first mounting one of the two real ESPs there manually -- there's no single "the" ESP the way a non-mirrored install would have, since either `esp-1` or `esp-2` would work but neither is chosen/mounted automatically by CoreOS's own tooling.

Not filing this as a new bug since the root cause and fix category are the same as this existing thread (`no volumes of type <ESP GUID>` because the real ESP isn't where fwupd's auto-detection looks) -- just adding this as a data point in case it helps someone else on Fedora CoreOS's mirrored-boot layout find their way here, or if it's useful input toward the `esp-not-found` wiki page covering this specific "real ESP exists but isn't mounted" variant alongside the "ESP is itself RAID-wrapped" one already covered.

ThePhatLeee, 2026-09-11:
Follow-up after actually testing `EspLocation` (the current key name in fwupd 2.1.7, `/etc/fwupd/fwupd.conf` -- `OverrideESPMountPoint` in `uefi_capsule.conf` from my comment above is the old 2022-era name/location and no longer works on current fwupd) on this box: the unmounted-ESP detail I described above turned out not to be the actual blocker. Even with a real ESP mounted read-only at `/boot/efi` and `EspLocation` pointing at it, `fwupdtool esp-list --verbose` still failed with:

```
Failed to get block devices: failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable
```

`udisks2` simply isn't installed on this Fedora CoreOS server image (it's not part of the base set for a minimal/headless install). fwupd's ESP auto-detection depends on it regardless of `EspLocation` being set correctly. So on a minimal CoreOS server install specifically, the real prerequisite is `rpm-ostree install udisks2` (+ reboot) before any of the ESP-detection discussion above becomes relevant at all -- worth knowing for anyone else hitting this on a similarly minimal server-oriented distro/install, since the symptom (same `esp-not-found` warning) looks identical to the mdraid case but the fix is completely different.

hughsie, 2026-09-12:
I believe the coreos team are ---><--- close to adding udisks to the default image.