fwupd/fwupd #5361: Please provide a quiet switch

noloader, 2022-12-24:
When I run `fwupdmgr` in my update scripts it produces some irrelevant noise. I want to see errors, but not warnings and not info. In the output below `Updating firmware` is a message from my script. The other output is generated by the tool.

```shell
Updating firmware
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates:
 • 032G74
No updatable devices
```

It would be nice to have a quiet switch, like `-q`, to cut down on the extra noise.

hughsie, 2022-12-24:
There's a --json flag you might be interested in...

closed 2023-02-19 by hughsie

noloader, 2023-02-19:
Thanks @hughsie.

Please forgive my ignorance... Could you provide the commit where the option was added?

hughsie, 2023-02-20:
Sure, it was added in 1.7.3 but there's been more and more support added for each command for each release. e.g. you can do `fwupdmgr --version --json` in dbd099507161816177028324467384c23ebbe668