fwupd/fwupd #5364: Rollback protection and TPM PRC0 reconstruction fail on ThinkPad T14s Gen 3 AMD

Jackaed, 2022-12-28:
I am running Kubuntu 22.10, (meaning fwupd is on version 1.8.4, so these issues may already be patched). Rollback protection is marked as disabled, however I have enabled it in the BIOS of my machine. TPM PRC0 reconstruction also fails, however when doing --verbose, the two strings listed are the same. This second issue is probably fixed on current versions, thanks to #5347 fixing #5344, so I will wait for that to make its way to ubuntu.

Both issues persisted when I moved to the Snap version (1.8.7). 

If the rollback protection issue is already fixed, or you would like me to test the current main build, let me know.

superm1, 2022-12-28:
I believe the pcr0 issue is fixed in main. You can try an edge snap if you want to confirm it.

For rollback protection, can you please provide output for `sudo fwupd.fwupdtool security --verbose`.

Jackaed, 2022-12-28:
The edge snap did fix the pcr0 issue, thank you for that.

Output for rollback protection: https://pastebin.com/PKdvCXsB

superm1, 2022-12-28:
Good!

It appears you have BIOS rollback protection enacted; so the BIOS setting worked.

The security processor rollback protection is something that would need to be enabled by the OEM. It's up to them to decide if they want that enabled.

We recently moved it to a higher level because there are reasons that OEMs will potentially turn it off and it's their decision to do so.

closed 2022-12-28 by superm1

Jackaed, 2022-12-28:
Understood, thank you so much for the quick responses!