fwupd/fwupd #5381: Not able to detect Jabra speak 810/710/510  through fwupdmgr get-device

kamasw, 2023-01-06:
**I have installed fwupd package from scratch in buildroot .** 
when check the device list which support firmware upgrade , I wont getting the list of Jabra Speakers of version 510/710/810 which is connected . Only getting 1 device list which is of multimedia card category 

root@milhouse ~ $ fwupdmgr get-devices --json
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
{
  "Devices" : [
    {
      "Name" : "DG4008",
      "DeviceId" : "5a769cf95c25c868bed01310839202749aafbe0c",
      "InstanceIds" : [
        "EMMC\\NAME_DG4008",
        "EMMC\\NAME_DG4008&REV_0x3733313033353137",
        "EMMC\\MAN_0045&OEM_0100",
        "EMMC\\MAN_0045&OEM_0100&NAME_DG4008",
        "EMMC\\MAN_0045&NAME_DG4008&REV_0x3733313033353137",
        "EMMC\\MAN_0045&OEM_0100&NAME_DG4008&REV_0x3733313033353137"
      ],
      "Guid" : [
        "d96b9ab7-ad8f-5a47-a32a-867ddb935935",
        "32c50c3d-3ad1-5334-b6da-2f17aec312d2",
        "5a02344d-585b-507b-ad59-19e96c03ca46",
        "423c0bc8-443b-5a53-93a8-27c621e790a4",
        "366e9264-71ef-55ae-90cc-c6122dbcb4b3",
        "835d06b7-8988-5737-9468-704978501dfa"
      ],
      "Serial" : "0xef7a2312",
      "Plugin" : "emmc",
      "Protocol" : "org.jedec.mmc",
      "Flags" : [
        "internal",
        "updatable",
        "registered"
      ],
      "VendorId" : "EMMC:0x000045",
      "Version" : "0x3733313033353137",
      "VersionFormat" : "number",
      "Icons" : [
        "media-memory"
      ],
      "Created" : 1672983380
    }
  ]
}

root@milhouse ~ $ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.7
runtime   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.gusb          0.4.3
runtime   org.kernel                    4.19.172
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.3

**Let me know if I am missing any thing while compiling fwupd . My assumption is that it should also show Jabra speak device  if connected .**

hughsie, 2023-01-06:
> if I am missing any thing while compiling fwupd

Are you missing USB support perhaps? What does the compile log say? You should also have `org.freedesktop.gusb` in `fwupdmgr --version` if USB support is compiled in.

kamasw, 2023-01-06:
as you can see  from the below output , I already have org.freedesktop.gusb 

root@milhouse ~ $ fwupdmgr --version
runtime org.freedesktop.fwupd 1.8.7
runtime com.hughsie.libjcat 0.1.11
compile org.freedesktop.gusb 0.4.3
runtime org.kernel 4.19.172
compile com.hughsie.libjcat 0.1.11
compile org.freedesktop.fwupd 1.8.7
**runtime org.freedesktop.gusb 0.4.3**

Do you have any idea why my Jabra Speaker not getting listed by fwupdmgr get-devices command .

Appreciate your answer

kamasw, 2023-01-09:
Jabra 810  and 710 VID/PID in lsusb is :
Bus 003 Device 003: ID **0b0e**:**2454** GN Netcom Jabra SPEAK 810
Bus 003 Device 004: ID **0b0e**:**2476** GN Netcom Jabra Speak 710

hughsie, 2023-01-09:
It  seems we don't support those devices: https://github.com/fwupd/fwupd/blob/main/plugins/jabra/jabra.quirk

You could try adding those VID/PIDs -- and that might work.

kamasw, 2023-01-09:
Thanks for this information .
Let me check and confirm

joakim-tjernlund, 2023-01-09:
> Thanks for this information . Let me check and confirm

Jabra has not updated it's FW for 5 years. I have added a couple of  PRs with newer FW but they just sit there.

hughsie, 2023-01-09:
I've been talking to Jabra more recently, as they want to get devices supported again. It's less dire than perhaps we fear.

joakim-tjernlund, 2023-01-09:
> I've been talking to Jabra more recently, as they want to get devices supported again. It's less dire than perhaps we fear.

ah, that would be great :) Any time frame ?

hughsie, 2023-01-09:
> Any time frame ?

We're at the "talking to legal" phase, so $DEITY$ only knows.

kamasw, 2023-01-13:
Thanks for the help @hughsie , your proposed work around works

superm1, 2023-01-13:
If the vids added to quirk list works; please submit a PR to make that happen upstream too.

superm1, 2023-12-11:
Can you please give the output for `sudo fwupdtool install foo.cab --verbose --verbose`?  Need to better understand everything that happened.

joakim-tjernlund, 2023-12-11:
Log for fwupdtool install  --verbose --verbose Jabra-SPEAK_510-2.32.cab
[log.txt](https://github.com/fwupd/fwupd/files/13639007/log.txt)

joakim-tjernlund, 2023-12-11:
FYI, This is how I make the CAB:
https://github.com/joakim-tjernlund/jabra-firmware/tree/speak_510/SPEAK_510/2.32

superm1, 2023-12-11:
I think you added the wrong log

joakim-tjernlund, 2023-12-11:
[log.txt](https://github.com/fwupd/fwupd/files/13641275/log.txt)

OOPS, and now ?

joakim-tjernlund, 2023-12-14:
Any ideas ?

joakim-tjernlund, 2023-12-14:
For good measure I tested with 1.9.3,  same problem there.

hughsie, 2023-12-14:
> writing firmware not supported by device

That feels like we're not attempting write_firmware on the correct device -- perhaps the DFU mode isn't being triggered correctly? I can test tomorrow when I'm back in the office.

joakim-tjernlund, 2023-12-14:
> > writing firmware not supported by device
> 
> That feels like we're not attempting write_firmware on the correct device -- perhaps the DFU mode isn't being triggered correctly? I can test tomorrow when I'm back in the office.

That would be awesome! Thanks.

joakim-tjernlund, 2023-12-18:
Any luck?

joakim-tjernlund, 2024-01-01:
a friendly ping

hughsie, 2024-01-02:
@joakim-tjernlund I tried locally here (first day back from Christmas) and it failed in the same way. So I can track down the regression, do you happen to know the newest fwupd version that it still worked with? e.g. 1.8.6 for example.

joakim-tjernlund, 2024-01-02:
hmm, not really. It was some time after JABRA SPEAK 510 version 2.31 came and that was a good while ago

closed 2024-01-02 by hughsie

joakim-tjernlund, 2024-01-09:
FYI , tested OK on my JABRA 510 now. Thanks

hughsie, 2024-01-10:
great, thanks for testing @joakim-tjernlund