fwupd/fwupd #5386: Thunderbolt controller not detected on Dell XPS 13 9360

lubo, 2023-01-06:
**Describe the bug**

In the title.

**Steps to Reproduce**

 1. Per [ArchWiki](https://wiki.archlinux.org/title/Dell_XPS_13_(9360)#Thunderbolt_firmware_updates), force-enable the controller:

    ```console
    # echo 1 > /sys/bus/wmi/devices/86CCFD48-205E-4A77-9C48-2021CBEDE341/force_power
    1
    ```

 2. Make sure the device is online:

    ```console
    $ cat /sys/bus/thunderbolt/devices/0-0/nvm_version                                        
    26.1
    ```

 3. Run `fwupdmgr get-devices`.

**Expected behavior**

`XPS13 9360 Thunderbolt Controller` should be detected when running `fwupdmgr get-devices`.

**fwupd version information**

```console
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.9
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.9
runtime   org.kernel                    6.1.3-arch1-1
```

Installed using Pacman from Arch Linux Community repository.

**fwupd device information**

<details>
  <summary>Click to expand</summary>

```console
$ fwupdmgr get-devices --show-all-devices
Dell Inc. XPS 13 9360
│
├─Unknown Device:
│     Device ID:          533f8dcc55554ba0d48582a72ece77dc0dd59d69
│     GUID:               c35c2451-ada5-556c-93eb-368ad00355a4 ← GPIO\ID_INT344B:00
│   
├─Core™ i7-7500U CPU @ 2.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Device Flags:       • Internal device
│   
├─HD Graphics 620:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
[... 172 lines cut ...]

```console
$ fwupdmgr get-history
Dell Inc. XPS 13 9360
│
├─XPS13 9360 Thunderbolt Controller:
│ │   Device ID:          f316028454ee963aeddc15ba7a94b81f1303ef82
│ │   Previous version:   21.00
│ │   Update State:       Failed
│ │   Update Error:       failed to get device after update: failed to wait for detach replug: device f316028454ee963aeddc15ba7a94b81f1303ef82 did not come back
│ │   GUID:               dfd51125-338f-56ff-b721-fa3bea8e534e
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Reported to remote server
│ │                       • Hardware is waiting to be replugged
│ │ 
│ └─  New version:      26.01
│       License:          Unknown
│       Description:      
│       The vendor did not supply any release notes.
│  
…
```

I don't use Thunderbolt though and I haven't noticed any problems besides this detection issue.

hughsie, 2023-01-11:
What does `sudo fwupdtool get-devices --plugins thunderbolt --verbose` say please?

lubo, 2023-01-11:
With the device force-enabled:

<details>
  <summary>Click to expand</summary>

```
22:12:11.517 FuDebug              Verbose debugging enabled (on console 0)
22:12:11.517 FuMain               locked /run/lock/fwupdtool
22:12:11.517 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
22:12:11.522 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…: 0%
22:12:11.522 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
22:12:11.522 FuCommon             reading /etc/fwupd/daemon.conf with 2494 bytes
22:12:11.522 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
22:12:11.522 FuConfig             using autodetected max archive size 4.1 GB
22:12:11.524 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
Loading…: 0%
22:12:11.524 FuRemoteList         loading /usr/share/fwupd/metainfo
22:12:11.524 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
22:12:11.524 XbSilo               file: a3ef0e06-e4b3-c5cc-43e5-05c84c32188d, current:a3ef0e06-e4b3-c5cc-43e5-05c84c32188d, cached: (null)
22:12:11.524 XbSilo               loading silo with file contents
22:12:11.524 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
22:12:11.524 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
22:12:11.524 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
22:12:11.525 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
22:12:11.525 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
22:12:11.525 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
22:12:11.525 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
22:12:11.525 FuRemoteList         ignoring unfound remote fwupd
22:12:11.525 FuRemoteList         ordering lvfs-testing=lvfs+1
Loading…: 1%
22:12:11.525 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
22:12:11.525 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
22:12:11.525 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
22:12:11.525 Jcat                 loaded 1 certificates
22:12:11.525 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
22:12:11.525 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
22:12:11.525 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
22:12:11.525 Jcat                 loaded 1 certificates
22:12:11.525 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
22:12:11.525 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
22:12:11.525 FuEngine             client certificate now exists: nothing to do
Loading…: 2%
22:12:11.526 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
22:12:11.526 FuHistory            got schema version of 8
Loading…: 3%
Loading…: 4%
22:12:11.533 FuPlugin             load(/usr/lib/fwupd-1.8.9/libfu_plugin_modem_manager.so)
22:12:11.534 FuContext            added udev subsystem watch of platform
22:12:11.534 FuContext            added udev subsystem watch of block
22:12:11.534 FuContext            added udev subsystem watch of pci
22:12:11.534 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
22:12:11.534 FuPluginDell         Using libsmbios 2.4
22:12:11.534 FuContext            added udev subsystem watch of i2c
22:12:11.534 FuContext            added udev subsystem watch of i2c-dev
22:12:11.534 FuContext            added udev subsystem watch of hidraw
22:12:11.534 FuContext            added udev subsystem watch of gpio
22:12:11.534 FuContext            added udev subsystem watch of mei
22:12:11.534 FuContext            added udev subsystem watch of iommu
22:12:11.534 FuContext            added udev subsystem watch of video4linux
22:12:11.534 FuContext            added udev subsystem watch of msr
22:12:11.534 FuContext            added udev subsystem watch of mtd
22:12:11.534 FuContext            added udev subsystem watch of nvme
22:12:11.536 FuContext            added udev subsystem watch of drm
22:12:11.536 FuContext            added udev subsystem watch of drm_dp_aux_dev
22:12:11.536 FuContext            added udev subsystem watch of serio
22:12:11.536 FuPluginTest         init
22:12:11.536 FuContext            added udev subsystem watch of thunderbolt
22:12:11.536 FuContext            added udev subsystem watch of tpm
22:12:11.536 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
[... 259 lines cut ...]
Loading…: 57%
Loading…: 58%
Loading…: 59%
Loading…: 60%
Loading…: 61%
Loading…: 62%
Loading…: 63%
Loading…: 64%
Loading…: 65%
Loading…: 66%
Loading…: 67%
Loading…: 68%
Loading…: 69%
Loading…: 70%
Loading…: 71%
Loading…: 99%
22:12:30.872 FuEngine             failed to update history database: device ID a60b665c769b019e30f31001e91ab31e24d7a035 was not found
Idle…: 0%
22:12:30.886 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
22:12:30.892 FuMain               No detected devices

22:12:30.895 FuPluginTest         destroy
```
</details>

hughsie, 2023-01-12:
> 22:12:11.940 FuDevice             using f316028454ee963aeddc15ba7a94b81f1303ef82 for dd030000-0082-7508-22f8-71d746f4d818

That looks promising. If you change it to be `sudo FWUPD_PROBE_VERBOSE=1 fwupdtool....` do you get anything more printed to the console about `f316028454ee963aeddc15ba7a94b81f1303ef82`?

lubo, 2023-01-12:
I think I do.

<details>
  <summary>Click to expand</summary>
  
```
10:14:24.542 FuDebug              Verbose debugging enabled (on console 0)
10:14:24.543 FuMain               locked /run/lock/fwupdtool
10:14:24.543 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
10:14:24.548 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…: 0%
10:14:24.548 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
10:14:24.548 FuCommon             reading /etc/fwupd/daemon.conf with 2494 bytes
10:14:24.548 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
10:14:24.548 FuConfig             using autodetected max archive size 4.1 GB
10:14:24.549 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
Loading…: 0%
10:14:24.550 FuRemoteList         loading /usr/share/fwupd/metainfo
10:14:24.550 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
10:14:24.550 XbSilo               file: a3ef0e06-e4b3-c5cc-43e5-05c84c32188d, current:a3ef0e06-e4b3-c5cc-43e5-05c84c32188d, cached: (null)
10:14:24.550 XbSilo               loading silo with file contents
10:14:24.550 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
10:14:24.550 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
10:14:24.551 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
10:14:24.551 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
10:14:24.551 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
10:14:24.551 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
10:14:24.551 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
10:14:24.551 FuRemoteList         ignoring unfound remote fwupd
10:14:24.551 FuRemoteList         ordering lvfs-testing=lvfs+1
Loading…: 1%
10:14:24.551 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:14:24.551 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
10:14:24.551 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
10:14:24.552 Jcat                 loaded 1 certificates
10:14:24.552 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
10:14:24.552 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
10:14:24.552 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
10:14:24.552 Jcat                 loaded 1 certificates
10:14:24.552 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
10:14:24.552 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
10:14:24.552 FuEngine             client certificate now exists: nothing to do
Loading…: 2%
10:14:24.552 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
10:14:24.552 FuHistory            got schema version of 8
Loading…: 3%
Loading…: 4%
10:14:24.562 FuPlugin             load(/usr/lib/fwupd-1.8.9/libfu_plugin_modem_manager.so)
10:14:24.562 FuContext            added udev subsystem watch of platform
10:14:24.562 FuContext            added udev subsystem watch of block
10:14:24.563 FuContext            added udev subsystem watch of pci
10:14:24.563 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
10:14:24.563 FuPluginDell         Using libsmbios 2.4
10:14:24.563 FuContext            added udev subsystem watch of i2c
10:14:24.563 FuContext            added udev subsystem watch of i2c-dev
10:14:24.563 FuContext            added udev subsystem watch of hidraw
10:14:24.563 FuContext            added udev subsystem watch of gpio
10:14:24.563 FuContext            added udev subsystem watch of mei
10:14:24.563 FuContext            added udev subsystem watch of iommu
10:14:24.563 FuContext            added udev subsystem watch of video4linux
10:14:24.563 FuContext            added udev subsystem watch of msr
10:14:24.563 FuContext            added udev subsystem watch of mtd
10:14:24.563 FuContext            added udev subsystem watch of nvme
10:14:24.565 FuContext            added udev subsystem watch of drm
10:14:24.565 FuContext            added udev subsystem watch of drm_dp_aux_dev
10:14:24.565 FuContext            added udev subsystem watch of serio
10:14:24.565 FuPluginTest         init
10:14:24.565 FuContext            added udev subsystem watch of thunderbolt
10:14:24.565 FuContext            added udev subsystem watch of tpm
10:14:24.565 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
[... 3502 lines cut ...]
  AcquiesceDelay:       2500
  Subsystem:            platform
  SysfsPath:            /sys/devices/platform/regulatory.0

10:14:25.172 FuEngine             /sys/devices/platform/rtc-efi.0 added FuUdevDevice:
  Flags:                none
  BackendId:            /sys/devices/platform/rtc-efi.0
  AcquiesceDelay:       2500
  Subsystem:            platform
  SysfsPath:            /sys/devices/platform/rtc-efi.0

10:14:25.174 FuEngine             /sys/devices/platform/rtc-efi.0 added FuUdevDevice:
  Flags:                none
[... cut ...] <- PLATFORM\MODALIAS_platform:rtc-efi
  Guid[quirks]:         d26c0d04-6d16-53d6-b712-0a737c6b8d50 <- PLATFORM
  BackendId:            /sys/devices/platform/rtc-efi.0
  AcquiesceDelay:       2500
  Subsystem:            platform
  SysfsPath:            /sys/devices/platform/rtc-efi.0

10:14:25.174 FuEngine             /sys/devices/platform/serial8250 added FuUdevDevice:
  Flags:                none
  BackendId:            /sys/devices/platform/serial8250
  AcquiesceDelay:       2500
  Subsystem:

hughsie, 2023-01-12:
@superm1 any idea what `thunderbolt ignoring: missing authorized attribute` means?

superm1, 2023-01-12:
Well so the check for it was introduced with 76cc23c736.  It was around the context of composite updates for docks.  If the attribute isn't visible, then it means that the security policy/security level in BIOS setup doesn't allow it.
https://github.com/torvalds/linux/blob/71a7507afbc3f27c346898f13ab9bfd918613c34/drivers/thunderbolt/switch.c#L2098

lubo, 2023-01-19:
So, what can be done about this?

hughsie, 2023-09-23:
@lubo I don't think anything I'm afraid -- I think the code is working as intended. The security policy/security level in BIOS is stopping the device to be enumerated.

closed 2023-09-23 by hughsie