fwupd/fwupd #5394: Feedback from Lenovo T14 Gen 3 AMD

patch-work, 2023-01-10:
**Describe the question**

Please check the following. 

Both the output of "fwupdmgr security" and Gnome's Device Security show that the "Processor rollback protection" is disabled. According to its description, "Rollback Protection prevents device software from being downgraded to an older version that has security problems". However, firmware downgrade protection is enabled in the BIOS. 

**fwupd version information**

```fwupdmgr --version

compile   org.freedesktop.fwupd         1.8.9
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.9
runtime   org.kernel                    6.0.18-300.fc37.x86_64
```

[image]
[image]
[image]

superm1, 2023-01-10:
That option controls BIOS rollback protection not secure processor rollback protection.
I guess this issue means we need to further disambiguate it in the docs.

patch-work, 2023-01-10:
The BIOS options with the "Secure Processor" (TPM/Pluton) do not include Pluton rollback protection, but they do include the option to disable Pluton (see photos). 

According to the propaganda, the "AMD Secure Processor" implements Intel Pluton, which claims to be "always up-to-date" "through Windows Update". 

I do not use Windows, so I disabled Pluton (see photo).

[image]
[image]

https://www.amd.com/en/technologies/pro-security

superm1, 2023-01-10:
I'm not referring to pluton.  Notice in the page you linked there are two security processors referred:

- Microsoft Pluton Security Processor
- AMD Secure processor

The check here is specifically for rollback protection of the AMD secure processor, it's up to the manufacturer to decide whether they want that enabled or not.  Let me draft up a PR to make the documentation a little more clear.

BTW in your above screenshots I notice GNOME panel showed encrypted RAM off but fwupdmgr showed it on.  Were those taken at different boots where you changed the BIOS setting?  Or do we actually have an inconsistency?

patch-work, 2023-01-10:
> The check here is specifically for rollback protection of the AMD secure processor, it's up to the manufacturer to decide whether they want that enabled or not. Let me draft up a PR to make the documentation a little more clear.

OK

> BTW in your above screenshots I notice GNOME panel showed encrypted RAM off but fwupdmgr showed it on. Were those taken at different boots where you changed the BIOS setting? Or do we actually have an inconsistency?

Same session, so there is an inconsistency.

superm1, 2023-01-10:
>Do we need to lobby Lenovo for rollback protection?

Not necessarily.  Manufacturers may have valid reasons to not need it enabled.  For example, if the EC validates the entire SPI image and you have BIOS/EC rollback protection enacted then it's **very difficult** to flash an updated security processor firmware.

This is part of the reason that we bumped it up to HSI-4 in the first place.  We found that there are reasons that manufacturers choose not to enable it, and it's up to their product decisions and other security mechanisms in place whether it's still a major threat.

>Same session, so there is an inconsistency.

Glad I checked.  OK... can you please open a bug in the appropriate GNOME tracker?  It looks like fwupd is probably doing it right to me, but the GNOME panel might not be using the flags properly.

patch-work, 2023-01-11:
https://gitlab.gnome.org/GNOME/gnome-control-center/-/issues/2176

SamuelBMartins, 2023-01-11:
Here it is. 
Btw I had "Rollback protection" disabled in the past and [these instructions](https://forums.lenovo.com/t5/Fedora/AMD-Rollback-protection-not-detected-by-fwupd-on-T14-G3-AMD/m-p/5182708?page=1#5810366) helped me enable it. I noticed the option "Rollback protection" does not exists anymore but now there is "Processor rollback protection" and "BIOS rollback protection". Maybe those instruction can help for your problem too.
[image]

patch-work, 2023-01-11:
I followed the instructions. I had the same options set, except for "OS Optimized Defaults". This is the result:

[image]

superm1, 2023-01-11:
@mrhpearson can you explain?  Is the action of loading "Optimized OS defaults" what is getting security processor rollback protection enabled?

If so; we may want to amend our documentation for this issue.  Thoughts?

smallorange, 2023-01-16:
Hi,
We've proposed our new GUI in GNOME44 to improve the user experience and the improvement also covers this issue. Could you try the latest version of the firmware security setting panel?

commit 41ea8c9a (2023-01-10): trivial: clarify documentation on PSP security processor (Fixes: #5394)
commit 17d92d96 (2023-01-10): trivial: clarify documentation on PSP security processor (Fixes: #5394)
commit b6b3f3b0 (2023-01-11): Update documentation to indicate that loading "OS Optimized Defaults"
may enable security processor rollback protection on Lenovo systems.

fixes: #5394
PR #5408 (2023-01-18): Update documentation to indicate that loading "OS Optimized Defaults" may enable security processor rollback protection on Lenovo systems.
fixes: #5394

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [x] Documentation
closed 2023-01-18 by hughsie