fwupd/fwupd #5402: s390x/ppc64el fail autopkgtest

superm1, 2023-01-10:
W/ 1.8.9 + 6c18cc0f66b10dd63ae2947f56ba9c3a078fc4d8 I'm finding that both Debian and Ubuntu are failing autopkgtest specific to s390x and ppc64el.

```
# Start of mtd tests
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.WWNEY1
# XbSilo-DEBUG: failed to load silo: blob too small
Bail out! FuContext-FATAL-WARNING: Failed to load SMBIOS: neither SMBIOS or DT found
```

I guess it's because that environment change we made isn't set for running installed tests.  I suspect it's because we forgot to install the symlink for `plugins/mtd/tests/dmi`

superm1, 2023-01-10:
Here's one link to the failure: https://autopkgtest.ubuntu.com/results/autopkgtest-lunar/lunar/ppc64el/f/fwupd/20230105_182402_168df@/log.gz.
The other 3 look the same.

hughsie, 2023-01-10:
Isn't the symlink set up in 6c18cc0f66b10dd63ae2947f56ba9c3a078fc4d8?

superm1, 2023-01-10:
It's not installed though into installed-tests.  Neither is the DMI data from libfwupdplugin.  So we either need to fail the test if missing data or install the data too.

hughsie, 2023-01-10:
Ahh, installed-tests. Which would you rather do?

superm1, 2023-01-10:
I think it's better to install the dmi data.  We can have more reproducible tests even in installed test environment then too.

superm1, 2023-01-18:
With W/ 1.8.9 + 6c18cc0f66b10dd63ae2947f56ba9c3a078fc4d8 + c2a494e8711dd4d5813d72aed126091fe6a3434f it's [still failing](https://ci.debian.net/data/autopkgtest/testing/ppc64el/f/fwupd/30480472/log.gz) unfortunately.

```
# Start of mtd tests
# GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
# XbSilo-DEBUG: attempting to load /tmp/.RPTZY1
# XbSilo-DEBUG: failed to load silo: blob too small
Bail out! FuContext-FATAL-WARNING: Failed to load SMBIOS: neither SMBIOS or DT found

(/usr/libexec/installed-tests/fwupd/mtd-self-test:4665): FuContext-WARNING **: 19:05:09.005: Failed to load SMBIOS: neither SMBIOS or DT found
```

Any other ideas?

superm1, 2023-01-18:
And I did confirm that c2a494e was applied properly since this showed up in the fwupd-tests binary package:
```
drwxr-xr-x root/root         0 2023-01-16 20:36 ./usr/share/installed-tests/fwupd/
drwxr-xr-x root/root         0 2023-01-16 20:36 ./usr/share/installed-tests/fwupd/dmi/
drwxr-xr-x root/root         0 2023-01-16 20:36 ./usr/share/installed-tests/fwupd/dmi/class/
-rw-r--r-- root/root         3 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/class/chassis_type
-rw-r--r-- root/root        10 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/class/sys_vendor
drwxr-xr-x root/root         0 2023-01-16 20:36 ./usr/share/installed-tests/fwupd/dmi/tables/
-rw-r--r-- root/root      2523 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/tables/DMI
-rw-r--r-- root/root        31 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/tables/smbios_entry_point
drwxr-xr-x root/root         0 2023-01-16 20:36 ./usr/share/installed-tests/fwupd/dmi/tables64/
-rw-r--r-- root/root      5829 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/tables64/DMI
-rw-r--r-- root/root        24 2023-01-03 12:15 ./usr/share/installed-tests/fwupd/dmi/tables64/smbios_entry_point
```

hughsie, 2023-01-19:
> Failed to load SMBIOS: neither SMBIOS or DT found

That string no longer exists; you sure that's the right patch?

superm1, 2023-01-20:
1.8.9 and those two patches it should still be there

hughsie, 2023-01-20:
... buuut 2b0f92506b51ffa230266b843e64b9230574da89 removed the `neither SMBIOS or DT found` string -- did you get that one too?

superm1, 2023-01-20:
I didn't add that one because it was a pretty big refactor for a distro patch, I was looking to mitigate risk.  How soon do you want to do 1.8.10?  Maybe it should just wait and try again with that.

superm1, 2023-01-20:
Yeah; that would be great thanks.

superm1, 2023-01-24:
1.8.10 seems to still fail autopkgtest.  It's failing on !x86 now :(

arm64:
```
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: 21:12:50.137 FuContext            loading HwId overrides from /etc/fwupd/daemon.conf
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: 21:12:50.137 FuContext            not loading HwId overrides from /var/lib/fwupd/daemon.conf
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: 21:12:50.137 FuContext            unable to read SMBIOS data from /sys/class/dmi/id/board_vendor: Failed to open file "/sys/class/dmi/id/board_vendor": No such file or directory
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: 21:12:50.137 FuContext            unable to read SMBIOS data from /sys/class/dmi/id/board_name: Failed to open file "/sys/class/dmi/id/board_name": No such file or directory
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: Failed to load daemon: failed to load engine: Failed to load HWIDs FDT: no data for compatible
Jan 24 05:12:50 ci-024-7f266a68 fwupd[4103]: 21:12:50.138 FuPluginTest         destroy
Jan 24 05:12:50 ci-024-7f266a68 systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Jan 24 05:12:50 ci-024-7f266a68 systemd[1]: fwupd.service: Failed with result 'exit-code'.
Jan 24 05:12:50 ci-024-7f266a68 systemd[1]: Failed to start Firmware update daemon.
FAIL: fwupd/fwupdmgr.test (Child process exited with code 1)
```

```
FuContext            unable to read SMBIOS data from /sys/class/dmi/id/board_vendor: Failed to open file "/sys/class/dmi/id/board_vendor": No such file or directory
FuContext            unable to read SMBIOS data from /sys/class/dmi/id/board_name: Failed to open file "/sys/class/dmi/id/board_name": No such file or directory
Failed to load daemon: failed to load engine: Failed to load HWIDs FDT: no data for compatible
```
https://ci.debian.net/data/autopkgtest/testing/arm64/f/fwupd/30640279/log.gz

superm1, 2023-01-24:
ppc64le:
```
Jan 23 21:55:00 ci-023-b7dc8098 fwupd[4629]: Failed to load daemon: failed to load engine: Failed to load HWIDs FDT: no image id (null) found in firmware
```

armel:
```
Jan 23 15:13:31 ci-023-644a0ca3 fwupd[4644]: Failed to load daemon: failed to load engine: Failed to load HWIDs FDT: no data for compatible
```

superm1, 2023-01-24:
So I guess we're missing some dummy data for !x86

hughsie, 2023-01-24:
> no image id (null) found in firmware

This means `/sys/firmware/fdt` (or the local `system.dtb`) exists, but had no root node.

> no data for compatible

This means `/sys/firmware/fdt` exists, has a root node, but has no `compatible` -- which is basically a hard requirement for DT.

Both unexpected failures, but not the kind of thing that should stop the daemon. Give me a few mins and I'll do something sensible in a new PR.

hughsie, 2023-01-24:
But I am really confused why these don't/can't fail CI.

superm1, 2023-01-25:
Good news everyone!

∙ ∙ autopkgtest for [fwupd](https://tracker.debian.org/pkg/fwupd)/1.8.10-2: [amd64](https://ci.debian.net/packages/f/fwupd/testing/amd64): [Pass](https://ci.debian.net/data/autopkgtest/testing/amd64/f/fwupd/30666782/log.gz), [arm64](https://ci.debian.net/packages/f/fwupd/testing/arm64): [Pass](https://ci.debian.net/data/autopkgtest/testing/arm64/f/fwupd/30667152/log.gz), [armel](https://ci.debian.net/packages/f/fwupd/testing/armel): [Pass](https://ci.debian.net/data/autopkgtest/testing/armel/f/fwupd/30667528/log.gz), [armhf](https://ci.debian.net/packages/f/fwupd/testing/armhf): [Pass](https://ci.debian.net/data/autopkgtest/testing/armhf/f/fwupd/30667904/log.gz), [i386](https://ci.debian.net/packages/f/fwupd/testing/i386): [Pass](https://ci.debian.net/data/autopkgtest/testing/i386/f/fwupd/30668299/log.gz), [s390x](https://ci.debian.net/packages/f/fwupd/testing/s390x): [Pass](https://ci.debian.net/data/autopkgtest/testing/s390x/f/fwupd/30668809/log.gz)

commit 16096ef7 (2023-01-11): trivial: install dmi data into installed-tests

fixes: #5402
PR #5404 (2023-01-11): trivial: install dmi data into installed-tests
fixes: #5402

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit c2a494e8 (2023-01-11): trivial: install dmi data into installed-tests

fixes: #5402
closed 2023-01-25 by superm1