fwupd/fwupd #5415: Ryzen 9 7950X: CPU not supported

kernie, 2023-01-15:
**Describe the bug**
I cannot get HSI classification, because there is data missing for my CPU.

```
fwupdmgr security --force
Host-Sicherheitskennung: HSI:INVALID:missing-data

HSI-1
✔ Sicherer UEFI-Boot:            Aktiviert
✔ TPM empty PCRs:                Gültig
✔ TPM v2.0:                      Gefunden
✔ UEFI-Plattformschlüssel:       Gültig
✘ Unterstützte CPU:              Fehlgeschlagen
```

**Steps to Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.9
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.4-1-default
runtime   org.freedesktop.fwupd         1.8.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
ASRock B650M PG Riptide
│
├─Unbekanntes Gerät:
│     Gerätekennung:      f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─Unbekanntes Gerät:
│     Gerätekennung:      0785bd3eac5b152d9fc34e5549136304ca025e4c
│     GUID:               bc9f6c6f-473f-5bc2-a0d1-3171c965d0d3 ← GPIO\ID_AMDIF031:00
│   
├─Mass storage controller:
│     Gerätekennung:      348a719caa1f7676550b5e92071aff0cdb038f1d
│     Aktuelle Version:   01
│     Anbieter:           Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│     GUIDs:              c12a74b6-78b0-5cfb-95a3-cd65a12f9e94 ← PCI\VEN_1022&DEV_43F6
│                         eb935b99-5869-5b16-ba2c-e2815a2ff8bd ← PCI\VEN_1022&DEV_43F6&REV_01
│                         38b47a77-8710-5b30-b79b-91df5c9a6d96 ← PCI\VEN_1022&DEV_43F6&SUBSYS_1B211062
│                         6f7c3552-52b3-571e-aee3-1135f28ed017 ← PCI\VEN_1022&DEV_43F6&SUBSYS_1B211062&REV_01
│                         67d3b17f-d200-5c56-979e-8a3a4ab46c2b ← PCI\VEN_1022&DEV_43F5
│                         a221019f-651b-56e2-ad6e-7460fd558d9b ← PCI\VEN_1022&DEV_43F5&REV_01
│                         6b2e126d-81a8-530b-a8d3-a93e65bd30b2 ← PCI\VEN_1022&DEV_43F5&SUBSYS_1B213328
│                         63647588-a7c1-5c44-b2d4-72128628c523 ← PCI\VEN_1022&DEV_43F5&SUBSYS_1B213328&REV_01
│     Device Flags:       • Internes Gerät
[... 126 lines cut ...]
│   
└─Unifying Receiver:
      Gerätekennung:      31421989e0cb406a6c65038dfbcc5c87d338d3b6
      Zusammenfassung:    Miniaturised USB wireless receiver
      Aktuelle Version:   RQR12.08_B0030
      Bootloader-Version: BOT01.02_B0014
      Anbieter:           HIDRAW:0x046D|USB:0x046D
      Installationsdauer: 30 Sekunden
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Aktualisierbar
                          • Unterstützt auf dem entfernten Server
                          • Unsignierte Nutzdaten
```

</details>

**Additional questions**
What can I do to receive a HSI classifcation?
I can deliver more logs. And I could also appraoch hardware vendors with bug reports...

- Operating system and version: Opensuse Tumbleweed, Current
- Have you tried rebooting? Yes
- Is this a regression? Don't know.

superm1, 2023-01-15:
> I cannot get HSI classification, because there is data missing for my CPU.

Yeah; unless the kernel exports the information this is the expected behavior.

> What can I do to receive a HSI classifcation?

From your side the only two things you can do are:
1) Enable the CCP kernel driver if you had it disabled.
2) Update your kernel to 5.19 or later if you're on something older (this is the minimum for the feature to work).

If neither of those are the problem, then you will need to contact your motherboard manufacturer to request them to enable this feature in the platform firmware.

kernie, 2023-01-16:
Thank you for your answer!

I checked for the CCP kernel driver:
```
lsmod | grep ccp
ccp                   122880  1 kvm_amd
```
I suspect it's there.

Kernel is 6.1.4-1. That means I contact the motherboard manufacturer...

commit fd4525a4 (2023-01-17): trivial: update supported cpu documentation to add more info for AMD

This should hopefully help disambiguate when information isn't available
what to do.

Fixes: #5415
PR #5419 (2023-01-17): trivial: update supported cpu documentation to add more info for AMD
This should hopefully help disambiguate when information isn't available what to do.

Fixes: #5415

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [x] Documentation
commit 450c3be9 (2023-01-17): trivial: update supported cpu documentation to add more info for AMD

This should hopefully help disambiguate when information isn't available
what to do.

Fixes: #5415
closed 2023-01-17 by superm1