fwupd/fwupd #5421: Upgrade of System Firmware not possible (no HWIDs matched 09cbbed7-aafc-5da4-b4e9-ed1f69c914fb)

silvio, 2023-01-17:
**Describe the bug**
I try to upgrade the system firmware without success.

**Steps to Reproduce**
```shell
$ fwupdmgr refresh
Firmware metadata last refresh: 7 minutes ago. Use --force to refresh again.
$ fwupdmgr upgrade
Devices with no available firmware updates: 
 • 0000:00:1f.5
 • Corsair MP400
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0x01090100 to 0x010b0000?                       ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Fixes and enhancements in System firmware 1.11.0.0:                          ║
║                                                                              ║
║ • Update BIOS to 01.11.00. Refer to hp.com for more details.                 ║
║                                                                              ║
║ HP ZBook Firefly 15 inch G8 Mobile Workstation PC must remain plugged into   ║
║ a power source for the duration of the update to avoid damage.               ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
no HWIDs matched 09cbbed7-aafc-5da4-b4e9-ed1f69c914fb
```

**Expected behavior**
Update of system firmware is possible.

**fwupd version information**
```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.9
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.9
runtime   org.kernel                    6.1.4-arch1-1

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Installed via *pacman*/Archlinux.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
HP HP ZBook Firefly 15 inch G8 Mobile Workstation PC                                                                                                                                                                                                                                                                                                                         
│                                                                                                                                                                                                                                                                                                                                                                            
├─Unknown Device:                                                                                                                                                                                                                                                                                                                                                            
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20                                                                                                                                                                                                                                                                                                           
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00                                                                                                                                                                                                                                                                                          
│                                                                                                                                                                                                                                                                                                                                                                            
├─0000:00:1f.5:                                                                                                                                                                                                                                                                                                                                                              
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1                                                                                                                                                                                                                                                                                                           
│     Summary:            Memory Technology Device                                                                                                                                                                                                                                                                                                                           
│     Vendor:             DMI:HP                                                                                                                                                                                                                                                                                                                                             
[... 156 lines cut ...]
│                         • Device can recover flash failures
│                         • Signed Payload
│   
└─USB4 host controller:
      Device ID:          3a20d193880fa5b5f3498674120b6e4d1dbee6bf
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
     
________________________________________________

Devices that were not updated correctly:

 • USB-C Dock G5 (1.0.13.0 → 1.0.16.0)

Devices that have been updated successfully:

 • System Firmware (0x01080100 → 0x01090100)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
```

</details>

hughsie, 2023-01-17:
Can you grab the output of `sudo fwupdtool hwids` please

hughsie, 2023-01-17:
Okay, thanks. The CDN metadata seems to be correct, but the inside-the-cab-file metadata seems to be incorrect. I've resigned the firmware and now the right thing has appeared. I assume this was a LVFS bug some months ago-- and we've since changed the task scheduler to something more stable than celery. It'll take ~4 hours to refresh stable, but after that's elapsed can you do `fwupdmgr refresh --force` and then `fwupdmgr update` again please. The file you want is https://fwupd.org/downloads/bcedd38efd969d738c8c14beec999a23343b84721e51c9d298a4970ded14f814-T76_01110000.cab if you're feeling impatient.

silvio, 2023-01-18:
@hughsie Thank you very much!
This has helped. I was able to update now!