fwupd/fwupd #5422: Check for MSI insecure firmwares

kerneis-anssi, 2023-01-17:
I don't really if this belongs to the server side or `fwupd security` HSI checks, but it would be nice to ban or warn about MSI firmwares that pretend to enable Secure Boot but actually don't: https://dawidpotocki.com/en/2023/01/13/msi-insecure-boot/

It requires parsing IFR data though, which could be a lot of work on the client side if there's no library available for it. On the server side, I guess it's possible to run UEFITool in a sandbox, see the script provided at the end of the article describing the issue.

hughsie, 2023-01-17:
Is the IFR specification public?

kerneis-anssi, 2023-01-17:
It is part of the UEFI specification. There is a Rust implementation here: https://github.com/LongSoft/IFRExtractor-RS

The part that is not standardized (but in practice every vendor uses the same) is the GUID: https://github.com/Foxboron/sbctl/issues/181#issuecomment-1384720460

kerneis-anssi, 2023-01-17:
If I read sbctl's thread correctly, parsing IFR was necessary to discover automatically vulnerable versions of MSI firmwares. But a more generic solution on a live system might be to check the state of some PCD variables: https://github.com/Foxboron/sbctl/issues/181#issuecomment-1371544441

dawidpotocki, 2023-01-17:
I'm not sure how you would go about reading those PCD variables (I don't
know much about UEFI), but there is a very hacky way to read HiiDB and
then extracting IFR data from that.

More about this here:
https://github.com/Foxboron/sbctl/pull/189#issuecomment-1383107867

It's probably not the best solution as it requires reading `/dev/mem`,
but it does work at runtime.

hughsie, 2023-01-17:
/dev/mem isn't going to work with Secure Boot actually turned on unfortunately.

Foxboron, 2023-01-17:
>/dev/mem isn't going to work with Secure Boot actually turned on unfortunately.

The quirk where linux distributions enable lockdown mode when secure boot is enabled is a patched behavior and absent from the vanilla kernel. Regardless of that I don't think reading /dev/mem is a good strategy.

hughsie, 2023-01-17:
> I don't really know if this belongs to the server side or fwupd security HSI checks

I think it makes sense as a HSI check -- if nothing else because MSI doesn't actually upload firmware for their own hardware on the LVFS...

heretical-blacksheep, 2023-01-17:
"Banning" MSI firmware with this bad default would leave users in an interminable state should yet another vulnerability show itself in AESA or AMD microcode which is still better dealt with at the firmware level than loading microcode after kernel boot.  It's a better idea to just issue a warning to those that care about secure boot (not everyone does!) to check the relevant options in their firmware setup and get on with business otherwise.  After all, MSI isn't on LVFS currently, and likely won't be in the short term future (they're actively hostile to any one mentioning anything but Windows in support tickets - I know from experience) so anyone loading MSI firmware via fwupd is doing so manually.

superm1, 2023-01-17:
If it's between IFR or PCD, it's probably better to build a path to get PCDs. I think this will scale better.

hughsie, 2023-01-17:
@superm1 got any cunning ideas on how to get the PCDs without using /dev/mem?

superm1, 2023-01-17:
Do it from a kernel driver that can access and parse the memory and then export the result from the kernel to userspace :)

NikolajSchlej, 2023-01-17:
No need to parse PCDs because most of them are static (i.e. converted into C `#define`s), so extracting them out of the binaries is way harder than parsing IFR. Also, for AMI Aptio PCDs are a secondary configuration mechanism, with the main one still being SDL TOKENs (that always convert to defines). It's up to you, folks, but I did try to make a tool to reconstruct PCD values of UEFI image and bailed as it proved to be hard even by hand and brain.

NikolajSchlej, 2023-01-17:
As for detecting things like that, it'll be good to have an EFI-level agent that could try to load an unsigned binary with `LoadImage`/`StartImage` and check if the error is returned or not. I've suggested an UEFI Mitigations Test Suite [once already](https://twitter.com/NikolajSchlej/status/1561967464208023552), and this test would be a nice to have there.

NikolajSchlej, 2023-01-17:
> The part that is not standardized (but in practice every vendor uses the same) is the GUID
The GUID comes from AMI, so it's not just not standardized, but AMI-specific, and the fact that downstream OEMs won't changing the GUID of their BIOS Setup (as Intel already does) can't be relied upon.

However, IFR extraction can be applied to every FFS file in the image, as the tool will only generate an `ifr.txt` file in case there are indeed both IFR form packages and IFR string packages in the input file.

We already made a run of IFR extractor across LVFS [once in this issue](https://github.com/LongSoft/IFRExtractor-RS/issues/9), there's a script attached there that could be used as an inspiration for a proper solution, i.e. run it every time a new image is uploaded, and check for configuration issues immediately.

superm1, 2023-01-17:
I think the problem there is we need the client side to make noise about problems.  If you're dependent upon extracting IFR, you need a way to get at it without `/dev/mem`.  Maybe we want an IFR parser in the kernel?

NikolajSchlej, 2023-01-17:
I think we need a generic way to dump the flash contents (that's not always available from `/dev/mem` because it could be unmapped), i.e. make `flashrom` or similar tools work with SecureBoot enabled.

kerneis-anssi, 2023-01-18:
> I think we need a generic way to dump the flash contents (that's not always available from `/dev/mem` because it could be unmapped)

Agreed, but that's https://github.com/fwupd/fwupd/issues/4094#issuecomment-1033439773 all over again, and it's not something easy to get in the kernel.

hughsie, 2023-02-20:
I'll move this to a discussion as it's not really something we can do yet.