fwupd/fwupd #5424: grub2-probe No path or device is specified.

hramrach, 2023-01-18:
**Describe the bug**
Error printed when generating grub configuration

```
No path or device is specified.
Usage: grub2-probe [OPTION...] [OPTION]... [PATH|DEVICE]
Try 'grub2-probe --help' or 'grub2-probe --usage' for more information.
```

**Steps to Reproduce**
`grub2-mkconfig`

**Expected behavior**
no error

**fwupd version information**

https://github.com/fwupd/fwupd/commit/684bc038132977ff28e54550f95941cff5e4ee84

This line https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/fwupd.grub.conf.in#L17 invokes grub-probe without arguments

OS: openSUSE

hughsie, 2023-01-18:
@superm1 is it missing a `${ESP}` perhaps?

superm1, 2023-01-18:
Can I see the generated grub.conf for the distro?

hramrach, 2023-01-18:
```
#! /bin/sh
# SPDX-License-Identifier: LGPL-2.1+
set -e

[ -d ${pkgdatadir:?} ]
# shellcheck source=/dev/null
. "$pkgdatadir/grub-mkconfig_lib"

if [ -f /var/lib/fwupd/uefi_capsule.conf ] &&
   ls /sys/firmware/efi/efivars/fwupd-*-0abba7dc-e516-4167-bbf5-4d9d1c739416 1>/dev/null 2>&1; then
      . /var/lib/fwupd/uefi_capsule.conf
      if [ "${EFI_PATH}" != "" ] && [ "${ESP}" != "" ]; then
      echo "Adding Linux Firmware Updater entry" >&2
cat << EOF
menuentry 'Linux Firmware Updater' \$menuentry_id_option 'fwupd' {
EOF
      ${grub_probe:?}
      prepare_grub_to_access_device '`${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/"'
cat << EOF
	chainloader ${EFI_PATH}
}
EOF
      fi
fi

```

hramrach, 2023-01-18:
hm, this version works, there must be something different on the broken machine

hramrach, 2023-01-18:
I see, the working machine does not have `/var/lib/fwupd/uefi_capsule.conf` which skips execution the broken code.

bzoltan1, 2023-01-18:
Here is the full grub2-mkconfig output with -x added to the script on the broken machine

`$ sudo grub2-mkconfig
[sudo] password for root: 
+ prefix=/usr
+ exec_prefix=/usr
+ datarootdir=/usr/share
+ prefix=/usr
+ exec_prefix=/usr
+ sbindir=/usr/sbin
+ bindir=/usr/bin
+ sysconfdir=/etc
+ PACKAGE_NAME=GRUB2
+ PACKAGE_VERSION=2.06
+ host_os=linux-gnu
+ datadir=/usr/share
+ '[' x = x ']'
+ pkgdatadir=/usr/share/grub2
+ export pkgdatadir
+ grub_cfg=
+ grub_mkconfig_dir=/etc/grub.d
++ basename /usr/sbin/grub2-mkconfig
+ self=grub2-mkconfig
+ grub_probe=/usr/sbin/grub2-probe
+ grub_file=/usr/bin/grub2-file
+ grub_editenv=/usr/bin/grub2-editenv
+ grub_script_check=/usr/bin/grub2-script-check
+ export TEXTDOMAIN=grub2
+ TEXTDOMAIN=grub2
+ export TEXTDOMAINDIR=/usr/share/locale
+ TEXTDOMAINDIR=/usr/share/locale
+ . /usr/share/grub2/grub-mkconfig_lib
++ prefix=/usr
++ exec_prefix=/usr
++ datarootdir=/usr/share
++ datadir=/usr/share
++ bindir=/usr/bin
++ sbindir=/usr/sbin
++ '[' x/usr/share/grub2 = x ']'
++ test x/usr/sbin/grub2-probe = x
++ test x/usr/bin/grub2-file = x
++ test x = x
++ grub_mkrelpath=/usr/bin/grub2-mkrelpath
++ command -v gettext
++ :
++ grub_tab='	'
+ test 0 -gt 0
+ '[' x0 = x ']'
+ '[' 0 '!=' 0 ']'
+ set /usr/sbin/grub2-probe dummy
+ test -f /usr/sbin/grub2-probe
+ :
++ awk '($9!="autofs")&&($9!="securityfs")' /proc/self/mountinfo
++ probe_nfsroot_device
++ read line
+++ echo 23 67 0:21 / /proc rw,nosuid,nodev,noexec,relatime shared:12 - proc proc rw
+++ sed -e 's! - .*$!!'
++ part1='23 67 0:21 / /proc rw,nosuid,nodev,noexec,relatime shared:12'
+++ echo 23 67 0:21 / /proc rw,nosuid,nodev,noexec,relatime shared:12 - proc proc rw
+++ sed -n -e 's! - \(.*\)$!\n\1!p'
+++ sed 1d
++ part2='proc proc rw'
++ set -- 23 67 0:21 / /proc rw,nosuid,nodev,noexec,relatime shared:12
++ path=/proc
++ set -- proc proc rw
++ fstype=proc
++ device=proc
++ '[' x/proc = x/ ']'
++ read line
+++ echo 24 67 0:22 / /sys rw,nosuid,nodev,noexec,relatime shared:2 - sysfs sysfs rw
[... 967 lines cut ...]
++ test x = x
++ grub_probe=/usr/sbin/grub2-probe
++ test x = x
++ grub_file=/usr/bin/grub2-file
++ test x = x
++ grub_mkrelpath=/usr/bin/grub2-mkrelpath
++ command -v gettext
++ :
++ grub_tab='	'
+ '[' -f /var/lib/fwupd/uefi_capsule.conf ']'
+ ls /sys/firmware/efi/efivars/fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
+ . /var/lib/fwupd/uefi_capsule.conf
++ EFI_PATH=/EFI/opensuse/fwupdx64.efi
++ ESP=/boot/efi
+ '[' /EFI/opensuse/fwupdx64.efi '!=' '' ']'
+ '[' /boot/efi '!=' '' ']'
+ echo 'Adding Linux Firmware Updater entry'
Adding Linux Firmware Updater entry
+ cat
menuentry 'Linux Firmware Updater' $menuentry_id_option 'fwupd' {
+ /usr/sbin/grub2-probe
No path or device is specified.
Usage: grub2-probe [OPTION...] [OPTION]... [PATH|DEVICE]
Try 'grub2-probe --help' or 'grub2-probe --usage' for more information.
`

bzoltan1, 2023-01-18:
[a.txt](https://github.com/fwupd/fwupd/files/10447147/a.txt)

superm1, 2023-01-18:
It seems like `${grub_probe:?}` might be the cause then.  Looking at other grub snippets, I don't see a need to keep it in place really.  Can you pull that line out?

hramrach, 2023-01-18:
and possibly similar lines added by 684bc03

superm1, 2023-01-18:
Yeah I think you're right.  Is there a parameter that can be used like `--version`?  If so That is what is needed or we need to waive the shellcheck failure in this case.

hramrach, 2023-01-18:
```
grub2-probe --help
Usage: grub2-probe [OPTION...] [OPTION]... [PATH|DEVICE]
Probe device information for a given path (or device, if the -d option is
given).

  -0                         separate items in output using ASCII NUL
                             characters
  -d, --device               given argument is a system device, not a path
  -m, --device-map=FILE      use FILE as the device map
                             [default=/boot/grub2/device.map]
  -t, --target=TARGET        print TARGET
                             available targets: abstraction, arc_hints,
                             baremetal_hints, bios_hints, compatibility_hint,
                             cryptodisk_uuid, device, disk, drive, efi_hints,
                             fs, fs_label, fs_uuid, gpt_parttype,
                             hints_string, ieee1275_hints, msdos_parttype,
                             partmap, partuuid, zero_check [default=fs]
  -v, --verbose              print verbose messages.
  -?, --help                 give this help list
      --usage                give a short usage message
  -V, --version              print program version

Mandatory or optional arguments to long options are also mandatory or optional
for any corresponding short options.

Report bugs to <bug-grub@gnu.org>.
```

superm1, 2023-01-18:
so `grub2-probe -V` has a zero return code right?

hramrach, 2023-01-18:
yes, and you probably want to redirect it to /dev/null to not break the grub configuration

superm1, 2023-01-18:
Would you mind updating it, trying it and sending up a PR?

hramrach, 2023-01-18:
@bzoltan1 can you test the patch?

It's applicable to the installed `/etc/grub.d/35_fwupd`

hramrach, 2023-01-18:
It's still broken:

```
Adding Linux Firmware Updater entry
+ cat
+ /usr/sbin/grub2-probe --version
+ prepare_grub_to_access_device '`${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/"'
+ old_ifs=' 	
'
+ IFS='
'
++ /usr/sbin/grub2-probe --device '`${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/"' --target=partmap
/usr/sbin/grub2-probe: error: cannot find a GRUB drive for `${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/".  Check your device.map.
+ partmap=

```
fwupd-1.8.9-1.1.x86_64

bzoltan1, 2023-01-18:
### BEGIN /etc/grub.d/35_fwupd ###
Adding Linux Firmware Updater entry
menuentry 'Linux Firmware Updater' $menuentry_id_option 'fwupd' {
/usr/sbin/grub2-probe: error: cannot find a GRUB drive for `${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/".  Check your device.map.

superm1, 2023-01-18:
That second line might need fixing too.

hramrach, 2023-01-18:
Updated the patch.

`zypper in -f fwupd`

and reapply patch please

hramrach, 2023-01-18:
Managed to simulate the capsule config:

`cat /var/lib/fwupd/uefi_capsule.conf`
```
EFI_PATH=foobar
ESP=/boot/efi
```

```
--- grub-config.0	2023-01-18 16:11:38.026771469 +0100
+++ grub-config.1	2023-01-18 16:47:14.871681177 +0100
@@ -411,6 +411,17 @@
 ### END /etc/grub.d/30_uefi-firmware ###
 
 ### BEGIN /etc/grub.d/35_fwupd ###
+menuentry 'Linux Firmware Updater' $menuentry_id_option 'fwupd' {
+	insmod part_gpt
+	insmod fat
+	set root='hd0,gpt1'
+	if [ x$feature_platform_search_hint = xy ]; then
+	  search --no-floppy --fs-uuid --set=root --hint-bios=hd0,gpt1 --hint-efi=hd0,gpt1 --hint-baremetal=ahci0,gpt1  8347-FFC6
+	else
+	  search --no-floppy --fs-uuid --set=root 8347-FFC6
+	fi
+	chainloader foobar
+}
 ### END /etc/grub.d/35_fwupd ###
 
 ### BEGIN /etc/grub.d/40_custom ###

```

PR #5426 (2023-01-18): uefi-capsule: Do not call grub2-probe without arguments
commit 684bc0381 ("trivial: fix various shellcheck warnings") adds call to ${grub_probe} without any argument which causes grub2-probe to return an error, and generating grub configuration fails.

Add --version argument to avoid the error, and redierct the output to /dev/null so that it's not included in the grub configuration file.

The commit also adds superfluous single quotes causing another error:

/usr/sbin/grub2-probe: error: cannot find a GRUB drive for `${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/".  Check your device [... cut]
commit 5688191a (2023-01-18): uefi-capsule: Do not call grub2-probe without arguments

commit 684bc0381 ("trivial: fix various shellcheck warnings") adds call
to ${grub_probe} without any argument which causes grub2-probe to return
an error, and generating grub configuration fails.

Add --version argument to avoid the error, and redierct the output to
/dev/null so that it's not included in the grub configuration file.

The commit also adds superfluous single quotes causing another error:

/usr/sbin/grub2-probe: error: cannot find a GRUB drive for `${grub_probe} --target=device \${ESP}` | sed -e "s/^/\t/".  Check your devic [... cut]
closed 2023-01-18 by superm1