fwupd/fwupd #5428: Failure to upgrade using FuPluginMm

superm1, 2023-01-18:
**Describe the bug**
I haven't used a laptop in my collection a while, and checked for updates and found one for the WWAN modem on it.  This WWAN modem upgrade hung `fwupdtool` for nearly 3 minutes, and then just sat there waiting for device restart for a very long time.

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM120R GL from EM120RGLAPR02A07M4G_10.010.10.010 to                  ║
║ EM120RGLAPR02A07M4G_20.020.20.020?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release :                                                        ║
║                                                                              ║
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Performance improvements                                                   ║
║                                                                              ║
║ EM120R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
WARNING: This package has not been validated, it may not work properly.
Writing…                 [***************************************]
Downloading…             [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
21:53:59.739 FuPluginMm           fu_sahara_loader_qdl_is_open: assertion 'self != NULL' failed

21:56:40.810 GLib-GObject         g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Restarting device…       [***************************************]
Waiting…                 [***************************************]
failed to get device before update reload: failed to wait for detach replug: device 1713527b2dca6a01577895eb7606708367bd9dcf did not come back
Devices with the latest available firmware version:
 • MZVLB512HBJQ-000L7
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Prometheus from 10.01.3273255 to 10.01.3478575?                      ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ New features and enhancements:                                               ║
║                                                                              ║
║ • Fix an authentication issue on FOP project.                                ║
║                                                                              ║
║ Prometheus and all connected devices may not be usable while updating.       ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: WARNING: This package has not been validated, it may not work properly.
21:58:40.871 FuProgress           child is at 0/1 step_max and parent done [../src/fu-tool.c:1228]
1) ../src/fu-tool.c:1228 (16/19)
0) ../src/fu-engine.c:2455 (0/1)

```

A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Upgrade modem (`fwupdtool update`)

**Expected behavior**
Success

**fwupd version information**

Git checkout of c647d0d0f683cbc3a39ff1db1e6462cac052d395 installed from source

superm1, 2023-01-18:
I didn't happen to run it with `--verbose`, after I reboot if the update triggers again.

superm1, 2023-01-18:
Seems the update took after a reboot at least.
```
└─EM120R GL:
      Device ID:          1713527b2dca6a01577895eb7606708367bd9dcf
      Summary:            Quectel EM120 firehose prog file
      Current version:    EM120RGLAPR02A07M4G_20.020.20.020
      Vendor:             quectel (PCI:0x1EAC)
      GUIDs:              d4002226-3331-5f42-aea3-a92371cc2d9b ← PCI\VID_1EAC&PID_1001&REV_0000&CARRIER_DEFAULT
                          4294f9ad-e7b2-53fb-bd16-be64092ebb1d ← PCI\VID_1EAC&PID_1001&REV_0000
                          3b2b9e7f-2620-546f-becb-727b6e4b662f ← PCI\VID_1EAC&PID_1001
                          8efebe65-85c5-58ff-937c-51f626d8fe2a ← PCI\VID_1EAC
      Device Flags:       • Updatable
                          • System requires external power source
                          • Unsigned Payload
```

superm1, 2023-01-18:
@ivanmikh ^

ivanmikh, 2023-01-18:
Hi @superm1 

The `failed to wait for detach replug` happens when fwupd waits for ModemManager to export the modem for too long.
The possible problems might be the modem not booting up after the update or an error during the ModemManager initialization procedure.

Did you see the modem in the list of the PCIe devices after update?

I will also ask someone in the test department to try to reproduce the issue.

Could you clarify a few more details:

- Laptop model
- Linux kernel version
- ModemManager version

superm1, 2023-01-18:
The part about waiting for replug isn't the most concerning to me actually.  It's the asserts..

* ThinkPad L14 Gen 2a
* Kernel 6.1.7
* Modem manager 1.18.6-1

I rebooted the system and the device is present in `lspci`, but I didn't check this after the update.

```
05:00.0 Unassigned class [ff00]: Quectel Wireless Solutions Co., Ltd. EM120R-GL LTE Modem
```

Here's the modem manager journal log from that run in case it helps:
[journal.txt](https://github.com/fwupd/fwupd/files/10451636/journal.txt)

ivanmikh, 2023-01-18:
> The part about waiting for replug isn't the most concerning to me actually. It's the asserts..

Oh yeah, its not a proper use of an assert. It should just check for the port status and return. Will fix it!

hughsie, 2023-09-23:
@superm1 close now?

superm1, 2023-09-23:
Yeah

closed 2023-09-23 by superm1