fwupd/fwupd #5430: Device disconnection / transfer failures when downgrading Intuos BT M and Intuos BT S (wacom_usb)

hardboprobot, 2023-01-19:
**Describe the bug**
`fwupdmgr local-install` fails to downgrade Intuos M and Intuos S devices randomly on some Chromebooks. They both run FW release 2.7 and the downgrade is to 2.6. During the process the devices seem to temporarily appear as disconnected and data transfers fail:

**Intuos BT M:**

```
...
Writing…: 66%
Writing…: 67%
Writing…: 68%
Writing…: 69%
Writing…: 70%
Waiting…: 70%
Verifying…: 70%
failed to refresh status: failed to GetReport: transfer failed
```

System logs show this when the problem happens:

```
2023-01-19T07:43:52.309050Z INFO kernel: [ 1914.394515] usb 3-9: USB disconnect, device number 15
2023-01-19T07:43:52.309652Z WARNING fwupd[15537]: 07:43:52:0309 FuDeviceLocker       failed to close device: failed to re-attach interface: USB error on device 056a:03c7 : No such device (it may have been disconnected) [-4]
2023-01-19T07:43:52.309652Z WARNING fwupd[15537]: 07:43:52:0309 FuHistory            modifying device Intuos BT M [Legacy Bluetooth Module (ID6)] [da35de871e10661b2aade04b20a94bd05d8b4785]
2023-01-19T07:43:52.370833Z WARNING fwupd[15537]: 07:43:52:0370 FuPlugin             attach(wacom_usb)
2023-01-19T07:43:52.370833Z WARNING fwupd[15537]: 07:43:52:0370 FuDeviceList         no replug or re-enumerate required
2023-01-19T07:43:52.371563Z WARNING fwupd[15537]: 07:43:52:0371 FuHistory            modifying device Intuos BT M [Legacy Bluetooth Module (ID6)] [da35de871e10661b2aade04b20a94bd05d8b4785]
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]: 07:43:52:0384 FuEngine             cleanup -> FuWacModuleBluetoothId6:
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  DeviceId:             da35de871e10661b2aade04b20a94bd05d8b4785
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  CompositeId:          4295296d98b3ba38c72f6baa33d24f03a1d428f6
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Name:                 Intuos BT M [Legacy Bluetooth Module (ID6)]
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Guid:                 a8a4e568-3ff2-568d-92bd-7c868efdaade <- USB\VID_056A&PID_03C7-bluetooth-id6
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Plugin:               wacom_usb
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Protocol:             com.wacom.usb
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Flags:                updatable|supported|registered|unsigned-payload
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Vendor:               Wacom Co.,Ltd.
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  VendorId:             USB:0x056A
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Version:              2.9
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  VersionFormat:        bcd
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  VersionRaw:           0x00000209
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Icon:                 input-tablet
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  InstallDuration:      120
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Created:              2023-01-19
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  PhysicalId:           usb:03:00:09
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  LogicalId:            bluetooth-id6
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  RemoveDelay:          10000
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  AcquiesceDelay:       50
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Order:                -1
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  BestDevice:           /dev/urandom
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  FwType:               bluetooth-id6
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Status:               busy
2023-01-19T07:43:52.384246Z WARNING fwupd[15537]:  Command:              end
2023-01-19T07:43:52.384377Z WARNING fwupd[15537]: 07:43:52:0384 FuDeviceLocker       ignoring close error on aborted open: failed to re-attach interface: USB error on device 056a:03c7 : No such device (it may have been disconnected) [-4]
2023-01-19T07:43:52.384377Z WARNING fwupd[15537]: 07:43:52:0384 FuEngine             failed to update-cleanup after failed update: failed to claim HID interface: USB error on device 056a:03c7 : No such device (it may have been disconnected) [-4]
```

**Intuos BT S:**

```
...
Writing…: 50%
Writing…: 63%
Writing…: 64%
Writing…: 65%
Writing…: 66%
Writing…: 67%
Writing…: 68%
Writing…: 69%
Writing…: 70%
Waiting…: 70%
Verifying…: 70%
[... 889 lines cut ...]
      "Flags" : [
        "internal",
        "require-ac",
        "registered",
        "signed-payload"
      ],
      "Vendor" : "INTEL",
      "VendorId" : "THUNDERBOLT:0x8087",
      "VersionFormat" : "pair",
      "Icons" : [
        "thunderbolt"
      ],
      "Created" : 1674113623
    }
  ]
}
```

</details>

**Additional questions**

- Operating system and version: ChromeOS 110 - 15276.0.0
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2023-01-19:
Same behaviour on Linux?

hardboprobot, 2023-01-20:
> Do you want me to poke Wacom about the v2 failure, or have you already done so?

I haven't, I don't have a contact. If you think they can take a closer look at this, please relay this to them. Thanks

jigpu, 2023-01-24:
It looks to me like your issue is the result of a race triggered by fwupd sending the `FU_WAC_MODULE_COMMAND_END` command after the flash finishes. Fwupd tries to query the hardware to see if the command was accepted but it looks like the hardware resets before the query can be sent. The wacom-usb plugin treats the resulting error as fatal and indicates the flash failed even though it actually completed without issue.

I'm not sure if this will help or not, but I've submitted a pull request that fixes a slightly different flash failure symptom that I was experiencing. This pull request allows the wacom-usb plugin to retry commands, which could allow the "Verify" step to complete. I'm a little worried that it might not have an effect though, since maybe the handles to the USB device are invalidated after reset and so the retries wouldn't be able to reach the device after the reset finishes anyway. @hughsie do you know if this is the case?

**Pull request:** https://github.com/fwupd/fwupd/pull/5445

Another possible workaround might be to not do a status query (or fail if a status query fails) when the command being sent is `FU_WAC_MODULE_COMMAND_END`.

hughsie, 2023-01-24:
> but it looks like the hardware resets before the query can be sent

Back a bit; why would `FU_WAC_MODULE_COMMAND_END` cause the hardware to reset? It should be only resetting for `FU_WAC_REPORT_ID_UPDATE_RESET` -- no?

jigpu, 2023-01-24:
Hmm, that's a good point. I wonder why the hardware seems to be resetting before the Verify step finishes...

hughsie, 2023-01-24:
@jigpu you don't want to ship me some hardware so I can test this myself, do you? If so, same address as last time.

hughsie, 2023-02-20:
@jigpu what's the status here? Did you work out why `FU_WAC_MODULE_COMMAND_END` would cause the hardware to reset?

jigpu, 2023-02-20:
@hughsie unfortunately I wasn't able. My understanding of the situation remains unchanged from the private mail I sent you on Jan 30. If you're able to find anything of interest or have questions for the firmware team, let me know and I can set up communication.

hughsie, 2023-06-27:
@hardboprobot does https://github.com/fwupd/fwupd/commit/1225bb4c297fd77933f0f3f863d1a8a131370cd4 fix this for good?

hardboprobot, 2023-06-27:
@hughsie I don't have the equipment but I pinged the guys who were testing it. I'll let you know if it works for them. Thanks!

jigpu, 2023-09-27:
Test results from the testers indicate that failure still occurs with fwupd version 1.9.3 (which has the above commit). Maybe the fixes we're cooking up for some of the other unreliability issues could help with this...

hughsie, 2024-11-11:
I'm pretty sure we fixed this when we moved the reboot into the composite_cleanup.

closed 2024-11-11 by hughsie