fwupd/fwupd #5447: Invalid Vendor name

jkolako-solidigm, 2023-01-25:
How the vendor name is determined when listing device info using fwupdmgr get-devices? VendorID is correct, but I'm expecting Solidigm string instead of Intel Corporation:
├─SOLIDIGM SSDPFKNU512GZL:
│     Vendor:             Intel Corporation (NVME:0x025E)

**fwupd version information**

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.11
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.11
runtime   org.kernel                    5.15.0-58-generic
```

hughsie, 2023-01-25:
It seems the systemd pci.ids was updated here:

    commit 3203190130c8b5150a7595b5efd41531fcb99115
    Author: Luca Boccassi <bluca@debian.org>
    Date:   Mon Oct 24 20:25:30 2022 +0100
    
        Update hwdb

I'm guessing you're running a systemd version older than that? If it bothers you, or you need it for compliance QA reasons, feel free to add:

    [NVME\VEN_025E]
    Vendor = Solidigm

to `nvme.quirk`

jkolako-solidigm, 2023-01-25:
Thank you. All is explained

closed 2023-01-25 by jkolako-solidigm