fwupd/fwupd #5462: Downgrade firmware LG gram

stpnwf, 2023-01-27:
**Describe the question**
Is it possible to downgrade the firmware of my LG Gram 14" 2018 laptop?

I have updated the firmware through Gnome Software and now I see "No device found" show up when I try to use the webcam. Mic also stopped working. :/
[image]

hughsie, 2023-01-30:
What firmware did you update? the dbx configuration?! If so I've never heard of that causing device regressions, although this is firmware so any weirdness is certainly possible. To "downgrade" the dbx in gnome-firmware isn't possible, but you should be able to achieve the same result by loading the factory defaults in the firmware setup menu.

stpnwf, 2023-01-30:
Hello, thanks for your reply.

>What firmware did you update? the dbx configuration?!

I saw that prompt in the Software Store and I clicked update. I do not know exactly what was updated, only that whatever it was, it went from version 77 -> 217.

>but you should be able to achieve the same result by loading the factory defaults in the firmware setup menu

Any advice/pointers about how I would do that? How would I access this firmware menu? Would it affect the OS installed?

hughsie, 2023-01-30:
> it went from version 77 -> 217

That certainly sounds like the dbx update, but I'm completely mystified how that possibly could affect a webcam. @superm1 any ideas?

> Any advice/pointers about how I would do that

superm1, 2023-01-30:
Seems implausible to have been caused by this.

Perhaps can we see the journal from before and after updates for further hints?

stpnwf, 2023-02-06:
Sorry for the delay, hectic week. I have installed a new OS since, Zorin is still on Gnome 38 and I wanted something newer. I do have a Timeshift backup however. Where would I find the journal regarding this update?

I am unsure of what might have caused it, but a few days ago the camera and the microphone just started working again. After a while though it stopped working. It has "come back to life" a couple times since. Even when it starts working, it never works with Cheese for some reason, just with Chromium (I also use Google Meet to test it)

Could such behavior be caused by that firmware update? I never had any issues with my webcam or mic before updating.

superm1, 2023-02-06:
I think it's most likely a coincidence that it stopped working when you took that update and you have some other problem with the machine.

hughsie, 2023-02-06:
> I think it's most likely a coincidence that it stopped working when you took that update

I think so too; sorry.

closed 2023-02-06 by hughsie

stpnwf, 2023-02-06:
Ok, thanks for your time guys.