fwupd/fwupd #5463: 8BitDo SN30 Pro+ is not seen in device list

gudvinr, 2023-01-29:
**Describe the bug**
SN30 Pro+ controller is not displayed when you run `fwupdmgr get-devices` unless you enter bootloader mode (L+R+Start) beforehand.  
But when it's already in bootloader, you can't update.

I can see it in lsusb:
`Bus 001 Device 044: ID 2dc8:6002 8BitDo 8BitDo SN30 Pro+`

And kernel recognizes it:
```kernel: usb 1-3: New USB device found, idVendor=2dc8, idProduct=6002, bcdDevice= 0.01
kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 1-3: Product: 8BitDo SN30 Pro+
kernel: usb 1-3: Manufacturer: 8Bitdo SF30 Pro  
kernel: usb 1-3: SerialNumber: SF30 Pro    
kernel: input: 8Bitdo SF30 Pro   8BitDo SN30 Pro+ as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:2DC8:6002.002B/input/input108
kernel: hid-generic 0003:2DC8:6002.002B: input,hidraw7: USB HID v1.11 Gamepad [8Bitdo SF30 Pro   8BitDo SN30 Pro+] on usb-0000:00:14.0-3/input0
```

**Steps to Reproduce**
1. Turn controller in dinput mode (B+Start)
2. Run `fwupdmgr get-devices`

**Expected behavior**
Controller is listed, firmware updates are available, etc

**fwupd version information**
1.8.10

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.1.8-zen1-1-zen
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: Arch Linux
- Have you tried rebooting? Yes
- Is this a regression? I don't know

superm1, 2023-01-29:
We added support for this device a long time ago (https://github.com/fwupd/fwupd/commit/d2701b12b2efdda664667020538cadd410962aec).  I suppose this means there was a regression along the way.

Can you please add `sudo fwupdtool get-devices --plugins ebitdo --verbose` output while in D-input mode so we can see the probing that fwupd does and any errors?

gudvinr, 2023-01-29:
> `sudo fwupdtool get-devices --plugins ebitdo --verbose` output

<details><summary> output </summary>

```
$ sudo fwupdtool get-devices --plugins ebitdo --verbose
21:36:54.744 FuDebug              Verbose debugging enabled (on console 1)
21:36:54.746 FuMain               locked /run/lock/fwupdtool
21:36:54.746 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
21:36:54.749 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [-                                      ]21:36:54.750 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
21:36:54.750 FuCommon             reading /etc/fwupd/daemon.conf with 2494 bytes
21:36:54.750 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
21:36:54.750 FuConfig             using autodetected max archive size 4.3 GB
21:36:54.750 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]21:36:54.750 FuRemoteList         loading /usr/share/fwupd/metainfo
21:36:54.751 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
21:36:54.751 XbSilo               file: 10d9a28a-c6da-5c45-fc31-6b7ea23792bc, current:0234160a-b688-3724-0dc6-1e24bd2cef84, cached: (null)
21:36:54.755 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
21:36:54.756 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
21:36:54.756 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
21:36:54.757 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
21:36:54.757 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
21:36:54.757 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
21:36:54.758 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
21:36:54.758 FuRemoteList         ignoring unfound remote fwupd
21:36:54.758 FuRemoteList         ordering lvfs-testing=lvfs+1
Loading…                 [                                       ]21:36:54.758 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:36:54.759 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:36:54.759 Jcat                 loaded 1 certificates
21:36:54.759 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:36:54.759 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:36:54.759 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:36:54.759 Jcat                 loaded 1 certificates
21:36:54.759 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:36:54.760 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
21:36:54.760 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
21:36:54.760 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]21:36:54.760 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
21:36:54.760 FuHistory            got schema version of 8
Loading…                 [*                                      ]21:36:54.766 FuPlugin             load(/usr/lib/fwupd-1.8.10/libfu_plugin_modem_manager.so)
21:36:54.771 FuContext            added udev subsystem watch of platform
21:36:54.771 FuContext            added udev subsystem watch of block
21:36:54.771 FuContext            added udev subsystem watch of pci
21:36:54.772 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
21:36:54.772 FuPluginDell         Using libsmbios 2.4
21:36:54.772 FuContext            added udev subsystem watch of i2c
21:36:54.772 FuContext            added udev subsystem watch of i2c-dev
21:36:54.772 FuContext            added udev subsystem watch of hidraw
21:36:54.772 FuContext            added udev subsystem watch of gpio
21:36:54.772 FuContext            added udev subsystem watch of mei
21:36:54.772 FuContext            added udev subsystem watch of iommu
21:36:54.773 FuContext            added udev subsystem watch of video4linux
21:36:54.773 FuContext            added udev subsystem watch of msr
21:36:54.773 FuContext            added udev subsystem watch of mtd
21:36:54.773 FuContext            added udev subsystem watch of nvme
21:36:54.774 FuContext            added udev subsystem watch of drm
21:36:54.774 FuContext            added udev subsystem watch of drm_dp_aux_dev
21:36:54.774 FuContext            added udev subsystem watch of serio
21:36:54.774 FuPluginTest         init
21:36:54.775 FuContext            added udev subsystem watch of thunderbolt
21:36:54.775 FuContext            added udev subsystem watch of tpm
21:36:54.775 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
21:36:54.775 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
21:36:54.775 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
[... cut ...]
21:36:54.834 FuDevice             removing vendor prefix of '8BitDo' from '8BitDo SN30 Pro+'
21:36:54.840 FuEngine             failed to add device usb:01:00:03: unexpected device response
Loading…                 [-                                      ]
21:36:55.151 FuCommon             creating path /var/cache/fwupd/motd.d
21:36:55.151 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
21:36:55.157 FuMain               No detected devices

21:36:55.159 FuPluginTest         destroy
```
</details>

superm1, 2023-01-29:
Here's the problem:

```
21:36:54.840 FuEngine failed to add device usb:01:00:03: unexpected device response
```

gudvinr, 2023-01-29:
I guess so. But what's the "expected" response and why it is unexpected?

superm1, 2023-01-29:
Comes from here

https://github.com/fwupd/fwupd/blob/3f105cf9ecaeb8226fee92f145d5f90369d31b3b/plugins/ebitdo/fu-ebitdo-device.c#L236

There is a verbose environment variable you can set 

`FWUPD_EBITDO_VERBOSE`

Which could help explain what data is failing.

gudvinr, 2023-01-29:
```
22:20:35.823 FuDevice             using 1f914e619b232dc4da68ec3389b98b20163003a2 for usb:01:00:03
22:20:35.824 FuDevice             removing vendor prefix of '8BitDo' from '8BitDo SN30 Pro+'
22:20:35.825 FuPluginEbitdo       ->DEVICE:       05 00 21 01 00 00 
PktLength:   0x05
PktType:     0x00 [user-cmd]
CmdSubtype:  0x21 [get-version]
CmdLen:      0x0001
Cmd:         0x00 [fw-update-data]
Payload Len: 0x0000
22:20:40.829 FuEngine             failed to add device usb:01:00:03: failed to retrieve from device on ep 0x81: transfer timed out
Loading…                 [-                                      ]
22:20:41.130 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
No detected devices
```

gudvinr, 2023-01-29:
It looks similar to #1681 and partially #165 (different controller, similar error). It seems like someone even made [PR for that](https://github.com/fwupd/8bitdo-firmware/pull/24)

superm1, 2023-01-30:
Yeah this looks the same.  Let's re-open that one as these are dups.

closed 2023-01-30 by superm1