fwupd/fwupd #5484: edge snap shows lots of `?` in the output

superm1, 2023-02-03:
Experimenting with the edge snap (1.8.10-67-gaef162223 rev 4646) with an edge snapd (2.58.2+git450.g4241ddc rev 18423) on Ubuntu 22.04.

I notice the following type of output doesn't seem to be using unicode.  

```
$ fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates: 
 ? ThinkPad Thunderbolt 3 Dock
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
????????????????????????????????????????????????????????????????????????????????
? Upgrade Embedded Controller from 0.1.51 to 0.1.54?                           ?
????????????????????????????????????????????????????????????????????????????????
? Lenovo ThinkPad Embedded Z-series Controller Firmware Version 1.54           ?
?                                                                              ?
? 21D2SIT061 must remain plugged into a power source for the duration of the   ?
? update to avoid damage.                                                      ?
????????????????????????????????????????????????????????????????????????????????

Perform operation? [Y|n]:  
```

Same thing seems to happen with `get-devices` output.
This is similar to we hit on Windows I recall.  @valentindavid any ideas?

hughsie, 2023-02-03:
Is the locale setup up correctly? e.g. en_GB**.UTF-8**?

superm1, 2023-02-03:
Running `snap run --shell fwupd.fwupdmgr` gets me a shell.
I then run `export | grep LANG` and it's set for `en_US.UTF-8`

valentindavid, 2023-02-05:
glib should find the terminal is UTF-8. Not sure why it does not. I have to debug that.

valentindavid, 2023-02-07:
Glib uses `nl_langinfo(CODESET)` from `glibc` to detect the encoding. glibc tries to read `locale-archive` which does not exist. And because of that it returns `ANSI_X3.4-1968` (this is ASCII).

Setting environment variable `CHARSET=UTF-8` works around this issue.

I will look into fixing it in the core base.

valentindavid, 2023-02-08:
It is not easily fixable on core side. I think I have to work that out.

We need the need glibc locales for the user's requested `LANG`/`LC_*`. There is this work-around on the forum: https://forum.snapcraft.io/t/the-locales-launch-launcher-fix-glibc-locales-in-the-snap-runtime/10296

Otherwise, users can just set `CHARSET=UTF-8` and it works without any change.

superm1, 2023-02-08:
> It is not easily fixable on core side. I think I have to work that out.
>
> We need the need glibc locales for the user's requested `LANG`/`LC_*`. There is this work-around on the forum: [...]

Is setting CHARSET in the wrapper viable then to avoid tripping this until it can be fixed in core?

valentindavid, 2023-02-08:
The issue is how to detect if we should use UTF-8. We can guess. But we cannot know for sure unless we have locale data.

superm1, 2023-02-08:
But what happens if we just guess that we need it? The case that you don't just shows the `?`?

I would think utf8 is the more common scenario and if we're going to workaround this until core can fix it I would rather make it look right for the more common case.

superm1, 2023-02-16:
I guess as https://github.com/snapcore/core-base/pull/94 was approved (but not yet merged), no actions needed for fwupd snap for this once that merges?

superm1, 2023-05-10:
@valentindavid I tried the upgraded core22, but I still see this issue.  Here are the versions I tried:

core22:  `latest/candidate: 20230503 2023-05-05 (634) 76MB -`
fwupd: `latest/candidate: 1.8.13-5-g33cc09899   2023-03-31 (4797) 23MB -`
snapd: `latest/stable:    2.59.2                 2023-05-02 (19122) 55MB -`

superm1, 2023-07-05:
I've upgraded core22 to latest on my system, and this is now fixed.
> latest/stable:    20230612 2023-06-24 (766) 77MB -

closed 2023-07-05 by superm1