fwupd/fwupd #5485: edge snap sets up "ubuntu-core" path for `fwupdx64.efi`

superm1, 2023-02-03:
Experimenting with the edge snap (1.8.10-67-gaef162223 rev 4646) with an edge snapd (2.58.2+git450.g4241ddc rev 18423) on Ubuntu 22.04.

I find that running an update that the path built for UEFI is `\efi\ubuntu-core\fwupdx64.efi`.  As the snap can run on Ubuntu desktop, or anything else that runs a snap I think we should change this, but I invite discussion.

I would propose the default for the snap should be `\efi\fwupd\fwupx64.efi`

superm1, 2023-02-03:
@valentindavid

superm1, 2023-02-03:
but also something that comes to mind is we have to think about what shim is executing us.  I'm afraid it really locks it down to secure boot from snap is "Ubuntu only".  That is unless we go and get fwupd64.efi signed by Microsoft directly...

superm1, 2023-02-03:
Hmm something else that comes to mind, for the snap we could just expect that the host OS provides the fwupd binary.  But that would mean that we need to have some sort of way to detect where it is.

valentindavid, 2023-02-05:
`/etc/os-release` is read. I think it reads the one from the core snap instead from host. We need to get access to the host one.

superm1, 2023-02-09:
I got around to testing this, and it gives me a few other thoughts I want to discuss.

1. Until the newer snapd lands this we need to better handle the "missing" file.
```
 sudo fwupd.fwupdtool get-devices 
Loading?                 [ -                                     ]Failed to load remotes: Failed to open file ?/var/lib/snapd/hostfs/etc/os-release?: Permission denied
````

2. This really won't work outside of Ubuntu.  Say you use it on Fedora, the Fedora shim won't be able to load the Ubuntu signed fwupdx64.efi that is contained in the snap with secure boot on.  So the way this is solved today will only really help the fwupd confined snap on Ubuntu desktop or Ubuntu core as a host.

So I guess this leaves two ideas for me on how it can be solved.
1. We can include the Ubuntu signed shimx64.efi in the snap and not use the lsb-release, but rather just hardcode is to "ubuntu".  If shim is already there we don't replace it though.
2. We get fwupdx64.efi signed directly from Microsoft and include that instead.  Then we install to a `fwupd` path on the ESP no matter what the host distro is.

@hughsie thoughts?

hughsie, 2023-02-09:
> We get fwupdx64.efi signed directly from Microsoft

That sounds like a ton of legal headaches. I'm not vetoing it, I just don't want to be the one on the hook for the compliance and security side of things.

> only really help the fwupd confined snap on Ubuntu

Maybe we just add checks to error out if the user is using the snap on non-Ubuntu? It seems to be the only real place people seem to be using it in the real world...

valentindavid, 2023-02-09:
> Until the newer snapd lands this we need to better handle the "missing" file.

We are going to get the fix released soon. We should keep this issue open though.

> 2. This really won't work outside of Ubuntu.  Say you use it on Fedora, the Fedora shim won't be able to load the Ubuntu signed fwupdx64.efi that is contained in the snap with secure boot on.  So [...]

This was also an issue with non confined fwupd, right?

superm1, 2023-02-09:
> Maybe we just add checks to error out if the user is using the snap on non-Ubuntu? It seems to be the only real place people seem to be using it in the real world...

I suppose we haven't had anyone complain about this yet, but also the classic snap didn't really work well on a lot of other distros previously.  The confined snap should work everywhere the classic didn't.

> That sounds like a ton of legal headaches. I'm not vetoing it, I just don't want to be the one on the hook for the compliance and security side of things.

Can you talk to folks internally that have gone through it about what it would take?  I naïvely think it's just pay a small amount of money and submit a binary to a portal, the black box runs some tests on it and if you pass and get back a signed version. 

> We are going to get the fix released soon. We should keep this issue open though.

OK.

> This was also an issue with non confined fwupd, right?

Yes it is.  The difference is that I expect we're going to be more widely encouraging the use of this snap now that it's confined and works elsewhere.

valentindavid, 2023-02-09:
We could try to check if secure boot is enabled and if the enrolled keys can verify Ubuntu's shim. So we know it will boot correctly.

If it cannot, we can print a message to ask the user to report the issue with what distro they use, what kind of keys are enrolled and decide how to support their case.

I would think it would not work to use Ubuntu's shim if the secure boot does not have the MicCorUEFCA certificate from Microsoft. In that case it could be:
 * The user enrolled its own keys. In that case we should probably document for the user how to re-sign `fwupx64.efi`, and provide it. Or how to add Ubuntu's certificate as a MOK. (There are plenty of ways).
 * The distro enrolls their keys from a setup mode. We need to know which distros do that.

hughsie, 2023-02-09:
This are the numbers of people using the snap:

[image]

hughsie, 2023-02-09:
> it also confirms we shouldn't over-engineer this solution

Right, this was my point; lets keep it simple as much as possible.

superm1, 2023-02-09:
> I suspect getting MS to agree to sign another project themselves would be a herculean effort. Better to just make sure shim and fwupd in the snap are a matched pair

We don't currently package a shim into the snap, but that gives me another idea.  How about we just do that and for the snap always use the `fwupd` directory on the ESP instead of the distro specific one?  We could then just install both binaries every time you're in SB mode.

It would potentially mean multiple copies of shim/fwupd on the ESP, but it would _always_ work.

vathpela, 2023-02-09:
> It would potentially mean multiple copies of shim/fwupd on the ESP, but it would _always_ work.

Sounds reasonable enough to me.

PR #5501 (2023-02-08): snap: read the host os-release instead of base snap
This should fix #5485. Depends on https://github.com/snapcore/snapd/pull/12545

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 70eed3cb (2023-02-08): snap: read the host os-release instead of base snap

This should fix #5485. Depends on https://github.com/snapcore/snapd/pull/12545
commit 781bba40 (2023-02-09): trivial: snap: install shim into the snap as well

Fixes: #5485
commit ef594ef2 (2023-02-09): trivial: snap: install shim into the snap as well

Fixes: #5485
closed 2023-02-16 by superm1