fwupd/fwupd #5486: Fwupd daemon not starting

jimmyd-be, 2023-02-04:
**Describe the bug**
Service cannot be started because of following error:
FuContext Failed to load SMBIOS: invalid DMI data size, got 4074 bytes, expected 7829
FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specification is unsupported, but /sys/firmware/efi exists: SMBIOS not supported

**Steps to Reproduce**
Install veraion 1.8.10 on Fedora 37. After downgrading back to 1.8.5 it works again.

**Expected behavior**
The service start normally without any errors.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Command ran with version 1.8.5 because versions 1.8.10 does not start
Unknown Vendor Unknown Product
│
├─Unknown Device:
│     Device ID:          21db7068bb0165add8279ad2b4c634bdeb89c035
│     GUID:               d621f1d9-a625-5d39-9359-61ed4baea480 ← GPIO\ID_bcma-gpio
│   
├─BCM20702A0:
│     Device ID:          65a54fb6ce182f0e75edf0e43047d547a0d61f0e
│     Current version:    1.12
│     Vendor:             Broadcom Corp (USB:0x0B05)
│     Serial Number:      40E23013BE22
│     GUIDs:              8e441291-f2bd-5d5b-b0d8-6479e3abd2a2 ← USB\VID_0B05&PID_180A
│                         ff3f3a18-1b33-5339-87fa-0d586aca4644 ← USB\VID_0B05&PID_180A&REV_0112
│     Device Flags:       • Updatable
│   
├─Core™ i7-5960X CPU @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000049
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         3dbf8c68-ea66-5b5e-8b50-e2e7c45dabdf ← CPUID\PRO_0&FAM_06&MOD_3F
│                         079c5e08-ec0d-5042-bec5-42d236d81c1c ← CPUID\PRO_0&FAM_06&MOD_3F&STP_2
│     Device Flags:       • Internal device
│   
├─Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT] (Sapphire Technology Limited Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT]):
│     Device ID:          594baa45f6742120d71dd82166e762f1529c2a8a
│     Current version:    c1
│     Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│     GUIDs:              dd0fc8c4-f037-5a78-bdea-c85eb0fe68fa ← PCI\VEN_1002&DEV_731F
│                         afa52ef7-5494-518d-980c-54328b60c0f3 ← PCI\VEN_1002&DEV_731F&REV_C1
│                         dac79440-d67c-5928-bb85-500544a6c76a ← PCI\VEN_1002&DEV_731F&SUBSYS_1DA2E409
│                         be658b1b-39e9-5883-9964-7c9c7e7af048 ← PCI\VEN_1002&DEV_731F&SUBSYS_1DA2E409&REV_C1
│                         800200db-2c4f-5f72-b833-04fc8a49959f ← PCI\VEN_1002&DEV_1479
│                         1be789d1-ab7b-5422-ba5e-122cd66318f6 ← PCI\VEN_1002&DEV_1479&REV_00
│                         a2e4f111-f931-56a4-aed7-cd100613e661 ← PCI\VEN_1002&DEV_1479&SUBSYS_10021479
│                         22ff632a-8a6d-500e-b8a5-5182e9c2793d ← PCI\VEN_1002&DEV_1479&SUBSYS_10021479&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─SSD 960 PRO 2TB:
│     Device ID:          310f45f1f223064b5c16bf6dff31146755a64480
│     Summary:            NVM Express solid state drive
│     Current version:    4B6QCXP7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S3EXNCAHB01279P
│     GUIDs:              5b3df2da-f745-5fd0-81de-5dafd7f0bf8c ← NVME\VEN_144D&DEV_A804
│                         f87b9ac8-1cb3-5c0a-ae57-7144f211fe5e ← NVME\VEN_144D&DEV_A804&REV_00
│                         aed4d3c0-fd97-5e46-a32f-ff35e0692f6d ← NVME\VEN_144D&DEV_A804&SUBSYS_144DA801
│                         030c853f-259a-57a0-b3fb-1c66100db94b ← NVME\VEN_144D&DEV_A804&SUBSYS_144DA801&REV_00
│                         3fc46114-619a-5be6-9e6d-88339c677ca2 ← Samsung SSD 960 PRO 2TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─ST3000DM001-1CH166:
│     Device ID:          602b0a6cc821d155208724f0e22f8d111542b74c
│     Summary:            ATA drive
│     Current version:    27
│     Vendor:             Seagate (ATA:0x1BB1, OUI:000c50)
│     Serial Number:      Z1F4KDN1
│     GUIDs:              fa3e2611-882a-5f26-bc28-14cfdda26f56 ← IDE\ST3000DM001-1CH166______________________27
│                         628768a6-af1c-5197-a267-47f87ad9c760 ← IDE\0ST3000DM001-1CH166______________________
│                         ecb04be5-0001-5f89-aa77-47d949a5cb81 ← ST3000DM001-1CH166
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    16641
  │   Minimum Version:    16641
  │   Update State:       Success
  │   Update Error:       No vendor ID set
  │   GUIDs:              7c3cf919-5658-5a5e-a94c-e8fb60f98afd
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • System requires external power source
  │                       • Needs a reboot after installation
  │                       • Device is usable for the duration of the update
  │                       • Updatable
  │ 
  └─UEFI dbx:
        Device ID:        362301da643102b9f38477387e2193e57abaa590
        Summary:          UEFI revocation database
        Current version:  217
        Minimum Version:  217
        Vendor:           UEFI:Linux Foundation
        Install Duration: 1 second
        GUIDs:            fda6234b-adcb-5105-8515-9af647d29775 ← UEFI\CRT_D7F66BE77CEF858C174BF4338A99263C8795B74E02026411F5F532F716AE3263
                          f8ff0d50-c757-5dc3-951a-39d86e16f419 ← UEFI\CRT_D7F66BE77CEF858C174BF4338A99263C8795B74E02026411F5F532F716AE3263&ARCH_X64
                          c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
                          7d5759e5-9aa0-5f0c-abd6-7439bb11b9f6 ← UEFI\CRT_ED1FE72CB9CA31C9AF5B757AFCD733323D675825032E6CED7FE1AE9EB767998C
                          0c7691e1-b6f2-5d71-bc9c-aabee364c916 ← UEFI\CRT_ED1FE72CB9CA31C9AF5B757AFCD733323D675825032E6CED7FE1AE9EB767998C&ARCH_X64
        Device Flags:     • Internal device
                          • Updatable
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Only version upgrades are allowed
                          • Signed Payload

```

</details>

**Additional questions**

- Operating system and version: Fedora kde 37
- Have you tried rebooting? yes
- Is this a regression? no

hughsie, 2023-02-04:
What hardware is this? Can you upload the /sys/firmware/dmi/tables/DMI file please.

jimmyd-be, 2023-02-04:
[tables.zip](https://github.com/fwupd/fwupd/files/10608584/tables.zip)
Here you have it.

hughsie, 2023-02-04:
Thanks -- can I have /sys/firmware/dmi/tables/smbios_entry_point too please.

jimmyd-be, 2023-02-04:
[tables.zip](https://github.com/fwupd/fwupd/files/10608635/tables.zip)

hughsie, 2023-02-04:
Okay; worked it out -- thanks.

closed 2023-02-04 by hughsie