fwupd/fwupd #5499: Provide some kind of error output when using `--json`

hardboprobot, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
When using `fwupdmgr --json` from a script, some commands output no info at all when they fail, making it harder for the script to tell what went wrong.  For instance, running `fwupdmgr local-install --json <fw_file> <dev_id>` might fail because the file couldn't be fetched or because it's incompatible with the specified device, but neither show any message or result other than an error code = 1.

**Describe the solution you'd like**
Either throw the same error message used for the non-json output in stdout or, ideally, output a json object describing the error.

hughsie, 2023-02-07:
@hardboprobot something like that^^ ?

hardboprobot, 2023-02-07:
Just like that! I tested it and left a comment there

closed 2023-02-08 by hughsie