fwupd/fwupd #5509: Fwupd Daemon 1.8.10 Fails On HP Elite Book 745 G6 With SMBIOS Structure Larger Than Available Data Error Message

arazilsongweaver, 2023-02-09:
**Describe the bug**
Fwupd daemon (/usr/lib/fwupd/fwupd) fails to start with the following message on my HP Elite Book 745 G6: Failed to load daemon: failed to load engine: Failed to load SMBIOS: structure larger than available data

This failure originated with version fwupd-1.8.10-1 from Arch Linux. The previous version from Arch Linux, fwupd-1.8.9-2, does not have this issue and works normally. Rolling back to the previous version (1.8.9-2) restores full functionality to the Fwupd daemon.

This issue causes service activation timeouts when software attempts to access the Fwupd daemon ("Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)").

**Steps to Reproduce**
Run the fwupd daemon through systemd or command line. This was repeatedly reproduced by running the following command as root:  /usr/lib/fwupd/fwupd -v

**Expected behavior**
The daemon should start normally.

**fwupd version information**

- Broken version: fwupd-1.8.10-1
- Last working version: fwupd-1.8.9-2

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Arch Linux Pacman - fwupd-1.8.10-1

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
HP HP EliteBook 745 G6
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─BC511 HFM512GDJTNI-82A0A:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    HPS0
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      NJ08N495811208A0J
│     GUIDs:              2cb66de9-4194-5b20-aa7d-f81ee7426aaa ← NVME\VEN_1C5C&DEV_1339
│                         30cd5b05-96f2-56e4-b543-27ee571185f7 ← NVME\VEN_1C5C&DEV_1339&REV_00
│                         326fe559-5d4b-5c4f-942c-876757cc1c32 ← NVME\VEN_1C5C&DEV_1339&SUBSYS_1C5C0000
│                         7e203a2b-6e4f-58c1-9d1e-6d9aec827484 ← NVME\VEN_1C5C&DEV_1339&SUBSYS_1C5C0000&REV_00
│                         a0e5391e-f04c-5fc8-92d2-c6c547570d19 ← SK hynix BC511 HFM512GDJTNI-82A0A
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08108109
│ │   Vendor:             AMD
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       997104da-ec11-5e04-9d94-ac5501fbf609 ← CPUID\PRO_0&FAM_17
│ │                       38cd08d3-c885-5730-bff8-fdf71774a4a2 ← CPUID\PRO_0&FAM_17&MOD_18
│ │                       cbe5ceff-2b79-5935-9e79-ad728587b62f ← CPUID\PRO_0&FAM_17&MOD_18&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           AMD (PCI:0x1022)
│       GUIDs:            0e8dc554-a0a2-51fb-b439-1eb72b14ec38 ← PCI\VEN_1022&DEV_15DF
│                         1adad2a8-21d1-5354-ae84-b68876203f50 ← PCI\VEN_1022&DEV_15DF&REV_00
│                         bd5a5081-60b8-57cb-91d6-78e850123204 ← PCI\VEN_1022&DEV_15DF&SUBSYS_102215DF
│                         e736aecb-53ae-53f7-a688-58233bf0507b ← PCI\VEN_1022&DEV_15DF&SUBSYS_102215DF&REV_00
│       Device Flags:     • Internal device
│     
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    71936
│ │   Minimum Version:    1
│ │   Vendor:             HP (DMI:HP)
│ │   Update State:       Success
│ │   GUIDs:              62d71d47-2bc9-45fd-778a-aaf1133750d0
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─UEFI Platform Key:
│ │     Device ID:        6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│ │     Summary:          HP Inc. PK 2016 CA
│ │     Vendor:           HP Inc.
│ │     GUID:             77c6cc94-a10e-5208-916d-2a805c661547 ← UEFI\CRT_9FD5D4B39F4374A3C6CAA212D4007AB905C58591
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  83
│       Minimum Version:  83
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       Update Error:     System firmware cannot accept DBX updates
│       GUIDs:            95641a6b-bc03-5b8b-bdfc-c92dab523a06 ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678
│                         7689caf4-c147-5c67-bff9-5dbe59a441bd ← UEFI\CRT_C8B4C6D9755968B6DEB48D4768DD5E9020D3FF4ADC9A62521F54252DD4241678&ARCH_X64
│                         c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Current version:    7.85.17.51968
      Vendor:             Infineon (TPM:IFX)
      GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
                          5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
                          ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
                          6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
                          301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
      Device Flags:       • Internal device
```

**Additional questions**

- Operating system and version:
Arch Linux
- Have you tried rebooting?
Yes.
- Is this a regression?
Yes.

hughsie, 2023-02-09:
> structure larger than available data

That in interesting indeed! Can you upload your /sys/firmware/dmi/tables/DMI and /sys/firmware/dmi/tables/smbios_entry_point files please.

hughsie, 2023-02-09:
The SMBIOS blob looks fine, right until the last entry where we get `str_type=0x70, str_len=0x74, str_handle=0x6c61` -- or the string `ptal` in ASCII -- @superm1 this is a Ryzen 7 machine, does that mean anything to you?

superm1, 2023-02-09:
Didn't this just get fixed in https://github.com/fwupd/fwupd/commit/4edf2c507968d3f1e9363d4cfdafc2e40cc6d9a1?

hughsie, 2023-02-09:
Nah, related, but different problems. In this one the `structure_table_len` is the exact same size as the buffer.

closed 2023-02-10 by hughsie