fwupd/fwupd #5517: Update failure on reboot

Vallas873, 2023-02-11:
**Describe the bug**
When I try to install the latest firmware update (1.21.0 -> 1.22.0) it doesn't work, it reboot without installing. 
I've seen [this](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot) (because when I run `fwupdmgr get-updates` it returns that is a known issue, with this link), but I don't have  any"Boot Order Lock" option and secure boot is already disabled.
I change the boot sequence to put "Linux Firmware Updater" before "debian", precisely when I reboot I have :
1. The Dell logo, followed by another reboot, 2 times
2. The Dell logo, and written on top : 

> fwpd-efi 1.4
> WARNING: No updates to process, exiting in 10 seconds

3. Then Debian boot

**Steps to Reproduce**

1. `fwupdmgr refresh --force`
2. `fwupdmgr get-updates`
3. `fwupdmgr update`
4. Reboot (and debian boot as usual, no screen with progress bar)

After putting "Linux Firmware Updater" on top of boot sequence, and repeat the same operations, same results with 3 reboots, as described above.

**Expected behavior**
I expected the normal installation process of a new firmware, with the Dell logo and a progress bar after the reboot 

**fwupd version information**
Please provide the version of the daemon and client.

```shell
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-3-amd64
runtime   com.dell.libsmbios            2.4
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.10
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`apt`
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. Inspiron 7400
│
├─DLL09D9:01 04F3:3147:
│     Device ID:          dca355a1153b3f01bec66167ec895d44bd3aa5dc
│     Résumé:             Touchpad
│     Version actuelle:   0x0004
│     Version du chargeur d’amorçage:0x0001
│     Fournisseur:        ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              d31718fc-3350-5b2e-87d5-817ccca56e28 ← HIDRAW\VEN_04F3&DEV_3147
│                         4bea3e35-eda3-59f9-a21c-7611603ecc66 ← HIDRAW\VEN_04F3&DEV_3147&REV_00
│                         ffac9b02-ca1e-5e13-91f8-dbaa64a8f254 ← HIDRAW\VEN_04F3&DEV_3147&MOD_006A
│                         4296cbee-f355-5bf6-b62d-a6a286b69414 ← ELANTP\ICTYPE_10
│                         db87776f-107c-5ed5-963f-6e8f4af69b34 ← ELANTP\ICTYPE_10&MOD_006A
│                         99fdc8fe-28ef-5afa-8428-747525ce23ca ← ELANTP\ICTYPE_10&MOD_006A&DRIVER_HID
│     Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│   
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
[... 488 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-6.1.0-2-amd64
├── initrd.img-6.1.0-3-amd64
├── System.map-6.1.0-2-amd64
├── System.map-6.1.0-3-amd64
├── vmlinuz-6.1.0-2-amd64
└── vmlinuz-6.1.0-3-amd64

6 directories, 331 files
```

**Additional questions**

- Operating system and version: Debian GNU/Linux bookworm/sid
- Have you tried rebooting? Many times
- Is this a regression? Nothing change
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Not a Lenovo system, and I don't see this option

hughsie, 2023-02-11:
Does this work correctly if you turn off secure boot?

Vallas873, 2023-02-11:
I assume you wanted to say on ? Because as I said, it is already disabled. Just in case I tried again with secure boot on, it is the same result than when it's off