fwupd/fwupd #5519: "Failed to download, server response was 412" when updating Lenovo laptop

thom-at-redhat, 2023-02-12:
**Describe the bug**
When updating a Lenovo laptop, receive "Failed to download, server response was 412".

**Steps to Reproduce**
1. On Lenovo P1Gen2, updating System Firmware from 0.1.45 to 0.1.46 using "fwuptool upgrade"

**Expected behavior**
Successful firmware update

**fwupd version information**

Daemon/client:
fwupd-1.8.10-2.fc37.x86_64

```shell
fwupdmgr --version
```
Loading…                 [*                                      ]16:47:08.360 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file “/sys/class/firmware-attributes/thinklmi/attributes/SecureBoot/type”: No such file or directory)
Loading…                 [-                                      ]
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.1.9-200.fc37.x86_64

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
dnf

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version:
Fedora 37
- Have you tried rebooting?
Yes
- Is this a regression?
Appears to be.  Worked for prior version
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
Believe so
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
Believe no

hughsie, 2023-02-12:
What does 'fwupdmgr update --verbose' say?

thom-at-redhat, 2023-02-12:
While collecting the information, noticed it now is working...
will advise after reboot

thom-at-redhat, 2023-02-12:
It did work this time after failing for days.  Is there information from the logs that may help or do we mark this as CLOSED WORKSFORME?

thom-at-redhat, 2023-02-13:
the only messages that may be related:
.
Feb 12 22:42:27 <system> fwupd[72784]: 03:42:27.018 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file "/sys/class/firmware-attributes/thinklmi/attributes/SecureBoot/type": No such file or directory)
Feb 12 22:42:28 <system> fwupd[72784]: 03:42:28.605 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

hughsie, 2023-02-13:
> messages that may be related:

Both of those are harmless, but I'm still very confused by the 412 status code. How many times did it fail with the 412 out of interest? Was it just one machine, or anything that shared the same IP address?

thom-at-redhat, 2023-02-13:
It was just one machine and it failed 5 or 6 times over several days.   Originally, I had thought it was a transient error or something being updated on the server side but didn't have enough information to pursue it.

hughsie, 2023-02-13:
`fwupdtool`! Not `fwupdmgr`!  Patch incoming.

closed 2023-02-13 by hughsie

colttt, 2023-03-21:
I  still get the issue..
```
flatpak run org.freedesktop.fwupd --version
Loading…                 [-                                      ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.12
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.3
runtime   com.hughsie.libjcat           0.1.12
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.3
runtime   org.freedesktop.fwupd         1.8.10
runtime   org.kernel                    6.0.0-0.deb11.6-amd64
```
and here is the output..
<details><summary>fwupd update --verbose</summary>

```

root@imperator:~# flatpak run org.freedesktop.fwupd update --verbose
09:05:52.530 FuDebug              Verbose debugging enabled (on console 1)
09:05:52.530 FuCommon             creating path /run/lock
09:05:52.530 FuMain               locked /run/lock/fwupdtool
Loading…                 [-                                      ]09:05:52.530 FuConfig             trying to load config values from /app/etc/fwupd/daemon.conf
09:05:52.530 FuCommon             reading /app/etc/fwupd/daemon.conf with 2494 bytes
09:05:52.530 FuConfig             trying to load config values from /var/data/etc/fwupd/daemon.conf
09:05:52.530 FuConfig             using autodetected max archive size 4.3 GB
Loading…                 [ -                                     ]09:05:52.530 FuRemoteList         loading /app/share/fwupd/metainfo
09:05:52.530 XbSilo               attempting to load /var/data/cache/fwupd/metainfo.xmlb
09:05:52.530 XbSilo               file: 6821c269-942e-8644-63c5-b517379043bd, current:6821c269-942e-8644-63c5-b517379043bd, cached: (null)
09:05:52.530 XbSilo               loading silo with file contents
09:05:52.530 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/dell-esrt.conf
09:05:52.531 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs-testing.conf
09:05:52.531 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/lvfs.conf
09:05:52.531 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor-directory.conf
09:05:52.531 FuRemoteList         loading remote from /app/etc/fwupd/remotes.d/vendor.conf
09:05:52.531 FuRemoteList         ordering lvfs-testing=lvfs+1
09:05:52.531 FuRemoteList         ignoring unfound remote fwupd
Loading…                 [                                       ]09:05:52.531 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
09:05:52.531 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:05:52.531 Jcat                 reading /app/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
09:05:52.531 Jcat                 loaded 1 certificates
09:05:52.531 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
09:05:52.531 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
09:05:52.531 Jcat                 reading /app/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
09:05:52.531 Jcat                 loaded 1 certificates
09:05:52.531 Jcat                 reading /var/data/lib/fwupd/pki/secret.key with 2455 bytes
09:05:52.531 Jcat                 reading /var/data/lib/fwupd/pki/client.pem with 1383 bytes
09:05:52.531 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]09:05:52.531 FuHistory            trying to open database '/var/data/lib/fwupd/pending.db'
09:05:52.531 FuHistory            got schema version of 8
Loading…                 [*                                      ]09:05:52.534 FuPlugin             load(/app/lib/fwupd-1.8.10/libfu_plugin_modem_manager.so)
09:05:52.535 FuContext            added udev subsystem watch of platform
09:05:52.535 FuContext            added udev subsystem watch of block
09:05:52.535 FuContext            added udev subsystem watch of pci
09:05:52.535 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
09:05:52.535 FuPluginDell         Using libsmbios 2.4
09:05:52.535 FuContext            added udev subsystem watch of i2c
09:05:52.535 FuContext            added udev subsystem watch of i2c-dev
09:05:52.535 FuContext            added udev subsystem watch of hidraw
09:05:52.535 FuContext            added udev subsystem watch of gpio
09:05:52.535 FuContext            added udev subsystem watch of mei
09:05:52.535 FuContext            added udev subsystem watch of iommu
09:05:52.535 FuContext            added udev subsystem watch of video4linux
09:05:52.535 FuContext            added udev subsystem watch of msr
09:05:52.535 FuContext            added udev subsystem watch of mtd
09:05:52.535 FuContext            added udev subsystem watch of nvme
09:05:52.536 FuContext            added udev subsystem watch of drm
09:05:52.536 FuContext            added udev subsystem watch of drm_dp_aux_dev
[... 1711 lines cut ...]
09:06:06.286 Fwupd                download progress: 100%
09:06:06.286 Fwupd                Emitting ::status-changed() [idle]
09:06:06.286 Fwupd                status-code was 412
Downloading…             [***************************************]
Failed to download, server response was 412
[... cut ...]
09:06:06.317 FuEngine             ignoring 11130101 < 11170101
 • WDC PC SN730 SDBQNTY-512G-1001
09:06:06.317 FuMain               current version is 11170101: 11170101=same, 11130101=older
09:06:06.320 FuPlugin             finalize(modem_manager)
09:06:06.320 FuPluginTest         destroy
09:06:06.323 FuProgress           raw timing data was { 0.074, 0.001, 0.018, 0.001, 0.034, 0.001 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 58 0 14 1 27 1 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:906
```
</details>

hughsie, 2023-03-21:
Guess who forgot to update the flatpak...