fwupd/fwupd #5520: Quectel EM120R-GL firmware upgrade now can not pass "restarting device"

ericqzou, 2023-02-12:
**Describe the bug**
fwupdmgr upgrade Quectel EM120R-GL modem can not pass "Restaring device" stage and reports "suitable port not found"

```
[journal.txt](https://github.com/fwupd/fwupd/files/10717473/journal.txt)
[journal.txt](https://github.com/fwupd/fwupd/files/10717476/journal.txt)

Devices with no available firmware updates: 
 • Integrated Camera
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM120R GL from EM120RGLAPR02A07M4G_10.010.10.010 to                  ║
║ EM120RGLAPR02A07M4G_20.020.20.020?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release :                                                        ║
║                                                                              ║
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Performance improvements                                                   ║
║                                                                              ║
║ EM120R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Restarting device…       [                                       ]
suitable port not found
```
**Steps to Reproduce**
$fwupdmgr upgrade

**Expected behavior**
success

**fwupd version information**
runtime   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    6.1.11
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.gusb          0.3.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt, ubuntu packages

</details>

**Additional questions**

- Lenovo T14 Gen 2a
- Kernel 6.1.11
- ModemManager 1.19.1

I am not sure if this is related to #5428 but in my journal there is an assert that failed again.

"fu_device_retry_full: assertion 'error == NULL || *error == NULL' failed"

ericqzou, 2023-02-12:
[journal.txt](https://github.com/fwupd/fwupd/files/10717491/journal.txt)

superm1, 2023-02-13:
@ivanmikh I guess this is because that fix from 5428 isn't landed in the version used here, right?

ivanmikh, 2023-02-13:
> @ivanmikh I guess this is because that fix from 5428 isn't landed in the version used here, right?

no, this one looks a bit different.
i don't have the setup right now, but I asked quectel engineer to try to reproduce it. will look into it

ericqzou, 2023-02-26:
Hi @ivanmikh 

Is there anything new from Quectel?

Sincerely
Eric

ivanmikh, 2023-02-27:
Hi @ericqzou 

Yes, they were able to reproduce the issue, but it seems to be tied to the `EM120RGLAPR02A07M4G_20.020.20.020`. Other version do not have this issue. Now they are looking into what might be the problem on the firmware side

WildCard1, 2023-03-16:
I experience exactly the same error.

WildCard1, 2023-03-17:
Suddenly it worked:

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM120R GL from EM120RGLAPR02A07M4G_16.016.16.016 to                  ║
║ EM120RGLAPR02A07M4G_20.020.20.020?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release :                                                        ║
║                                                                              ║
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Performance improvements                                                   ║
║                                                                              ║
║ EM120R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Waiting…                 [***************************************] Less than one minute remaining…       [*******************                    ] 1 minute remaining…
Waiting…                 [***************************************]
Successfully installed firmware / / /
```

ericqzou, 2023-03-17:
Thanks @WildCard1 

Very nice you put the feedback here, I will try it when I get home. My guess is that the firmware team figured out the problem, fixed it, and just pull the old package and put a new one online.

lnagel, 2023-03-17:
```
root@LNT480s:~# fwupdmgr upgrade
Devices with no available firmware updates: 
 • SSD 970 EVO Plus 2TB
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade EM120R GL from EM120RGLAPR02A07M4G_10.010.10.010 to                  ║
║ EM120RGLAPR02A07M4G_20.020.20.020?                                           ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release :                                                        ║
║                                                                              ║
║ Firmware release EM120RGLAPR02A07M4G_20.020.20.020 packaged for LVFS.        ║
║                                                                              ║
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Performance improvements                                                   ║
║                                                                              ║
║ EM120R GL and all connected devices may not be usable while updating.        ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Restarting device…       [                                       ]
suitable port not found
```

ericqzou, 2023-03-17:
@ivanmikh  just for feedback:

Unfortunately neither is the case for me.

Suitable Port not found problem is still present.

After a reboot, the firmware is still 10, not 20, and fwupdmgr says the firmware is not updated correctly (as all other times), and asked if I want to same the report (which I did many times too).

For the record, I never has a sim card in the slot. I don't use it. I turned on modem just for fw upgrade purpose..

Fryie, 2023-04-24:
Unfortunately, I have the same issue

lnagel, 2023-04-27:
The upgrade worked for me on Ubuntu 23.04 on the first attempt. Also the modem seems much more stable now.

Fryie, 2023-05-03:
Upgrading to the new Ubuntu version also fixed the problem for me.