fwupd/fwupd #5521: Add async variant of fwupd_client_download_file()

pwithnall, 2023-02-13:
There’s not currently an async version of fwupd_client_download_file() in the API. It would be useful to have one, so we can use it in gnome-software.

The implementation of it could use `g_output_stream_splice_async()` to avoid having to heap allocate a single buffer to contain the entire downloaded firmware (which I’m assuming might be tens of megabytes in size?)

hughsie, 2023-02-13:
You're correct. I see a `fwupd_client_download_bytes_async()` which I guess you could use in gnome-software -- that's existed since 1.5.0 (Oct 2020) which might be a more palatable thing than adding `fwupd_client_download_file_async()` and depending on git main branch? I'm happy to add the new API if it makes your life easier.

>  avoid having to heap allocate a single buffer to contain the entire downloaded firmware

I'm not so worried about that; we need to do that anyway when we pass the function to the firmware plugin install func.

pwithnall, 2023-02-13:
Yeah, I’m [now using `fwupd_client_download_bytes_async()`](https://gitlab.gnome.org/GNOME/gnome-software/-/merge_requests/1616) and then `g_file_replace_contents_bytes_async()`.

pwithnall, 2023-02-13:
OK then!

closed 2023-02-13 by pwithnall