fwupd/fwupd #5522: Support progress reporting for multiple devices, or expose current device ID as a property

pwithnall, 2023-02-13:
Currently, it’s possible to call `fwupd_client_install_async()` multiple times in parallel to kick off multiple firmware updates in parallel. (Whether this results in the firmware updates actually happening in parallel, or whether the daemon serialises them is irrelevant for this issue.)

Progress reporting is done via the `FwupdClient:percentage` and `FwupdClient:status` properties. There is currently no way to relate these back to the device which is currently being updated: the caller has to track this themselves. That’s straightforward when only one `fwupd_client_install_async()` call is in flight at once, but impossible when there are multiple ones in flight.

It would be useful if either:
 1. The `FwupdClient:percentage` and `FwupdClient:status` properties were deprecated, and replaced with a `FwupdProgress` object passed in to the `fwupd_client_install_async()` call, which the caller could then receive per-operation signals on.
 2. Or, a `FwupdClient:current-device-id` property was added, which exposes the ID of the device currently being updated, which is the one the `percentage` and `status` properties currently relate to.

Either of these would be useful for gnome-software.

If I’m holding the fwupd API incorrectly, please say!

hughsie, 2023-02-13:
> Currently, it’s possible to call fwupd_client_install_async() multiple times in parallel to kick off multiple firmware updates in parallel

Right; in this case we'd serialize them -- we'd never attempt to do more than one at a time - it's scary enough to do in order!

> Or, a FwupdClient:current-device-id property was added, which exposes the ID of the device currently being updated, which is the one the percentage and status properties currently relate to.

I think this is the least about of new API; @superm1 any concerns? I'm not sure what to do if the device updating actually updates a few sub-components, e.g. on a dock. I think it makes sense to say "what percentage is referring to" so it would always point to the "thing we asked to install" rather than the things we're operating on -- which I think is fine from a "global progressbar" point of view.

The other thing we could do is have `FwupdDevice.percentage` property that gets updated on device-changed(). It would do the right thing in the event of a composite update -- we already have the `status` property there.

hughsie, 2023-09-23:
> is have FwupdDevice.percentage property

We did that :)

closed 2023-09-23 by hughsie