fwupd/fwupd #5525: Logitech Unifying Receiver (Lightspeed devices) not listed

paulmenzel, 2023-02-13:
Factored out from issue https://github.com/fwupd/fwupd/issues/4648.

     $ lsusb -d 046d:c53f
    Bus 001 Device 005: ID 046d:c53f Logitech, Inc. USB Receiver

    $ lsusb -d 046d:c52b
    Bus 002 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver

> I'm afraid the Lightspeed devices don't have a firmware update capability (at least in fwupd, no idea about Windows) -- We only support the Unifying and Bolt devices I'm afraid.

>> Would it be possible to list the device in fwupdmgr get-devices and denote somehow, that it’s unsupported
>
> Sure, we could certainly enumerate them and show that as not UPDATABLE. Do you know of the method solaar uses to get the version numbers of Lightspeed devices? Maybe it's small tweak to the existing code? i.e. add it to `logitech_hidpp.quirk` and see what `fwupdtool --plugins logitech_hidpp get-devices --verbose` says.

superm1, 2023-02-13:
Can we please also see `sudo fwupdtool get-devices --verbose --plugins logitech_hidpp` output says.

hughsie, 2023-02-13:
With the VID PID added to the quirk file.

paulmenzel, 2023-02-13:
How can I do that in Debian sid/unstable? Do I need to build it manually? In *fwupd* 1.8.10-2 I only found `/usr/share/fwupd/quirks.d/builtin.quirk.gz`.

superm1, 2023-02-13:
> then restart fwupd

(that's for `fwupd` daemon + `fwupdmgr` client; with `fwupdtool` no need to do this)

superm1, 2023-02-13:
This one is also already supported in the quirk file: https://github.com/fwupd/fwupd/blob/main/plugins/logitech-hidpp/logitech-hidpp.quirk#L2

superm1, 2023-02-13:
If you delete your `test.quirk` does it stop showing up?

paulmenzel, 2023-02-13:
> I don't think this is a Lightspeed receiver. It's a Unifying according to https://support.logi.com/hc/en-us/articles/360023462213-K270-Technical-Specifications

Sorry for not seeing the difference in the lsusb output. I updated the title and description.

> If you delete your test.quirk does it stop showing up?

Yes, without that, it’s not showing up.

superm1, 2023-02-13:
Is `builtin.quirk.gz` missing in the fwupd package perhaps?

paulmenzel, 2023-02-13:
Hmm, I had to restart the system to reproduce the issue.

First it’s not listed.

```
$ fwupdmgr get-devices
WARNUNG: Aktualisierung der UEFI-Kapsel nicht verfügbar oder in der Firmware-Einrichtung aktiviert
  Siehe https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported für weitere Informationen.
Dell Inc. Latitude E7250
│
├─AMT [unprovisioned]:
│     Gerätekennung:      8d5470e73fd9a31eaa460b2b6aea95483fe3f14c
│     Zusammenfassung:    Hardware and firmware technology for remote out-of-band management
│     Aktuelle Version:   10.0.56.3002
│     Bootloader-Version: 10.0.56.3002
│     Anbieter:           Intel Corporation (MEI:0x8086)
│     GUIDs:              12f80028-b4b7-4b2d-aca8-46e0ff65814c
│                         4e4029a8-fe0b-5e35-99a7-eca3cbb8c828 ← MEI\VEN_8086&DEV_9CBA
│                         74076579-a16c-5c7f-95c0-a0dc583ed192 ← MEI\VEN_8086&DEV_9CBA&REV_03
│                         97293933-0211-5453-a256-36a549b4ad14 ← MEI\VEN_8086&DEV_9CBA&SUBSYS_1028062D
│                         45d491b0-3586-5374-932b-7d65d380a019 ← MEI\VEN_8086&DEV_9CBA&SUBSYS_1028062D&REV_03
│     Device Flags:       • Internes Gerät
│   
├─Core™ i7-5600U CPU @ 2.60GHz:
│     Gerätekennung:      4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Aktuelle Version:   0x0000002f
│     Anbieter:           Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         bcad68bd-c33a-56cd-a8a6-b9a78d62ee6e ← CPUID\PRO_0&FAM_06&MOD_3D
│                         a702f59f-10c2-5561-976c-b94c89cb42d8 ← CPUID\PRO_0&FAM_06&MOD_3D&STP_4
│     Device Flags:       • Internes Gerät
│   
├─HD Graphics 5500:
│     Gerätekennung:      5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Aktuelle Version:   09
│     Anbieter:           Intel Corporation (PCI:0x8086)
│     GUIDs:              64c0ea55-d861-5dd0-aa6b-bcc307f57ed4 ← PCI\VEN_8086&DEV_1616
│                         19e1f63e-bef0-5582-b236-364069bcd0d5 ← PCI\VEN_8086&DEV_1616&REV_09
│                         3c729b01-2419-57c8-8dda-052bab778c92 ← PCI\VEN_8086&DEV_1616&SUBSYS_1028062D
│                         5f5a0a08-69b7-5b63-91fc-c3e20c2f0b58 ← PCI\VEN_8086&DEV_1616&SUBSYS_1028062D&REV_09
│                         62eb842b-53f4-586a-ac7c-054f041e77d6 ← PCI\VEN_8086&DEV_1616&REV_00
│                         975510f1-65b5-5107-83d8-d46432f273b7 ← PCI\VEN_8086&DEV_1616&SUBSYS_1028062D&REV_00
│     Device Flags:       • Internes Gerät
│                         • Kryptografische Hash-Verifizierung ist verfügbar
│   
├─SSD PM851 mSATA 256GB:
│     Gerätekennung:      0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
│     Zusammenfassung:    ATA drive
│     Aktuelle Version:   EXT4AD0Q
│     Anbieter:           Samsung (ATA:0x144D, OUI:002538)
│     Seriennummer:       S1EVNSAG104351
│     GUIDs:              16edd6b3-f18c-5eee-be5a-e7cb89d592f2 ← IDE\SAMSUNG_SSD_PM851_mSATA_256GB___________EXT4AD0Q
│                         cae6f248-5571-598e-8d17-226faea2805e ← IDE\0SAMSUNG_SSD_PM851_mSATA_256GB___________
│                         3be83754-dd2c-556e-8c44-2fd08f362ade ← SAMSUNG SSD PM851 mSATA 256GB
│     Device Flags:       • Internes Gerät
│                         • Aktualisierbar
│                         • System benötigt externe Stromquelle
│                         • Benötigt einen Neustart nach der Installation
│                         • Das Gerät ist während der Dauer der Aktualisierung nutzbar
│   
└─UEFI dbx:
      Gerätekennung:      362301da643102b9f38477387e2193e57abaa590
      Zusammenfassung:    UEFI revocation database
      Aktuelle Version:   217
      Minimale Version:   217
      Anbieter:           UEFI:Linux Foundation
      Installationsdauer: 1 Sekunde
      GUIDs:              c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
[... 240 lines cut ...]
│     Anbieter:           UEFI:Linux Foundation
│     Installationsdauer: 1 Sekunde
│     GUIDs:              c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│     Device Flags:       • Internes Gerät
│                         • Aktualisierbar
│                         • Unterstützt auf dem entfernten Server
│                         • Benötigt einen Neustart nach der Installation
│                         • Das Gerät ist während der Dauer der Aktualisierung nutzbar
│                         • Nur Versionsaktualisierungen sind erlaubt
│                         • Signierte Nutzdaten
│   
└─Unifying Receiver:
      Gerätekennung:      53d907a89b8cf6db282e318dcb99ad85850fb789
      Zusammenfassung:    Miniaturised USB wireless receiver
      Aktuelle Version:   RQR12.11_B0032
      Bootloader-Version: BOT01.04_B0016
      Anbieter:           HIDRAW:0x046D|USB:0x046D
      Installationsdauer: 30 Sekunden
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Aktualisierbar
                          • Signierte Nutzdaten
```

superm1, 2023-02-13:
Can you please add `--verbose` into the systemd unit for `fwupd.service` and then reboot and share the journal output after it reproduces then?

paulmenzel, 2023-02-13:
I am sorry, some more details. During debugging, more details emerged. The Logitech device is listed, if it’s already plugged in during boot or when being plugged in GDM (and then log in to check). I can only reproduce the problem, when plugging the device with my user session being active.

superm1, 2023-02-13:
> Feb 13 16:50:01 e7250 fwupd[2113]: 15:50:01.219 FuEngine             disabling plugin because: failed to startup using logitech_hidpp: no kernel support for CONFIG_HIDRAW

That makes perfect sense now with your description.  Basically if not plugged in you don't have `/sys/class/hidraw` and we disable the plugin.  That's not sensible, and we should change it.

hughsie, 2023-02-13:
> Basically if not plugged in you don't have /sys/class/hidraw and we disable the plugin

@superm1 can we just drop the check for CONFIG_HIDRAW? I think that might be the least magic. @paulmenzel do you want to give a PR a go please?

superm1, 2023-02-13:
Thanks for tracking this down btw.  It's been broken a long time, but since people usually leave their dongles plugged in all the time or having something else that provides `/sys/class/hidraw`, no one saw it for a while.

closed 2023-02-13 by superm1

paulmenzel, 2023-02-13:
No, thank @hughsie and you, for your great support. I wish, Logitech would have done it.

commit 383ae945 (2023-02-13): trivial: logitech-hidpp: move the check for CONFIG_HIDRAW into probe

The kernel should have loaded hidraw support by the time that we probe
a device.  This fixes logitech devices not showing up after startup
if they weren't plugged in initially.

Fixes: #5525
commit ea7b2757 (2023-02-13): trivial: logitech-hidpp: move the check for CONFIG_HIDRAW into probe

The kernel should have loaded hidraw support by the time that we probe
a device.  This fixes logitech devices not showing up after startup
if they weren't plugged in initially.

Fixes: #5525
PR #5527 (2023-02-13): trivial: logitech-hidpp: move the check for CONFIG_HIDRAW into probe
The kernel should have loaded hidraw support by the time that we probe
a device.  This fixes logitech devices not showing up after startup
if they weren't plugged in initially.

Fixes: #5525

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 54441317 (2023-02-13): trivial: logitech-hidpp: move the check for CONFIG_HIDRAW into probe

The kernel should have loaded hidraw support by the time that we probe
a device.  This fixes logitech devices not showing up after startup
if they weren't plugged in initially.

Fixes: #5525