fwupd/fwupd #5530: Can't update/downgrade firmware in community firmware in PinePhone / Pro's modem anymore

Biktorgj, 2023-02-14:
**Describe the bug**
There seems to be a regression in recent fwupd versions affecting the PinePhone and PinePhone Pro devices (at least). Community firmware for the Quectel EG25-G inside these phones can't find any firmware _at all_ in an up to date distro (not even using the embargoed remote.

When using an older version of fwupdmgr, the updates do appear and everything seems to work. After some digging, it seems like the current fwupdmgr (as shipped in Mobian at least) can't correctly identify the manufacturer and product, as `fwupdtool hwids` repeatedly outputs `not available as 'Manufacturer' unknown` and `fwupdtool export-hwids filename` returns `g_key_file_set_string: assertion 'string != NULL' failed` for each line where hwids would report the `not available` error

**Steps to Reproduce**
Steps to reproduce the behavior.
1. Run `fwupdmgr get-updates` in a Pinephone
2. Alternatively, go to the Firmware GUI application and select the Quectel EG25-G mode

**Expected behavior**
The different firmware versions should appear as available for update/downgrade

**fwupd version information**
```
compile   org.freedesktop.fwupd         1.8.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1-rockchip
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Comes preinstalled with the distro, updated via `apt`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Unknown Vendor Unknown Product
?
??Unknown Device:
?     Device ID:          37ffb960cf01d183b5e2fdc36ebc3b599caf8014
?     GUID:               9d052585-63c3-5fee-b009-97faa715eb68 ? GPIO\ID_gpio0
?   
??Unknown Device:
?     Device ID:          b0609ed13f23900c32cb5cf13c031721cea7a02b
?     GUID:               1e2879be-5f17-57bc-ac71-6442cadfbec0 ? GPIO\ID_gpio1
?   
??Unknown Device:
?     Device ID:          4389cc5774949d3aef78e64fc72430a0601d9390
?     GUID:               0722edb7-de76-574a-b826-abff44261ce5 ? GPIO\ID_gpio2
?   
??Unknown Device:
?     Device ID:          8fd4c139e223699b0c2d001c96b43a2e027fbe01
?     GUID:               6dcb93b1-3f1c-5af7-b278-87a3c898d859 ? GPIO\ID_gpio3
?   
??Unknown Device:
?     Device ID:          6c121d22aaad6290713b9b7aead3127b004b0b26
?     GUID:               d776fae8-f6c0-52c9-aa5a-f60913db5069 ? GPIO\ID_gpio4
?   
??A3A442:
?     Device ID:          46dad0f891e0142d68b298b33140ecd4555e7d75
?     Current version:    0x1000000000000000
?     Vendor:             EMMC:0x0000d6
?     Serial Number:      0xadbb9523
?     GUIDs:              6c528c16-67bd-5f93-9859-db787c5e10f9 ? EMMC\NAME_A3A442
?                         34685747-6e52-591b-b6bf-9f0d7bc2f9af ? EMMC\NAME_A3A442&REV_0x1000000000000000
?                         f0f9ca5c-cde7-5e61-9d9a-d39deed578d2 ? EMMC\MAN_00D6&OEM_0103
?                         42e489fc-0597-528f-b2f2-c05a8d90c10c ? EMMC\MAN_00D6&OEM_0103&NAME_A3A442
?                         7060db51-5f9d-5f53-a268-1cda05556454 ? EMMC\MAN_00D6&NAME_A3A442&REV_0x1000000000000000
?                         173e3fcb-036b-55a4-9145-6914d4c2e940 ? EMMC\MAN_00D6&OEM_0103&NAME_A3A442&REV_0x1000000000000000
?                         4ad1c540-521f-5fc8-86a0-b77fdc0ce206 ? EMMC\0214&0259&A3A442
?     Device Flags:       ? Internal device
?                         ? Updatable
?   
??QUECTEL Mobile Broadband Module:
?     Device ID:          a479a7a712f900f527bc80df33034bc7df0a7a32
?     Summary:            Quectel EG25-G modem
?     Current version:    0.7.3
?     Vendor:             Qualcomm (USB:0x2C7C)
?     Release Branch:     community
?     GUIDs:              db379a33-254f-5140-b37e-d36ae7e5c039
?                         d18f31f1-a3fa-55a2-b4ed-decfbc1e004d ? USB\VID_2C7C&PID_0125&REV_0318&NAME_EG25GGB
?                         1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ? USB\VID_2C7C&PID_0125&REV_0318
?                         587bf468-6859-5522-93a7-6cce552a0aa3 ? USB\VID_2C7C&PID_0125
?                         22ae45db-f68e-5c55-9c02-4557dca238ec ? USB\VID_2C7C
?     Device Flags:       ? Updatable
?                         ? System requires external power source
?                         ? Unsigned Payload
?   
??spi0.0:
      Device ID:          c483d3f8b5d38f125bdc73f32d990dbea9c32768
      Summary:            Memory Technology Device
      Update Error:       No vendor ID set
      GUID:               9101d2aa-f9e7-5b49-8720-16fc71fe9e69 ? MTD\NAME_spi0.0
      Device Flags:       ? Internal device
                          ? Needs a reboot after installation
                          ? Cryptographic hash verification is available
                          ? Updatable
    
```

</details>

**Additional questions**

- Operating system and version: `Mobian/bookworm`
- Have you tried rebooting? `Yes`
- Is this a regression? `Yes`

For completeness, here's the output from various commands:
1. fwupdmgr get-updates:
```
 fwupdmgr get-updates
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Updating pine64_community_embargo
Downloading?             [***************************************]
Updating lvfs
Downloading?             [***************************************]
Downloading?             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
Devices with no available firmware updates: 
 ? spi0.0
 ? A3A442
 ? QUECTEL Mobile Broadband Module
No updatable devices
```
2.  fwupdmgr get-updates a479a7a712f900f527bc80df33034bc7df0a7a32
```
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Updating pine64_community_embargo
Downloading?             [                          |            ]Updating lvfs
Downloading?             [***************************************]
Downloading?             [***************************************]
Downloading?             [***************************************]
Successfully downloaded new metadata: 0 local devices supported
Devices with no available firmware updates: 
 ? QUECTEL Mobile Broadband Module
No updatable devices
```
3. fwupdtool hwids
```
Computer Information
--------------------

Hardware IDs
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
```
4. fwupdtool export-hwids out.log
```
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
04:46:08.128 GLib                 g_key_file_set_string: assertion 'string != NULL' failed
```

Biktorgj, 2023-02-14:
Hi @hughsie!

This is what I get (hope I did it right)
```
./fwupdtool --version
Loading?                 [-                                      ]
WARNING: This package has not been validated, it may not work properly.
runtime   org.freedesktop.fwupd         1.8.11
runtime   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1-rockchip
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.11
runtime   org.freedesktop.gusb          0.3.10
```
```
./fwupdtool hwids --verbose
09:23:03.406 FuDebug              Verbose debugging enabled (on console 1)
Loading?                 [-                                      ]09:23:03.409 GLib-GIO             09:23:03.409 FuContext            loading HwId overrides from /usr/local/etc/fwupd/daemon.conf
Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
09:23:03.409 FuContext            not loading HwId overrides from /var/local/lib/fwupd/daemon.conf
09:23:03.410 FuContext            failed to load smbios: SMBIOS tables not found at /sys/firmware/dmi/tables
Loading?                 [                                       ]09:23:03.410 FuHwids              HardwareID-0 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-1 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-2 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-3 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-4 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-5 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-6 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-7 is not available, not available as 'Manufacturer' unknown
09:23:03.410 FuHwids              HardwareID-8 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-9 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-10 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-11 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-12 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-13 is not available, not available as 'Manufacturer' unknown
09:23:03.411 FuHwids              HardwareID-14 is not available, not available as 'Manufacturer' unknown
Loading?                 [*                                      ]09:23:03.411 FuContext            added udev subsystem watch of firmware-attributes
09:23:03.411 FuContext            Error opening directory ?/sys/class/firmware-attributes?: No such file or directory
09:23:03.411 FuProgress           raw timing data was { 0.001, 0.001, 0.000, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 34 51 3 11 ] at ../libfwupdplugin/fu-context.c:810
Computer Information
--------------------

Hardware IDs
------------
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
not available as 'Manufacturer' unknown
09:23:03.412 FuProgress           raw timing data was { 0.001, 0.001, 0.000, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 34 51 3 11 ] at ../libfwupdplugin/fu-context.c:810
```

hughsie, 2023-02-14:
Thanks. I might have broken something with the fdt context rework. Can you attach the /sys/firmware/fdt file please.

Biktorgj, 2023-02-14:
Sure thing:
[fdt_pinephone_pro.zip](https://github.com/fwupd/fwupd/files/10730846/fdt_pinephone_pro.zip)
Do you need the one on the OG Pinephone too?

hughsie, 2023-02-14:
nah, I've got it -- patch incoming.

closed 2023-02-14 by hughsie