fwupd/fwupd #5533: Not able to detect AVer CAM520 pro2 through fwupdmgr get-device

kamasw, 2023-02-15:
**Describe the question**
I have installed fwupd package from scratch in buildroot .
when check the device list which support firmware upgrade , I wont getting the list of Jabra Speakers of version 510/710/810 which is connected . Only getting 1 device list which is of multimedia card category

root@milhouse ~ $ fwupdmgr get-devices --json
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
WARNING: This package has not been validated, it may not work properly.
{
"Devices" : [
{
"Name" : "DG4008",
"DeviceId" : "5a769cf95c25c868bed01310839202749aafbe0c",
"InstanceIds" : [
"EMMC\NAME_DG4008",
"EMMC\NAME_DG4008&REV_0x3733313033353137",
"EMMC\MAN_0045&OEM_0100",
"EMMC\MAN_0045&OEM_0100&NAME_DG4008",
"EMMC\MAN_0045&NAME_DG4008&REV_0x3733313033353137",
"EMMC\MAN_0045&OEM_0100&NAME_DG4008&REV_0x3733313033353137"
],
"Guid" : [
"d96b9ab7-ad8f-5a47-a32a-867ddb935935",
"32c50c3d-3ad1-5334-b6da-2f17aec312d2",
"5a02344d-585b-507b-ad59-19e96c03ca46",
"423c0bc8-443b-5a53-93a8-27c621e790a4",
"366e9264-71ef-55ae-90cc-c6122dbcb4b3",
"835d06b7-8988-5737-9468-704978501dfa"
],
"Serial" : "0xef7a2312",
"Plugin" : "emmc",
"Protocol" : "org.jedec.mmc",
"Flags" : [
"internal",
"updatable",
"registered"
],
"VendorId" : "EMMC:0x000045",
"Version" : "0x3733313033353137",
"VersionFormat" : "number",
"Icons" : [
"media-memory"
],
"Created" : 1672983380
}
]
}

**fwupd version information**
root@ClickShare-1862300249 ~ $ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.7
runtime   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.gusb          0.4.3
runtime   org.kernel                    4.19.172
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.7
runtime   org.freedesktop.gusb          0.4.3

I have checked the dfu.quirk file found below Aver CAM entry 

# AVer CAM520 Pro2
[USB\VID_2574&PID_0A30]
Plugin = dfu
Flags = detach-for-attach
RemoveDelay = 180000

Let me know if I am missing any thing while compiling fwupd . My assumption is that it should also show Aver CAM device if connected .

hughsie, 2023-02-15:
> checked the dfu.quirk file found

Did you `ninja install` the modified quirk file? What does lsusb show for this device? What does `fwupdtool --plugins dfu get-devices --verbose` say with the device plugged in?

kamasw, 2023-02-16:
Hi @hughsie 

**lsusb command output for Aver CAM520 Pro2**
Bus 001 Device 016: ID 2574:0a30 AVer Information, Inc. AVer CAM520 Pro2
Bus 001 Device 015: ID 2574:1003 AVer Information, Inc. CAM520 PRO2

root@ClickShare-1862300249 ~ $ **fwupdtool --plugins dfu get-devices --verbose**
03:21:47.465 FuDebug              Verbose debugging enabled (on console 1)
03:21:47.466 FuMain               locked /run/lock/fwupdtool
Loading?                 [-                                      ]03:21:47.466 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
03:21:47.466 FuCommon             reading /etc/fwupd/daemon.conf with 2224 bytes
03:21:47.466 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
03:21:47.467 FuConfig             using autodetected max archive size 513.8?MB
03:21:47.467 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
Loading?                 [ -                                     ]03:21:47.467 FuRemoteList         loading /usr/share/fwupd/metainfo
03:21:47.468 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
03:21:47.468 XbSilo               file: b3bd409c-29fa-273c-0e14-c45f708aa06d, current:b3bd409c-29fa-273c-0e14-c45f708aa06d, cached: (null)
03:21:47.468 XbSilo               loading silo with file contents
03:21:47.468 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
03:21:47.468 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
03:21:47.469 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
03:21:47.469 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
03:21:47.470 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
03:21:47.470 FuRemoteList         ordering lvfs-testing=lvfs+1
03:21:47.470 FuRemoteList         ignoring unfound remote fwupd
Loading?                 [                                       ]03:21:47.470 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
03:21:47.471 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
03:21:47.471 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
03:21:47.471 Jcat                 loaded 1 certificates
03:21:47.471 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
03:21:47.471 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
03:21:47.471 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
03:21:47.471 Jcat                 loaded 1 certificates
03:21:47.471 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
03:21:47.472 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
03:21:47.472 FuEngine             client certificate now exists: nothing to do
Loading?                 [                                       ]03:21:47.472 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
03:21:47.473 FuHistory            got schema version of 8
Loading?                 [*                                      ]03:21:47.474 FuContext            added udev subsystem watch of block
03:21:47.474 FuContext            added udev subsystem watch of pci
03:21:47.475 FuContext            added udev subsystem watch of i2c
03:21:47.475 FuContext            added udev subsystem watch of i2c-dev
03:21:47.475 FuContext            added udev subsystem watch of hidraw
03:21:47.475 FuContext            added udev subsystem watch of mei
03:21:47.476 FuContext            added udev subsystem watch of mtd
03:21:47.476 FuContext            added udev subsystem watch of nvme
03:21:47.481 FuContext            added udev subsystem watch of drm
03:21:47.482 FuContext            added udev subsystem watch of drm_dp_aux_dev
03:21:47.482 FuContext            added udev subsystem watch of serio
03:21:47.482 FuPluginTest         init
03:21:47.483 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
03:21:47.483 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
03:21:47.483 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
03:21:47.483 XbSilo               file: fa0dd588-5450-45fe-b813-085fb962e328, current:fa0dd588-5450-45fe-b813-085fb962e328, cached: (null)
03:21:47.483 XbSilo               loading silo with file contents
Loading?                 [*                                      ]03:21:47.484 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
03:21:47.484 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
03:21:47.484 FuHwids              smbios property Manufacturer=rockchip
03:21:47.484 FuHwids              ignoring EnclosureKind: offset bigger than size 0
03:21:47.484 FuHwids              ignoring Family: offset bigger than size 6
03:21:47.484 FuHwids              smbios property ProductName=Rockchip RK3399 Excavator Board (Linux Opensource)
03:21:47.484 FuHwids              ignoring ProductSku: offset bigger than size 6
03:21:47.484 FuHwids              ignoring BiosVendor: offset bigger than size 0
03:21:47.484 FuHwids              ignoring BiosVersion: offset bigger than size 0
03:21:47.484 FuHwids              ignoring BiosMajorRelease: offset bigger than size 0
[... cut ...]
Loading?                 [***                                    ]03:21:47.510 FuContext            battery threshold now 10
Loading?                 [*****                                  ]03:21:47.513 FuEngine             using plugins: dfu
Loading?                 [-                                      ]
03:21:47.680 FuEngine             writing motd target /var/cache/fwupd/motd.d/85-fwupd
WARNING: This package has not been validated, it may not work properly.
No detected devices
03:21:47.681 FuMain               No detected devices

03:21:47.684 FuPluginTest         destroy

hughsie, 2023-02-16:
> Bus 001 Device 016: ID 2574:0a30 AVer Information, Inc. AVer CAM520 Pro2
> Bus 001 Device 015: ID 2574:1003 AVer Information, Inc. CAM520 PRO2

Do these have a DFU runtime interface? Can you grab the "lsusb -vvv" for either of them please.

kamasw, 2023-02-17:
Bus 001 Device 018: ID 2574:1003 AVer Information, Inc. CAM520 PRO2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x2574 AVer Information, Inc.
  idProduct          0x1003
  bcdDevice           50.10
  iManufacturer           1 AVer Information Inc.
  iProduct                2 CAM520 PRO2
  iSerial                 3 00000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Single TT
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
[... 740 lines cut ...]
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat           1 micro seconds
    bU2DevExitLat         500 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

hughsie, 2023-02-17:
@kamasw - neither of those have a USB DFU runtime interface -- so fwupd doesn't know what to do with them. The DFU runtime is an additional interface the USB IF defines so that firmware update software can switch (so called "detach") the device into DFU update mode.

How do you switch the devices right now into DFU mode? Presumably there's some vendor-specific control-transfer that reboots the device into the DFU mode? If you can share some sample code or some documentation we could add support that way.

closed 2023-09-23 by hughsie