fwupd/fwupd #5547: Flakiness with fu_progress_get_duration() unit tests

djcampello, 2023-02-17:
assertion failed (fu_progress_get_duration(progress) == 0.1f (+/- 0.05)): (0.154924 == 0.100000001)

This has happen multiple times in different builders.

From a comment on the original bug report: "These all look like we're just outside the "epsilon" of a floating point comparison. Should the epsilon be bumped a bit? (espsilon == 0.1 instead of 0.05 would cover all this.)"

hughsie, 2023-02-17:
100ms is a pretty huge epsilon; got any clues on why the timing would be so funky?

djcampello, 2023-02-17:
It seems it happens when running the unit tests virtualized via qemu as part of building other archs via cross-compiling... Expanding on the above error log:

```
ERROR:../fwupd-1.8.9/libfwupdplugin/fu-self-test.c:3534:fu_progress_func: assertion failed (fu_progress_get_duration(progress) == 0.1f (+/- 0.05)): (0.169768 == 0.100000001)
qemu: uncaught target signal 6 (Aborted) - core dumped
```

closed 2023-02-18 by hughsie