fwupd/fwupd #5548: Firmware install failed with error: org.freedesktop.fwupd.AuthFailed updating Dell WD19 in ChromeOS

djcampello, 2023-02-17:
I believe the issue might be related to how the FWUPD_DEVICE_FLAG_TRUSTED flag is applied when dealing with composite updates. If I understand the logic right it seems that the flag is not propagated to child devices and then the daemon fails the dbus call from chrome.

hughsie, 2023-02-17:
Can you paste in the daemon --verbose log please; it's entirely plausible it's because we compile out polkit on ChromeOS iirc.

djcampello, 2023-02-17:
Output is not too useful it goes from waiting-for-auth to idle without more information:

```
2023-02-08 15:47:08.086 4 fwupd[3601]: 15:47:08.086 FuMain               Emitting PropertyChanged('Status'='waiting-for-auth')
2023-02-08 15:47:08.086 4 fwupd[3601]: 15:47:08.086 FuMain               Emitting PropertyChanged('Status'='idle')

djcampello, 2023-02-17:
Also important to point out is that this happens when issuing dbus calls from chrome and not from fwupdmgr. fwupdmgr behavior is to prompt for confirmation on every component update

hughsie, 2023-02-17:
What dbus calls do you do? It should be enough to just call one install with the same ID as the composite ID, i.e. you don't need x dbus calls for x subcomponents.

djcampello, 2023-02-17:
https://source.chromium.org/chromium/chromium/src/+/main:chromeos/ash/components/dbus/fwupd/fwupd_client.cc;l=369 is where the error is being thrown

Here is the method that created the Install dbus call: https://source.chromium.org/chromium/chromium/src/+/main:chromeos/ash/components/dbus/fwupd/fwupd_client.cc;l=139;bpv=1;bpt=1

And this is somewhat higher in the stack where arguments are collected: https://source.chromium.org/chromium/chromium/src/+/main:chromeos/ash/components/fwupd/firmware_update_manager.cc;l=442

hughsie, 2023-02-20:
So, if I understand correctly:

 * chromeOS is compiling without polkit
 * chrome-the-UI is running as the normal user, not euid=0
 
if I understand the code correctly, I don't see how this works for any device, let alone for composite devices. Don't you get "Failed to obtain auth as not trusted user" for all update actions?

`FWUPD_DEVICE_FLAG_TRUSTED` is being (mis?)used by `FuEngineRequest` for the case where point-to-point D-Bus is used, or when the *caller*'s UID is in the `/etc/fwupd/daemon.conf` default-0 `TrustedUids` entry.

I'm guessing you must have some kind of patch which allows all kinds of authentication, and instead you're hard-relying on the `OnlyTrusted=true` behaviour, i.e. that you can only install signed updates.

djcampello, 2023-02-22:
I think you are right a no update is working. The culprit seems to be ae8e553cb270b5592893d74434ccb354726d050e. Before then it seems that there was no check for the TRUSTED flag. I'll go ahead and add chronos UID to the list of trusted ids.

hughsie, 2023-02-22:
> I'll go ahead and add chronos UID to the list of trusted ids.

MSTM. Yell if it doesn't work.

closed 2023-03-15 by djcampello