fwupd/fwupd #5549: Release request

jayofdoom, 2023-02-18:
**Describe the question**
Would it be possible for you to cut a release? I was impacted by https://github.com/fwupd/fwupd/commit/dc6054aadf6c1846c5e0056733f5380cb2dcf219 and it seemed quite serious -- I've patched locally but I'd like it to be fixed for all Gentoo users (and preferably without a distro backport).

Thanks for the software!

-Jay

superm1, 2023-02-18:
IMO In general most people should be doing their downloads using clients like `fwupdmgr` rather than `fwupdtool`.  So hopefully it's not super widespread.

hughsie, 2023-02-18:
I was planning a release on the 23rd Feb, if that helps.

jayofdoom, 2023-02-18:
> I was planning a release on the 23rd Feb, if that helps.

That should be soon enough to prevent me from wanting to offer a backport.

and... 

> IMO In general most people should be doing their downloads using clients like fwupdmgr rather than fwupdtool. So hopefully it's not super widespread.

Makes me wonder if I've been doing it wrong. I'll re-evaluate my own workflow to see if I avoid the bug; thanks for the pointer!

closed 2023-02-18 by jayofdoom