fwupd/fwupd #5573: Cannot update FPC Fingerprint Reader Firmware on Lenovo ThinkPad E15

qwetwe, 2023-02-22:
**Stand**
- **Notebook:** ThinkPad E15 Gen 4 (21ED004XRT)
- **SecureBoot:** Disabled
- **System:** ALT Workstation K 10.1 ([alt-kworkstation-10.1-install-x86_64.iso](http://ftp.altlinux.org/pub/distributions/ALTLinux/p10/images/kworkstation/alt-kworkstation-10.1-install-x86_64.iso))
- **Kernel:** 5.15.94-un-def-alt1

**Description**
Cannot update FPC Fingerprint Reader Firmware on Lenovo ThinkPad E15 using command ```# fwupdmgr update```:
<details>

**<summary># fwupdmgr update -v</summary>**

```
# LANG=en_US.UTF-8 G_DEBUG=all G_MESSAGES_DEBUG=all fwupdmgr update -v
(fwupdmgr:15381): GLib-DEBUG: 15:24:20.983: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:15381): GLib-DEBUG: 15:24:20.983: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:15381): GLib-GIO-DEBUG: 15:24:21.000: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.000: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.002: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:15381): GLib-GIO-DEBUG: 15:24:21.004: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.010: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.010: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.010: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.010: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:15381): dconf-DEBUG: 15:24:21.010: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:21.011: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • Integrated RGB Camera
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • USB2.0 Hub
Devices with the latest available firmware version:
 • Micron MTFDKCD256TFK
(fwupdmgr:15381): FuMain-DEBUG: 15:24:21.034: current version is 7003V5LN: 7003V5LN=same
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Sensor Controller L:0001 FW:16.26.23.14 from 16.26.23.14 to          ║
║ 27.26.23.19?                                                                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Update FPC Fingerprint Reader Firmware:                                      ║
║                                                                              ║
║ • Support FPC Disum Fingerprint Reader Device(10A5:9800)                     ║
║                                                                              ║
║ Sensor Controller L:0001 FW:16.26.23.14 and all connected devices may not    ║
║ be usable while updating.                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:25.457: downloading https://fwupd.org/downloads/bbc350a290baed298fca24be8aadb9b3b78ecc806845a42ba9e796b9fc45e5d8-fpc-lenfy-moh-v27_26_23_19.cab
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:25.458: Emitting ::status-changed() [downloading]
Downloading…             [                       \               ](fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.364: download progress: 100%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.364: download progress: 100%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.364: download progress: 100%
Downloading…             [***************************************]
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.508: download progress: 1%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.508: download progress: 1%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.508: download progress: 1%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.508: download progress: 1%
Downloading…             [                                       ](fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.509: download progress: 12%
(fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.509: download progress: 12%
Downloading…             [****                                   ](fwupdmgr:15381): Fwupd-DEBUG: 15:24:26.510: download progress: 18%
[... 322 lines cut ...]
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

</details>

**Additional questions**

- **Operating system and version:** ALT Workstation K 10.1 (KDE) ([alt-kworkstation-10.1-install-x86_64.iso](http://ftp.altlinux.org/pub/distributions/ALTLinux/p10/images/kworkstation/alt-kworkstation-10.1-install-x86_64.iso))
- - Operating System: ALT 10.1
- - KDE Plasma Version: 5.26.5
- - KDE Frameworks Version: 5.102.0
- - Qt Version: 5.15.7
- - Kernel Version: 5.15.72-un-def-alt1 (64-bit)
- - Graphics Platform: X11
- - Processors: 12 × AMD Ryzen 5 5625U with Radeon Graphics
- - Memory: 6.6 ГиБ of RAM
- - Graphics Processor: AMD Radeon Graphics
- - Manufacturer: LENOVO
- - Product Name: 21ED004XRT
- - System Version: ThinkPad E15 Gen 4
- **Have you tried rebooting?** Yes and more than once.
- **Is this a regression?** This is not a regression.

haoweilo, 2023-02-23:
From the log, 16.26.23.14 would be an early release version FW. It might be a compatibility issue. We will look into this.

jimzhang2, 2023-02-23:
Hello~ @qwetwe 
We have tried the latest source code of fwupd.  It seems work fine. So please install fwupd with the latest source code or make sure your fwupd version is above 1.8.7. If there is any problem, pls tell us. Thanks!

hughsie, 2023-02-23:
@jimzhang2 there were no changes to the `fpc` plugin between 1.8.10 (the reporters version) and the current git `main` branch.

jimzhang2, 2023-02-23:
@hughsie Hello~ We have some change in 1.8.7. Thanks!

hughsie, 2023-02-23:
@jimzhang2 I know, but the reporter of the bug is using a newer version than that -- they're using 1.8.10. Or do you mean it works in the older version but does not work in the newer version?

jimzhang2, 2023-02-23:
@hughsie It's weird. We have tested it in ubuntu20.04 with latest source code. We need more check.

jimzhang2, 2023-02-24:
@qwetwe Hi~ We have problems about installing ALT linux in our desktop. So, do you have an email or a telegram account? Thanks!

Blarse, 2023-02-24:
> @qwetwe Hi~ We have problems about installing ALT linux in our desktop. So, do you have an email or a telegram account? Thanks!

Hi, I am the maintainer of fwupd in ALT Linux, you can contact me via egori@altlinux.org or telegram @blarse

hughsie, 2023-02-24:
@jimzhang2 I don't think we have any expectations that you need to install a different distro :) @qwetwe can you confirm if it also fails on a Fedora or Ubuntu LIveUSB please?

qwetwe, 2023-03-01:
> @jimzhang2 I don't think we have any expectations that you need to install a different distro :) @qwetwe can you confirm if it also fails on a Fedora or Ubuntu LIveUSB please?

I won't do this, because we will not be able to reproduce this error again on our OS.
Or do you think that we will be able to downgrade the firmware later?

hughsie, 2023-03-01:
> I won't do this, because we will not be able to reproduce this error again on our OS.

If you think the problem is OS specific then it's not the hardware vendors problem.

> Or do you think that we will be able to downgrade the firmware later?

I think that might be a dangerous assumption.

superm1, 2023-03-01:
I think the only part that might factor into it being OS specific is the gusb version.  Can you upgrade to the latest gusb and still reproduce it?

Blarse, 2023-03-01:
> > I won't do this, because we will not be able to reproduce this error again on our OS.
> 
> If you think the problem is OS specific then it's not the hardware vendors problem.

Well, I think the end goal for both hardware and OS vendors is to find and fix this issue for all users, not just our case and hope that it won’t break later elsewhere.
I'm pretty sure this problem isn't OS specific, but I would wait before @jimzhang2 can reproduce it on his side.
We are ready to help him and provide any necessary diagnostic information.

Best we can do until then is to try and confirm if it also fails after updating the system from our rolling release repository.

Blarse, 2023-03-01:
> I think the only part that might factor into it being OS specific is the gusb version. Can you upgrade to the latest gusb and still reproduce it?

Hm, sounds promising. We will try this tomorrow anyway, but do you think this problem could have been fixed somewhere between gusb 0.4.1 and 0.4.5

Blarse, 2023-03-02:
> Can you upgrade to the latest gusb and still reproduce it?

Unfortunately, it still reproduces with gusb 0.4.5

jimzhang2, 2023-03-15:
Hello, every one !
I have installed  http://beta.altlinux.org/kworkstation/ on HP G4 laptop.
And connect a fingerprint scanner through USB.

fwupdmgr version:
[root@lemon2onward ~]# fwupd
fwupdagent  fwupdate    fwupdmgr    fwupdtool
[root@lemon2onward ~]# fwupdmgr  --version
compile   org.freedesktop.fwupd         1.8.10
compile   com.hughsie.libjcat           0.1.11
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.4.1
runtime   com.hughsie.libjcat           0.1.11
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.1
runtime   org.kernel                    5.15.80-un-def-alt1
runtime   org.freedesktop.fwupd         1.8.10

firmware upgrade log:
 Update FPC Fingerprint Reader Firmware:                                      ║
║                                                                              ║
║ • Support FPC Disum Fingerprint Reader Device(10A5:9800)                     ║
║                                                                              ║
║ Sensor Controller L:0001 FW:16.26.23.14 and all connected devices may not    ║
║ be usable while updating.                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
Waiting…                 [***************************************]
Idle…                    [-                                      ]
Idle…                    [***************************************]
Successfully uploaded 1 report
Devices with the latest available firmware version:
 • UEFI dbx

The firmware upgrading seems successful.

Blarse, 2023-03-15:
> The firmware upgrading seems successful.

Hi, did you use fwupd from repository or build it yourself?

> Could you try the same OS on your laptop? Thanks!

Ok, we will try

jimzhang2, 2023-03-17:
@Blarse Hello~
fwupd is not built by the source code.
It's built in the system.
Thanks!

Blarse, 2023-04-11:
Hello, everyone!

I wanted to see if this fingerprint device would work on Windows, so I installed it and all of a sudden Windows automatically upgraded the device firmware to 26.26.23.31. What's interesting is that fwupd still fails to update firmware with the same error, now from 26.26.23.31 to 27.26.23.19.

So the device itself is working. I would try fwupd on Ubuntu or Fedora if it would be possible to downgrade the firmware back in case of success, so that the problem can be reproduced on AltLlnux. Is it even possible to downgrade firmware from either Linux or Windows?

hughsie, 2023-09-23:
@qwetwe is this good now?

qwetwe, 2023-09-25:
@hughsie greetings!

At the moment I can't get updates for the device:

<details>

<summary># fwupdmgr get-updates -v</summary>

```
[root@qwetwe ~]# LANG=en_EN fwupdmgr get-updates -v
(fwupdmgr:12570): GLib-DEBUG: 19:17:37.517: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:12570): GLib-GIO-DEBUG: 19:17:37.526: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ?gsettings-backend?
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.526: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.527: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.527: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.527: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:12570): GLib-GIO-DEBUG: 19:17:37.527: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ?gio-proxy-resolver?
(fwupdmgr:12570): Fwupd-DEBUG: 19:17:37.530: Emitting ::status-changed() [idle]
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.531: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.531: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.532: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.532: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:12570): dconf-DEBUG: 19:17:37.532: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:12570): FuMain-DEBUG: 19:17:37.541: current version is 7003V5LN: 7003V5LN=same
(fwupdmgr:12570): FuMain-DEBUG: 19:17:37.547: current version is 371: 371=same
Devices with no available firmware updates: 
 ? Integrated RGB Camera
 ? System Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
 ? UEFI Device Firmware
Devices with the latest available firmware version:
 ? Micron MTFDKCD256TFK
 ? UEFI dbx
(fwupdmgr:12570): FuMain-DEBUG: 19:17:37.568: lvfs has no RemoteURI
(fwupdmgr:12570): FuMain-DEBUG: 19:17:37.568: no unreported devices
No updates available
```

</details>

Also, the sensor device has stopped being displayed in the list of all devices that support firmware:

<details>

<summary># fwupdmgr get-devices -v</summary>

```
[root@qwetwe ~]# fwupdmgr get-devices -v
(fwupdmgr:12725): GLib-DEBUG: 19:21:07.475: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:12725): GLib-GIO-DEBUG: 19:21:07.508: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.508: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.509: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.511: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.511: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:12725): GLib-GIO-DEBUG: 19:21:07.513: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.517: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:12725): dconf-DEBUG: 19:21:07.517: watch_established: "/system/proxy/http/" (establishing: 0)
[... 172 lines cut ...]
  VersionRaw:           0x0001000a
  VersionLowestRaw:     0x0001000a
  Created:              2023-09-25
  UpdateState:          success

(fwupdmgr:12725): FuMain-DEBUG: 19:21:07.567: lvfs has no RemoteURI
(fwupdmgr:12725): FuMain-DEBUG: 19:21:07.567: no unreported devices
```

</details>

Fwupd version info:
```
[root@qwetwe ~]# LANG=en_EN fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.8
compile   com.hughsie.libjcat           0.1.11
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.11
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.55-un-def-alt1
runtime   org.freedesktop.fwupd         1.9.5
```

hughsie, 2023-09-26:
> fwupdmgr get-devices -v

Can you get the output from `sudo fwupdtool --plugins fpc get-devices -vv` -- thanks!

qwetwe, 2023-09-27:
> > fwupdmgr get-devices -v
> 
> Can you get the output from `sudo fwupdtool --plugins fpc get-devices -vv` -- thanks!

Sure. Here it is: <details>
```
[qwetwe@qwetwe ~]$ sudo fwupdtool --plugins fpc get-devices -vv
16:53:23.398 FuDebug              verbose to debug (on console 1)
16:53:23.398 FuEngine             starting fwupd 1.9.5…
16:53:23.399 FuMain               locked /run/lock/fwupdtool
16:53:23.404 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Загрузка…                [ -                                     ]16:53:23.405 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
16:53:23.405 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
16:53:23.405 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
16:53:23.405 FuCommon             reading /etc/fwupd/fwupd.conf with 119 bytes
16:53:23.405 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
16:53:23.405 FuConfig             Не удалось открыть файл «/var/etc/fwupd/fwupd.conf»: Нет такого файла или каталога
16:53:23.405 FuConfig             ::configuration loaded
16:53:23.405 FuRemoteList         loading /usr/share/fwupd/metainfo
16:53:23.405 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:53:23.405 XbSilo               file: 9267b256-5394-b966-8def-acb1877adb8a, current:9267b256-5394-b966-8def-acb1877adb8a, cached: (null)
16:53:23.405 XbSilo               loading silo with file contents
16:53:23.405 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
16:53:23.405 FuRemoteList         ::remote_list changed
16:53:23.405 FuEngine             priority local and dell-esrt is not download remote, so bumping
16:53:23.405 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:53:23.405 FuRemoteList         ::remote_list changed
16:53:23.405 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:53:23.405 FuRemoteList         ::remote_list changed
16:53:23.406 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:53:23.406 FuRemoteList         ::remote_list changed
16:53:23.406 FuEngine             priority local and vendor-directory is not download remote, so bumping
16:53:23.406 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:53:23.406 FuRemoteList         ::remote_list changed
16:53:23.406 FuEngine             priority local and vendor is not download remote, so bumping
16:53:23.406 FuRemoteList         ordering lvfs-testing=lvfs+1
16:53:23.406 FuRemoteList         enabled remotes: dell-esrt[1000], vendor-directory[1000], lvfs[0]
Загрузка…                [                                       ]16:53:23.406 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:53:23.406 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:53:23.406 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:53:23.406 Jcat                 loaded 1 certificates
16:53:23.406 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:53:23.406 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:53:23.406 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:53:23.406 Jcat                 loaded 1 certificates
16:53:23.406 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
16:53:23.406 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:53:23.406 FuEngine             client certificate now exists: nothing to do
Загрузка…                [                                       ]16:53:23.406 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:53:23.407 FuHistory            got schema version of 9
Загрузка…                [*                                      ]16:53:23.409 GLib                 setenv()/putenv() are not thread-safe and should not be used after threads are created
16:53:23.409 FuPluginTest         init
16:53:23.409 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:53:23.409 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
16:53:23.409 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:53:23.409 XbSilo               file: c6352be8-d0f8-51b4-0aa2-59094417dfcc, current:c6352be8-d0f8-51b4-0aa2-59094417dfcc, cached: (null)
16:53:23.409 XbSilo               loading silo with file contents
Загрузка…                [*                                      ]16:53:23.410 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
16:53:23.410 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
16:53:23.410 FuCommon             reading /etc/fwupd/fwupd.conf with 119 bytes
16:53:23.410 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
16:53:23.410 FuConfig             Не удалось открыть файл «/var/etc/fwupd/fwupd.conf»: Нет такого файла или каталога
16:53:23.410 FuConfig             ::configuration loaded
16:53:23.410 FuStruct             SmbiosEp64:
  anchor_str: _SM3_
  entry_point_csum: 0xc4
  entry_point_len: 0x18
  smbios_major_ver: 0x3
[... cut ...]
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuFpcPlugin:
  Name:                 fpc

16:53:23.491 FuEngine             resetting update motd timeout

No detected devices
16:53:23.491 FuMain               No detected devices

16:53:23.492 FuPluginTest         destroy
16:53:23.493 FuProgress           raw timing data was { 0,001, 0,000, 0,004, 0,023 } -- steps were set as [ 1 1 3 95 ] but should have been [ 2 0 13 84 ] at ../libfwupdplugin/fu-context.c:967
```
</details>

hughsie, 2023-09-28:
Adding to this a bit, `libusb_get_string_descriptor_ascii()` requests a string from the device with a 255 byte buffer (`usbi_string_desc_buf`) -- I think `LIBUSB_TRANSFER_OVERFLOW` is being returned because the device is sending more data than requested. Is that plausible?

Winson-Huang, 2023-12-23:
I have similar issue in ThinkPad neo 14, my OS is Debian 12.

kauzu, 2024-01-02:
This still seems to be a problem. I'm getting the same error message as @qwetwe

`22:27:13.951 FuEngine             usb:03:00:03 added FuUsbDevice:
  Guid:                 bcb666d2-d13b-5295-b6ef-32a3f4ba8737 ← USB\VID_10A5&PID_9800 ⚠
  Flags:                none
  VendorId:             USB:0x10A5
  Version:              23.14
  VersionFormat:        bcd
  VersionRaw:           0x00002314
  Guid[quirk]:          dd4a4bc8-89bb-5453-9c8d-70ccd761b725 ← USB\VID_10A5
  Guid[quirk]:          4bd986bc-4aaf-5baf-8bc2-9f7928182387 ← USB\CLASS_FF
  Guid[quirk]:          9931332d-2ffb-56ab-9b87-180a924d40e7 ← USB\CLASS_FF&SUBCLASS_FF
  Guid[quirk]:          b4010dfb-3098-5ca7-9b60-73c1c541288e ← USB\CLASS_FF&SUBCLASS_FF&PROT_FF
  PhysicalId:           usb:03:00:03
  AcquiesceDelay:       2500
  CustomFlags:          moh-device,rts,lenfy,enforce-requires
  PossiblePlugin:       fpc
  ParentPhysicalIds:    usb:03:00,usb:03
  InternalFlags:        enforce-requires
  UsbDeviceClass:       interface-desc

22:27:13.953 FuDevice             using 0d5d05911800242bb1f35287012cdcbd9b381148 for usb:03:00:03
22:27:14.072 FuDevice             removing vendor prefix of 'FPC' from 'FPC Sensor Controller L:0001 FW:16.26.23.14'
22:27:14.073 FuDevice             0d5d05911800242bb1f35287012cdcbd9b381148 device overwriting name value: Sensor Controller L:0001 FW:16.26.23.14->Sensor Controller
22:27:14.073 FuUsbDevice          re-binding kernel driver as not waiting for replug
22:27:14.074 FuEngine             failed to add device usb:03:00:03: failed to get firmware version: only sent 0x0028 of 0x0048
` 
`m@ubuntu-tp-1 ~ % sudo dmidecode -t 2
# dmidecode 3.5
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0010, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: LENOVO
	Product Name: 21EES00100
	Version: ThinkPad
	Serial Number: L1HF237021D
	Asset Tag: Not Available
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Available
	Chassis Handle: 0x0000
	Type: Motherboard
	Contained Object Handles: 0`

`lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0bda:4853 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 002: ID 10a5:9800 FPC FPC Sensor Controller L:0001 FW:16.26.23.14
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Bus 001 Device 005: ID 1532:024e Razer USA, Ltd Razer BlackWidow V3
Bus 001 Device 007: ID 0409:05cc NEC Corp. EA242F
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 002: ID 5986:2146 Bison Electronics Inc. Integrated RGB Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

`m@ubuntu-tp-1 ~ % cat /etc/issue
Ubuntu 23.10 \n \l

m@ubuntu-tp-1 ~ % uname -a
Linux ubuntu-tp-1 6.5.0-14-generic #14-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:59:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

helpvisa, 2024-08-24:
Still experiencing a similar issue on Fedora 40 on a Thinkpad E16 Gen1 AMD. Update fails from within GNOME software and also when using fwupdmgr from the commandline.

Typing ```sudo fwupdmgr update -v``` returns the following:
```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Sensor Controller from 27.26.23.33 to 27.26.23.50?                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Update FPC Fingerprint Reader Firmware:                                      ║
║                                                                              ║
║ • Support FPC linux Fingerprint Reader Device(10A5:9800)                     ║
║                                                                              ║
║ Sensor Controller and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Fwupd-INFO: 19:15:17.024: downloading https://fwupd.org/downloads/36e1ad9e0547885c8dccd5f917148c938452b7111652ab486d3f8d701e747bf9-fpc-lenfy-moh-v27_26_23_50.cab
(fwupdmgr:10035): pxbackend-DEBUG: 19:15:17.024: px_manager_get_proxies_sync: url=https://fwupd.org/downloads/36e1ad9e0547885c8dccd5f917148c938452b7111652ab486d3f8d701e747bf9-fpc-lenfy-moh-v27_26_23_50.cab online=1
(fwupdmgr:10035): pxbackend-DEBUG: 19:15:17.025: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.025: Emitting ::status-changed() [downloading]
Downloading…             [           |                           ]Fwupd-INFO: 19:15:17.445: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 19:15:17.571: download progress: 22%
Downloading…             [********                               ]Fwupd-INFO: 19:15:17.571: download progress: 24%
Downloading…             [*********                              ]Fwupd-INFO: 19:15:17.572: download progress: 26%
Downloading…             [**********                             ]Fwupd-INFO: 19:15:17.572: download progress: 28%
Downloading…             [**********                             ]Fwupd-INFO: 19:15:17.572: download progress: 30%
Downloading…             [***********                            ]Fwupd-INFO: 19:15:17.572: download progress: 32%
Downloading…             [************                           ]Fwupd-INFO: 19:15:17.572: download progress: 36%
Downloading…             [**************                         ]Fwupd-INFO: 19:15:17.573: download progress: 37%
Downloading…             [**************                         ]Fwupd-INFO: 19:15:17.573: download progress: 41%
Downloading…             [***************                        ]Fwupd-INFO: 19:15:17.573: download progress: 43%
Downloading…             [****************                       ]Fwupd-INFO: 19:15:17.574: download progress: 44%
Downloading…             [*****************                      ]Fwupd-INFO: 19:15:17.574: download progress: 48%
Fwupd-INFO: 19:15:17.574: download progress: 52%
Downloading…             [********************                   ]Fwupd-INFO: 19:15:17.574: download progress: 53%
Downloading…             [********************                   ]Fwupd-INFO: 19:15:17.576: download progress: 55%
Downloading…             [*********************                  ]Fwupd-INFO: 19:15:17.576: download progress: 63%
Downloading…             [************************               ]Fwupd-INFO: 19:15:17.576: download progress: 65%
Downloading…             [*************************              ]Fwupd-INFO: 19:15:17.576: download progress: 67%
Downloading…             [**************************             ]Fwupd-INFO: 19:15:17.577: download progress: 71%
Downloading…             [***************************            ]Fwupd-INFO: 19:15:17.577: download progress: 72%
Downloading…             [****************************           ]Fwupd-INFO: 19:15:17.578: download progress: 78%
Downloading…             [******************************         ]Fwupd-INFO: 19:15:17.578: download progress: 84%
Downloading…             [********************************       ]Fwupd-INFO: 19:15:17.579: download progress: 89%
Downloading…             [**********************************     ]Fwupd-INFO: 19:15:17.580: download progress: 100%
Downloading…             [***************************************](fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.580: Emitting ::status-changed() [idle]
Fwupd-INFO: 19:15:17.580: status-code was 200
(fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.583: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.611: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.632: Emitting ::status-changed() [device-busy]
Waiting…                 [ -                                     ](fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.632: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.632: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:10035): Fwupd-DEBUG: 19:15:17.664: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
[... cut ...]
Writing…                 [                                       ] 4 minutes remaining…(fwupdmgr:10035): Fwupd-DEBUG: 19:15:22.314: Emitting ::status-changed() [decompressing]
Decompressing…           [                                       ](fwupdmgr:10035): Fwupd-DEBUG: 19:15:22.316: Emitting ::status-changed() [device-busy]
Waiting…                 [                                       ](fwupdmgr:10035): Fwupd-DEBUG: 19:15:22.326: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)

failed to initial update: failed to clear status: endpoint stalled or request not supported
```

Have tried disabling Secure Boot but end up with the same results.

bluesceada, 2024-09-02:
I am having a similar problem on a Thinkpad E14 Gen 4 AMD. 

It gets stuck at "Restarting device..." for a few seconds and looks as follows in the end:

```
 ~/ > sudo fwupdmgr update -v
(pkttyagent:8041): GLib-GIO-DEBUG: 18:44:15.470: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Devices with no available firmware updates: 
 • Integrated RGB Camera
 • SSD 980 1TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Sensor Controller from 16.26.23.14 to 27.26.23.50?                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Update FPC Fingerprint Reader Firmware:                                      ║
║                                                                              ║
║ • Support FPC linux Fingerprint Reader Device(10A5:9800)                     ║
║                                                                              ║
║ Sensor Controller and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [                                       ] 4 minutes remaining…
failed to initial update: failed to clear status: endpoint stalled or request not supported
```

sokol-44, 2024-09-18:
I have similar problem like @bluesceada my system is Fedora 40 on Lenovo Thinkpad E14 Gen 4 AMD.

```
#  fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.23
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.7
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.10.8-200.fc40.x86_64
runtime   org.freedesktop.fwupd         1.9.23
```

```
# fwupdmgr update -v
Devices with the latest available firmware version:
 ? Micron MTFDKCD512TFK
(fwupdmgr:380006): FuMain-DEBUG: 09:36:48.527: current version is 7003V5LN: 7003V5LN=same
????????????????????????????????????????????????????????????????????????????????
? Upgrade Sensor Controller from 16.26.23.14 to 27.26.23.50?                   ?
????????????????????????????????????????????????????????????????????????????????
? Update FPC Fingerprint Reader Firmware:                                      ?
?                                                                              ?
? ? Support FPC linux Fingerprint Reader Device(10A5:9800)                     ?
?                                                                              ?
? Sensor Controller and all connected devices may not be usable while          ?
? updating.                                                                    ?
????????????????????????????????????????????????????????????????????????????????
Perform operation? [Y|n]: y
Fwupd-INFO: 09:36:52.637: downloading https://fwupd.org/downloads/36e1ad9e0547885c8dccd5f917148c938452b7111652ab486d3f8d701e747bf9-fpc-lenfy-moh-v27_26_23_50.cab
[..... downloading ......]
Decompressing?           [***************************************](fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.774: Emitting ::status-changed() [waiting-for-auth]
Authenticating?          [***************************************](fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.793: Emitting ::status-changed() [device-busy]
Waiting?                 [***************************************](fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.793: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.793: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.801: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.801: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.900: Emitting ::status-changed() [device-write]
Writing?                 [                                       ](fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.900: Emitting ::status-changed() [device-restart]
Restarting device?       [                                       ](fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.901: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.967: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:53.967: Emitting ::changed()
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.409: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.409: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.409: Emitting ::changed()
(fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.415: Emitting ::status-changed() [device-write]
Writing?                 [                                       ] 4 minutes remaining?(fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.415: Emitting ::status-changed() [decompressing]
Decompressing?           [                                       ](fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.416: Emitting ::status-changed() [device-busy]
Waiting?                 [                                       ](fwupdmgr:380006): Fwupd-DEBUG: 09:36:58.425: Emitting ::device-changed(0d5d05911800242bb1f35287012cdcbd9b381148)

failed to initial update: failed to clear status: endpoint stalled or request not supported
```
[fwupdtool --plugins fpc get-devices -vv --json](https://github.com/user-attachments/files/17040442/fwupdtool_--plugins_fpc_get-devices_-vv.json)

JeffKernsGitHub, 2024-10-01:
Same issue on Lenovo ThinkPad E16 Gen 1 AMD

hughsie, 2024-11-11:
> Lenovo ThinkPad E16 Gen 1 AMD

@superm1 could you get hold of one of these and grab some emulation data?

superm1, 2024-11-11:
> > Lenovo ThinkPad E16 Gen 1 AMD
> 
> @superm1 could you get hold of one of these and grab some emulation data?

I don't have access to one no. Maybe @mrhpearson can.

kauzu, 2024-11-11:
@hughsie  i have access to a ThinkPad E15 Gen 4 AMD with the same fingerprint reader and the same problem. 

If someone can tell me how to grab some emulation data ill do that

hughsie, 2024-11-11:
@kauzu if you can build the main branch, there are instructions here: https://github.com/fwupd/fwupd/blob/main/docs/device-emulation.md -- but @jimzhang2 will need to help us as we need an emulation where it *work* rather than *fails* if you see what I mean.

mrhpearson, 2024-11-12:
I don't have any of these systems on hand and they're not in the Linux program so I can't make any promises...
I've created internal ticket LO-3396 to see if we can get hold of one to repro and help out.

mrhpearson, 2024-11-14:
@hughsie - discussing with FPC, and the FW update is only supposed to be available for specific CHIDs (in the metadata).
How can we check what the CHID is on a system, to understand why the FW is being offered. I couldn't figure out how to find it.
As a note (to be confirmed) - I think the E16 G1 AMD should work, but the E14/E15 G4 shouldn't. It's not the complete answer, but it does seem the FW is being offered in some cases when it shouldn't be.

mrhpearson, 2024-11-14:
Found the answer (actually my colleague did :)). Can those who reported the issue on the ticket do:
`fwupdtool hwids`
and share the results please.

hughsie, 2024-11-14:
> How can we check what the CHID is on a system

`sudo fwupdtool hwids` is what you want.

helpvisa, 2024-11-14:
Adding mine here as well:
```
Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: R2CET29W (1.11 )
BiosMajorRelease: 1
BiosMinorRelease: 11
FirmwareMajorRelease: 01
FirmwareMinorRelease: 0b
Manufacturer: LENOVO
Family: ThinkPad E16 Gen 1
ProductName: 21JT001PUS
ProductSku: LENOVO_MT_21JT_BU_Think_FM_ThinkPad E16 Gen 1
EnclosureKind: a
BaseboardManufacturer: LENOVO
BaseboardProduct: 21JT001PUS
Hardware IDs
------------
{bf9b69a7-7c9a-51a0-a333-309d7b5c4e98}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{9ec85115-3c7e-509e-8891-eeabdd8f479a}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{1fca72ae-e766-5ba3-8df0-865efa2a6dab}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{c9507d51-4d67-5f58-b8e5-001d64a6a902}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{18b7e9d9-5029-5e42-9b0a-f29ad955fe72}   <- Manufacturer + Family + ProductName + ProductSku
{e453d4b6-8ab5-50ea-8134-647059a23351}   <- Manufacturer + Family + ProductName
{97eba7e0-ef0c-521c-a63c-6268817bc5a2}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{e1daca05-c165-5ecf-832e-a4190066facd}   <- Manufacturer + ProductSku
{493eb478-5c16-5585-95b8-3c063c3503ce}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{e650b6c8-cf71-51d0-8a82-38e0240f7de3}   <- Manufacturer + ProductName
{91efa1c5-711e-5480-932d-3285936acd88}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{a8b209c1-d74d-5de0-9ee7-3458cc4ae4dd}   <- Manufacturer + Family
{e093d715-70f7-51f4-b6c8-b4a7e31def85}   <- Manufacturer + EnclosureKind
{1c8e1222-4887-5f16-a50c-2d84609d0478}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer
Extra Hardware IDs
------------------
{bc09bbe7-7282-58ca-be7d-31826c70da43}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{8d4586a6-c20c-5a24-aa50-a50143c0dd15}   <- Manufacturer + Family + ProductName + BiosVendor
{99431f53-09a1-5869-be79-65e2fa3f341d}   <- Manufacturer + BiosVendor

```

hughsie, 2024-11-14:
Aha, so on the LVFS we have two firmwares that match the GUID of `bcb666d2-d13b-5295-b6ef-32a3f4ba8737`:

 * https://fwupd.org/lvfs/firmware/88372 - `com.fingerprints.Lenovo.cs23.9800.firmware` (v27.26.23.50)
 * https://fwupd.org/lvfs/firmware/14580 - `com.fingerprints.lenfy.firmware` (v27.26.23.19)

The first firmware has a boatload of CHIDs restricting it down to 14 different (Lenovo?) models -- but the second has no restrictions and seems to be available to everyone.

Looking at the logs here in a bit more detail, that GUID seems to map to `USB\VID_10A5&PID_9800` which looks like a *generic* VID/PID -- and looking at https://github.com/fwupd/fwupd/blob/main/plugins/fpc/fpc.quirk#L8 we can see that it has the enforce-requires flag. Which should mean that com.fingerprints.lenfy.firmware isn't installable on any device as it doesn't have any CHID requirements -- which was functionality we added in https://github.com/fwupd/fwupd/commit/8dba5fad5b56ad08f9d42251a8024c3e9a81157b which was included in fwupd 1.9.8.

So, like all tricky issues we have a few bugs:

 * Ideally the FPC devices should have a unique VID/PID for each firmware stream
 * Maybe the LVFS could have _warned_ the uploader that `com.fingerprints.lenfy.firmware` and `com.fingerprints.Lenovo.cs23.9800.firmware` are both targetting the same GUID
 * We shouldn't be suggesting a non-CHID'd update to anyone running older than fwupd 1.9.8
 * The LVFS should probably have a list of generic GUIDs that absolutely need fwupd 1.9.8

The easiest fix is to change the org.freedesktop.fwupd requirement from `ge` **1.3.6** to `ge` **1.9.8** to the `com.fingerprints.lenfy.firmware` firmware, and that's just what I've done. I'll ponder the others. It'll take 4h for metadata to regenerate.

mrhpearson, 2024-11-14:
Thanks Richard! (side note - the CHIDs cover 5 platforms - some of them have multiple CHIDs)

I don't have access to view the 14580 FW, but should that just be demoted if it can't/shouldn't be installed on anything? I wonder if it's a leftover from earlier debug (I know it took a few iterations for FPC to get it working). I can ask them on that.

We still have the open question as to why it didn't work on @helpvisa system (which is supported) - but I think that's a separate issue.

Mark

helpvisa, 2024-11-14:
It may be worth mentioning that the FPC fingerprint sensor has never properly functioned for any considerable period of time under Fedora (or any other distro) for me. It did briefly work at one point with a fork of fprintd and a bespoke driver but after a fresh reinstall stopped working once more.

I can do some digging through old files and see if I can retrace my steps re: the sensor itself. Perhaps it's a hardware issue on my end.

hughsie, 2024-11-14:
> but should that just be demoted if it can't/shouldn't be installed on anything

Ohh it's valid; it's just unfortunate that all the FPC devices seem to [intentionally] have the same VID/PID. e.g. see [xxx](https://lore.kernel.org/lvfs-announce/CAD2FfiF5CbWH0gsJmGg5zW2F57Z6t-CBThe2b54iMDakdyf32Q@mail.gmail.com/T/#u) for some bedtime reading.

sokol-44, 2024-11-15:
So now i can just run `fwupdmgr update -v` and it should work ? [Or do you prefer that i go to build  "device information" and grab some more data] ?

hughsie, 2024-11-15:
> So now i can just run fwupdmgr update -v and it should work

Well, if you were matching the `com.fingerprints.lenfy.firmware` component then that just won't be suggested any more hopefully.

xuiqzy, 2024-11-22:
Below is the output for `sudo fwupdtool hwids` on the Lenovo E16 Gen 1, because I have slightly different output than @helpvisa who also has an E16 Gen 1, maybe due to different BIOS version or hardware versions, see these values in the output.

FYI, I get offered the update `from 27.26.23.33 to 27.26.23.50` (so ending with `50` but it seems to download the one with `lenfy` in the name but still ending with `50`. So not matching the relationship between name and version from comment https://github.com/fwupd/fwupd/issues/5573#issuecomment-2477210132 by @hughsie  !), but it fails to install while at the waiting stage (showed `restarting device...` and `writing` already before that) with:
```
sudo fwupdmgr update -v
(pkttyagent:67196): GLib-GIO-DEBUG: 00:53:33.718: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:67194): GLib-GIO-DEBUG: 00:53:33.728: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.728: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.728: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.728: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.728: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.728: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:67194): GLib-GIO-DEBUG: 00:53:33.728: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.733: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.733: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.733: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.733: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:67194): dconf-DEBUG: 00:53:33.733: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:67194): GLib-GIO-DEBUG: 00:53:33.734: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:67194): Fwupd-DEBUG: 00:53:33.821: Emitting ::status-changed() [idle]
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Sensor Controller from 27.26.23.33 to 27.26.23.50?                   ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Update FPC Fingerprint Reader Firmware:                                      ║
║                                                                              ║
║ • Support FPC linux Fingerprint Reader Device(10A5:9800)                     ║
║                                                                              ║
║ Sensor Controller and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Fwupd-INFO: 00:53:35.322: downloading https://fwupd.org/downloads/36e1ad9e0547885c8dccd5f917148c938452b7111652ab486d3f8d701e747bf9-fpc-lenfy-moh-v27_26_23_50.cab
(fwupdmgr:67194): Fwupd-DEBUG: 00:53:35.323: Emitting ::status-changed() [downloading]
Downloading…             [                   -                   ]Fwupd-INFO: 00:53:36.085: download progress: 100%
Downloading…             [***************************************]Fwupd-INFO: 00:53:36.532: download progress: 22%
Downloading…             [********                               ]Fwupd-INFO: 00:53:36.532: download progress: 46%
Downloading…             [*****************                      ]Fwupd-INFO: 00:53:36.532: download progress: 69%
Downloading…             [**************************             ]Fwupd-INFO: 00:53:36.532: download progress: 83%
Downloading…             [********************************       ]Fwupd-INFO: 00:53:36.557: download progress: 89%
Downloading…             [**********************************     ]Fwupd-INFO: 00:53:36.557: download progress: 91%
Downloading…             [***********************************    ]Fwupd-INFO: 00:53:36.558: download progress: 96%
Downloading…             [*************************************  ]Fwupd-INFO: 00:53:36.558: download progress: 100%
(fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.558: Emitting ::status-changed() [idle]
Fwupd-INFO: 00:53:36.558: status-code was 200
(fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.563: Emitting ::status-changed() [decompressing]
Decompressing…           [***************************************](fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.619: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [  -                                    ](fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.670: Emitting ::status-changed() [device-busy]
Waiting…                 [  -                                    ](fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.670: Emitting ::device-changed(ff75e69eba7f7bc73bf2e2524de4c41d32d61506)
(fwupdmgr:67194): Fwupd-DEBUG: 00:53:36.670: Emitting ::device-changed(ff75e69eba7f7bc73bf2e2524de4c41d32d61506)
[... cut ...]
{920421a6-e3a9-53a7-8876-5a08339228b0}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer
Extra Hardware IDs
------------------
{788ebe0b-3d84-5919-bfb5-ed0171af7a5f}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{cf8ea1df-04fb-5514-9480-3d9ca08b9c13}   <- Manufacturer + Family + ProductName + BiosVendor
{99431f53-09a1-5869-be79-65e2fa3f341d}   <- Manufacturer + BiosVendor
```

Is the update supposed to work already on the ThinkPad E16 Gen 1?

mrhpearson, 2024-11-26:
@xuiqzy - None of those CHIDs match - so the FW won't load on your device.

The E16 G1 AMD was not Linux certified, so my guess is that as we won't have tested it won't get the .50 update.

I don't know why the .50 version only applies to a few platforms I'm afraid - I'll see if I can find out. It's curious as the .19 version is there for everything. It might just be a decision to limit it to platforms where the update can be tested? I'm guessing.

sokol-44, 2024-12-23:
After some time on Fedora 40 (and 41) i don't see update request, and current version stays at: 16.26.23.14.
```10:59:30.712 FuMain               FuFpcDevice:
  DeviceId:             0d5d05911800242bb1f35287012cdcbd9b381148
  Name:                 Sensor Controller
  Guid:                 bcb666d2-d13b-5295-b6ef-32a3f4ba8737 ← USB\VID_10A5&PID_9800
  Summary:              FPC fingerprint sensor
  Plugin:               fpc
  Protocol:             com.fingerprints.dfupc
  Flags:                updatable|registered|use-runtime-version|signed-payload
  Vendor:               FPC
  VendorId:             USB:0x10A5
  Version:              16.26.23.14
  VersionFormat:        quad
  VersionRaw:           0x101a170e
  InstallDuration:      15
  Created:              2024-12-23
```
So i didn't get update but nagging stopped.

closed 2025-05-04 by hughsie